ttlappalainen/NMEA2000 #213: delete ( (tNMEA_SocketCAN *)) issues

lsoltero, 2021-04-16:
the following code

`
#define SOCKET_CAN_PORT (char *)"can1"
tNMEA2000_SocketCAN &NMEA2000=*(new tNMEA2000_SocketCAN(SOCKET_CAN_PORT));

main() {
      tNMEA2000_SocketCAN *t=new tNMEA2000_SocketCAN((char *)"can0");
      delete (t);  // causes illigal instruction in N2kStream...  so live with the memory leak
  }

causes the following compiler warning.

g++ -c -std=c++11 -g -Wall -Werror -I /usr/local/include/nmea0183 -I /usr/local/include/n2k nmead.cpp -o nmead.o
nmead.cpp: In function ‘bool setup_n2k()’:
nmead.cpp:367:16: error: deleting object of polymorphic class type ‘tNMEA2000_SocketCAN’ which has non-virtual destructor might cause undefined behaviour [-Werror=delete-non-virtual-dtor]
       delete (t);  // causes illigal instruction in N2kStream...  so live with the memory leak
`

easy enough to fix by adding a virtual destructor to tNMEA2000_SocketCAN.  After doing so the compile is happy but on the execution of 

`delete (t);`

you get the following

`
(gdb) 
367	      delete (t);
(gdb) p t
$1 = (tNMEA2000_SocketCAN *) 0x6483b0
(gdb) p *t
$2 = {<tNMEA2000> = {_vptr.tNMEA2000 = 0x42a260 <vtable for tNMEA2000_SocketCAN+16>, static FwdModeBit_EnableForward = 1, static FwdModeBit_SystemMessages = 2, 
    static FwdModeBit_OnlyKnownMessages = 4, static FwdModeBit_OwnMessages = 8, static HandleModeBit_OnlyKnownMessages = 16, dbMode = tNMEA2000::dm_None, 
    N2kMode = tNMEA2000::N2km_ListenOnly, ForwardType = tNMEA2000::fwdt_Actisense, ForwardMode = 11, ForwardStream = 0x0, MsgHandlers = 0x0, DeviceReady = false, 
    AddressChanged = false, DeviceInformationChanged = false, Devices = 0x0, DeviceCount = 1, LocalConfigurationInformationData = 0x0, ConfigurationInformation = {
      ManufacturerInformation = 0x42a060 <DefManufacturerInformation> "NMEA2000 library, https://github.com/ttlappalainen/NMEA2000", 
      InstallationDescription1 = 0x42a09c <DefInstallationDescription1> "", InstallationDescription2 = 0x42a09d <DefInstallationDescription2> ""}, SingleFrameMessages = {0x0, 0x0}, 
    FastPacketMessages = {0x0, 0x0}, N2kCANMsgBuf = 0x0, MaxN2kCANMsgs = 0 '\000', CANSendFrameBuf = 0x0, MaxCANSendFrames = 40, CANSendFrameBufferWrite = 28732, 
    CANSendFrameBufferRead = 65535, MaxCANReceiveFrames = 0, MsgHandler = 0x0, ISORqstHandler = 0x0, pGroupFunctionHandlers = 0x0, InstallationDescriptionChanged = false}, skt = -1, 
  _CANport = 0x7fffffffe7d8 "can0"}
(gdb) n

Thread 1 "nmead" received signal SIGILL, Illegal instruction.
0x000000000042a318 in typeinfo for N2kStream ()

(gdb) where
#0  0x000000000042a318 in typeinfo for N2kStream ()

`

not sure why... 

comment out the delete() and everything works fine...

ttlappalainen, 2021-04-16:
You are the first one who complains about delete; I have not met any real case yet, where you would need to delete NMEA2000 object. Normally library will be used on devices communicting with N2 bus. So they will be turned on and turned of - delete has nothing to do there, when you turn it off. That is simple reason why there is no destructor and I did not even tried to think how and in which order everything must be cleared. It would be mostly just wasting of time and code space.

lsoltero, 2021-04-16:
Yes.. this is really not much of a problem since usually there is one object created.  In my case I created an NMEA2000 object using "new" for a second CAN bus interface, then copied it, and when done thought I would delete is a good house keeping measure.  Was very surprised to see that I could not delete an object I had just created with new.... Here is a very simple full program that shows the problem... I just compiled and tested to make sure that it wasn't anything silly on my end.

as you see there is nothing to it.  new followed by delete causes a core dump.  comment out the delete and the application exits without issue.

Anyway... no action needed here unless there is an underlying problem that could cause other issues down the line.

`
lsoltero@ubuntu16:~/openwrt.ee/local_packages/nmead/src$ cat test.cpp 

#include <N2kMessages.h>
#include <N2kDeviceList.h>
#include <NMEA2000_SocketCAN.h>

using namespace std;

int main()
{
    tNMEA2000_SocketCAN *t=new tNMEA2000_SocketCAN((char *)"can0");
    delete (t); // causes illigal instruction in N2kStream... so live with the memory leak
    
    if ( t == NULL ) return 1;
    return 0;
}

`

ttlappalainen, 2021-04-17:
I do not understand why do you need to create new and then delete it? If you have second interface, don't you need it over the program. Or even if you do not, just leave it. When your program closes, os will clean everything for that process.

lsoltero, 2021-04-17:
the board I am using has a number of different CAN interfaces all running different protocols (n2k, rv-c, j1939, ...) so the application opens each interface and in this case specifically looking for N2K data.  If it finds no N2K on the current interface it goes on to the next. in doing so it closes and deletes the NMEA2000 instance since it no longer needs it.  

But... it's not a big deal... the embedded system I am working with has plenty of RAM so I can leave the unused instances of NMEA2000 open and live with the small memory usage required for each. 

So no need to fix it and the issue is now documented for others.

Take care.

ttlappalainen, 2025-08-26:
Destructor has not been handled at all, since point has been that NMEA2000 object will be created once. Delete and recreate may leave holes to memory reservations, which is not good with small MCU:s. When I started library with Arduino environment, I found problems with several things, so I left e.g., destructor out. Since those days things had goen better and it could be handled properly, but there would be a some work. 

Anyway I have not found any need for delete object in any case and neither commercial users, with whom I co-operate.