ttlappalainen/NMEA2000 #215: Bilge alarm in N2K?

virtuvas, 2021-05-01:
hello Timo,

since I have three individual compartments in my boat and three float switches and pumps with three controls on the lower helm (that I'm often not using...) I'm thinking of "stealing" signal from when one of the pumps is on (they are 24V) and via a suitable voltage divider, get all three of them in different analogue pins of a nearby teensy.
The problem is what to do with that in order to show as pop-up warnings like high engine temp or low oil pressure in the garmin plotter or GMI10 on the f/b helm.
Since I don't own or drive a Ferrari (see #201) is there any other trick I can do?
I don't have a custom box up there, else I could simply use the switching state as you helped me work out on #210 broadcast state from the lower helm teensy, read them on the f/b and run a buzzer or something...
Is there any other cheap trick I could use?

thanks

Vassilis

ttlappalainen, 2021-05-01:
I do not believe that GMI10 would even support alert PGNs. But rather many MDF:s supports simple engine dynamic and alerts as you thought for engine failure bits. So Just test sending Engine dynamic PGN 127489 with some of the Engine Discrete Status bits on and look does GMI10 alert.

Or buy an Ferrari https://www.kobelt.com/products/product-detail/6700-vitals-vessel-system-monitoring-smart-alarm-device It is incredibly programmable for alarming and displays.

virtuvas, 2021-05-01:
GMI10 happily showing 127489 engine dynamic alarms, so that's fine. Using it now since last season.
and no, I'm not buying yet another screen, not much space to put them!
basically my question was if there is anything else other than 127489 that issues alarms that is supported by you!

virtuvas, 2021-05-01:
BTW, I could use the Maretron DSM250 I own, but it's on the lower helm and no use when you're helming from the f/b...

ttlappalainen, 2021-05-01:
SetN2kTransmissionParameters has similar flags as engine dynamic. I did not found alert PGNs even on DSM250 list.

It is also possible to fool with e.g. some temp message of tank level. Before I got Vitals, I used temperature for chain out on GMI20 - shows nicely temperature 0 - 50. GMI20 "alarms" at least for waste tank high level - it does not beep and either does no popup, which is small problem.

virtuvas, 2021-05-01:
ah, sorry, yes, haven't checked for alert pgns in DSM250, but it does have an Indicator Status page where you can configure colours for different binary status on different devices. Granted wont make any noise, and that's what I want.
Looks like I'll have to integrate a display screen on my yet to be built 5in touch screen for operating the stabilisers and the generator from the f/b. So when indicator status for any of them goes ON, I get a pop up on this custom screen...
maybe end of summer!

closed 2023-02-03 by virtuvas