ttlappalainen/NMEA2000 #216: ESP32 issues in ListenAndNode mode

titio72, 2021-05-03:
Hi, 

I'm using an ESP32 with a SN65HVD230 and it works perfectly to read from the bus, but when I try to send something the lib hangs (it stops reading and does not send anything).
Using an MCP2515 everything works fine sending and receiving.

Any idea on what could be wrong? You can find the code here https://github.com/titio72/n2k_evo_autopilot_remote

Thanks in advance for your help!

ttlappalainen, 2021-05-03:
No idea. I use ES32 a lot without problems. I have never used SN65HVD230. Check that you do not have it on standby mode. But even on that mode it should not stop library. Try to isolate problem and test some of the simple examples - e.g. TemperatureMonitor.

dirk99, 2021-05-03:
Hi, i have with some sn65hvd230 the same issues. I have two types of this CAN drivers. One works without any problems, but the other type(   https://www.googleadservices.com/pagead/aclk?sa=L&ai=DChcSEwjjz6bfsa3wAhUP43cKHedhDkAYABAFGgJlZg&ae=2&ohost=www.google.com&cid=CAESQOD2q33XNe--FXKveWBF9sEGZ49BITwqHlzJYKmHK1L-wgco8HRd30Axh0cL9Yn33gS2vgqcScLKm5CjMJEjEWM&sig=AOD64_1jqFABilnc21Niotwblfb_L7IV1g&ctype=5&q=&ved=2ahUKEwjK253fsa3wAhWw_7sIHVkKBfIQwg96BAgBEBk&adurl=.) do not send any packets on the same code.

This drive works fine
https://eckstein-shop.de/Waveshare-SN65HVD230-CAN-Board-33V-ESD-protection&ved=2ahUKEwjK253fsa3wAhWw_7sIHVkKBfIQFjARegQIGRAC&usg=AOvVaw3_CkEBU5O179gX1kVdgFN_

Greetings
Dirk

ttlappalainen, 2021-05-03:
They seem to use same chip. The one you say will not send seem to have fixed 120 ohm termination resistor. Check that you have exactly 2 termination resistors - one on each end of main bus. Measure also unpowered bus that resistor between CAN H/L is 60 +- 1 ohm. If is less, then you have somewhere extra resistor and some drivers can not drive the bus. If it is more, you are missing termination resistor.

titio72, 2021-05-03:
@dirk99 thanks for the hint

@ttlappalainen - yes, I've checked. Also tried to remove the smd resistor and use my termination. Also tried a different unit, just in case...
What is weird is that it receives perfectly, and when you try sending something it stops also receiving. This is why I was thinking to a software issue, but if dirk99 has seen the some problem maybe I just have to buy a different transceiver.
Any idea on how to debug it?

ttlappalainen, 2021-05-03:
As I said I use it a lot with ESP32. And also used in some commercial solutions so I would have heard of troubles. I also have tested without bus connection and it will not got stuck. Starts normally, when bus is connected.
Try with TemperatureMonitor.

titio72, 2021-05-03:
No luck... I've tried the original TemperatureMonitor code and got the same. Then I've tried using a low level CAN lib with the same result.
I'm now convinced that the problem is in the transceiver - I've ordered a couple of the same type/brand indicated by dirk99; in the meantime I can use the MCP to finish the code.

ttlappalainen, 2021-05-04:
Why people use those models? I have mentioned in several places to use MCP2562 or ISO1050. I have never had problems with those.

titio72, 2021-05-04:
@ttlappalainen thanks for the advice - I didn't really think too much about the transceivers, just bought some on Amazon.
I guess I'll try the MCP2562 too.

closed 2021-05-04 by titio72