ttlappalainen/NMEA2000 #217: Engine box monitoring

AntonisPapa, 2021-05-03:
Hi to everyone,

I am starting to develop an engine monitor box with an arduino due.
I would like to know if someone has an experience about the measurement of  rpm's?
I used an frequency counter ,when I made the first test to an electromagnetic sensor ,the deviation of the measurement was too high.
If someone have any idea ,please let me know???

ttlappalainen, 2021-05-05:
First, please skip DUE and go to Teensy 3.2 or ESP32.

I have used Teensy FreqMeasure library directly from altenator W+ - naturally requires rectification. In ESP32 it is also possible to use internal counters for measuring frequency. In sw your need just simple lowpass filtering:
```
double ReadPulseWidth=FreqMeasure.read();
PulseWidth= (FilterWeight * ReadPulseWidth+ (1 - FilterWeight) * PulseWidth);
```
FilterWeight =< 1 and depends how stable your data is. If there is lot variation, you need to use rather small FilterWeight, but then it also takes longer to reach right RPM, when you change it fast.

virtuvas, 2021-05-05:
+1 on what Timo says, using teensy 3.5 and v.happy.

re RPM have a look at the code of this guy:
https://www.youtube.com/watch?v=u2uJMJWsfsg&ab_channel=InterlinkKnight
maybe complicated but works absolutely fine on my main engines (IVECO straight 6 diesels) as well as the yanmar 2GMF generator 3000rpm one.

ttlappalainen, 2021-05-05:
Yes, Teensy 3.5 or 3.6 has also build in SD card, if you need to save anything. Or even Teensy 4.0/4.1 - I do not remember which one is cheaper.

virtuvas, 2021-05-05:
I've decided to stick with teensy 3.5 mainly due to the fact that analog pins are 5V tolerant. OK, will only measure 3V3, but wont destroy the chip with a 5V input.
Further I'm using the SD card to keep a log of the hours the engine has run (so if your VDO tacho LCD screen dies, you still know how many hours the engine run in the season :-) ) To avoid destroying the microSD quickly, I'm just writting to it every 120sec. Works fine now for 2yrs.

Mind, still I've destroyed 3teensies allready but not on engine monitoring, on a general monitor that does tanks and other analog and digital senders.
3.6 and the 4 series are 3V3 ONLY!