ttlappalainen/NMEA2000 #218: PGN059904 ISO Request

preeve9534, 2021-05-05:
I'm working on an N2k project that has to respond to these. I notice that the ISO Request PGN isn't in N2kMessages - I'm happy to add it and raise a pull request if you would like me to. Reason I'm asking is I was surprised not to find it there already and wondered if there is a reason - other than nobody needed it yet...

P

ttlappalainen, 2021-05-05:
Please do not. You should use tNMEA2000.SetISORqstHandler to register your own handler for your PGNs. Library responds automatically to internal requests. Library also responds NAK, if your handler returns false.

preeve9534, 2021-05-05:
Thank you for stopping me being a twit. I did search the package and
documentation for ISORequest, not expecting a pesky abbreviation.

ttlappalainen, 2021-05-05:
Documentation could be better. But it would be difficult to write short and still explaining everything so I'd rather respond questions.

preeve9534, 2021-05-05:
Understand that. Also not easy in a non native tongue. Thank you for your
support I'm getting to grips with it ... I think. Just about to test some
group function handler stuff .... Di I may have more questions...

P