ttlappalainen/NMEA2000 #219: Adding new Garmin Device in bus, locks teensy output?

virtuvas, 2021-05-05:
hi Timo,

must have done something silly again (not surprising...)
A teensy hooked up to a custom board (I have 4 of them around the boat doing different things) stops transmitting as soon as a GMI10 that was unconnected (or off) on the bus is turned on.
So 4GMIs, two on, two off, connect the teensyboard, the 2 connected GMIs are reporting data from the teensyboard.
As soon as I turn on the third GMI, the two connected stop reporting data from the teensyboard. Obvs the one I just turned on is not getting data either.
Now, if I disconnect the teensyboard, and now with 3 (or 4) GMIs on,  I reconnect the teensyboard, data flows to all 4 GMIs, no problems.

Some peculiar addressing issue?
Termination is fine, almost 20 devices on the bus, all working happily reporting whatever goes around the bus...

```
void setup() {
  //setup internal temperature reading NOT TO alter ADC settings!
  InternalTemperature.begin(TEMPERATURE_NO_ADC_SETTING_CHANGES);
  TeensyTemp_Filter.begin(SMOOTHED_AVERAGE, 70);

  //12bit 4095 for accuracy!
  analogReadResolution(12);
//  Serial.begin(115200);

  // NMEA2000 routines
  //OutputStream=&Serial;
  // Set Product information
  NMEA2000.SetN2kCANSendFrameBufSize(150);
    NMEA2000.SetProductInformation("1212131",                     // Manufacturer's Model serial code
                                  1210,                           // Manufacturer's product code
                                  "GVD ECU-PORT",                 // Manufacturer's Model ID
                                  SWVER,
                                  MANMODEL,                       // Manufacturer's Model version
                                  0x02,                           // load equivalency - use default
                                  0xffff,                         // NMEA 2000 version - use default
                                  0xff                            // Certification level - use default
                                 );
    NMEA2000.SetDeviceInformation(1212131,  // Unique number. Use e.g. Serial number.
                                 160,           
                                 50,            
                                 1210,      
                                 4                                
                                );  
    NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly, 131);

  NMEA2000.EnableForward(false); // Disable all msg forwarding to USB (=Serial)
  // Here we tell, which PGNs we transmit for each device
  NMEA2000.ExtendTransmitMessages(TransmitMessages);
  NMEA2000.ExtendFastPacketMessages(ExtendedFastPacketMessage);
  NMEA2000.SetMsgHandler(HandleNMEA2000Msg);
  NMEA2000.Open();

```

and on the loop I have:

```
#define NMEA2000ParamsUpdatePeriod 20000 //or 30sec?
#define TeensyInternalTempUpdatePeriod 1000 

static unsigned long NMEA2000ParamsNextUpdate = millis();
static unsigned long TeensyInternalTempNextUpdate = millis();
float TeensyTemp = 0.0;
int   boardID;

void loop() {

// var routines...
    EGTParams();
    NMEA2000.ParseMessages();
    
    EngineDynamicParams();
    NMEA2000.ParseMessages();

    EngineRapidParams();
    NMEA2000.ParseMessages();

  // internal Teensy Temperature as BaitWell with the board #
  tN2kMsg N2kMsg;
  if (TeensyInternalTempNextUpdate < millis() ) {
    TeensyInternalTempNextUpdate += TeensyInternalTempUpdatePeriod;

    TeensyTemp_Filter.add(InternalTemperature.readTemperatureC());
    TeensyTemp = TeensyTemp_Filter.get();

    //Serial.println("Internal Teensy temp: " + String(TeensyTemp));
    if (EngineInst == 0)
      boardID = 131;
      else if (EngineInst == 1)
      boardID = 136;
      else if (EngineInst == 2)
      boardID = 138;
      else if (EngineInst == 3)
      boardID = 139;     
    SetN2kTemperature(N2kMsg, 0xff, boardID, N2kts_BaitWellTemperature, CToKelvin(TeensyTemp)); 
    NMEA2000.SendMsg(N2kMsg);

  }

  if (NMEA2000ParamsNextUpdate < millis() ) {
    NMEA2000ParamsNextUpdate += NMEA2000ParamsUpdatePeriod;
    NMEA2000.SendIsoAddressClaim();
    NMEA2000.SendProductInformation();

  //??? needed or not here?
  NMEA2000.ParseMessages();
  }

}

```

Must be something I've done lately as it was definitely not a problem before Easter...
any ideas?

cheers

V.

ttlappalainen, 2021-05-06:
Can not say with this information. Some notes:
- Add some 1 sec periodic prints to loop to see does loop lock up. Simple debugging!
- What messagesyou handle with `NMEA2000.SetMsgHandler(HandleNMEA2000Msg);`?
- Do yo really need `NMEA2000.ExtendFastPacketMessages(ExtendedFastPacketMessage);`? Nearly all fast packets has been listed. Also proprietary messages will be handled right.
- As I understood you use Teensy ADC for temperature. It would be best to have ADC reading as own task and do it e.g. in every 1 ms or even faster. Then do low pass filtering for that. I use analogReadAveraging(1);  analogReadResolution(13); so that one single analogRead is fast and does not stop process for long time. Then I filter readings. 
-  if your "tasks" has been designed right, it is enough to call  NMEA2000.SendMsg(N2kMsg); once in loop. Tasks should be designed so that they do not spent long time.
- Timer test of style
  ```
if (NMEA2000ParamsNextUpdate < millis() ) {
    NMEA2000ParamsNextUpdate += NMEA2000ParamsUpdatePeriod;
```
    may cause you problems on some dat, if device is running long. millis() runs out on about 50 days. So lets expect you do  
    `NMEA2000ParamsNextUpdate += NMEA2000ParamsUpdatePeriod;`
    near end (FFFF D8F0=10000 before run out) , which causes your NMEA2000ParamsNextUpdate==20000
    Now on each loop time NMEA2000ParamsNextUpdate < millis() is true and will cause sending information and also cause
    NMEA2000ParamsNextUpdate += NMEA2000ParamsUpdatePeriod; until millis() finally runs out (10 sec). The your 
    NMEA2000ParamsNextUpdate has grown high and stops sending for long time. The better way to test it is
    `if ( millis()-NMEA2000ParamsNextUpdate < LONG_MAX ) {`
    At the end is class, which dows job for you. You just use it:
    ```
tmsScheduler NMEA2000ParamsNextUpdate(5000) ;
    ...
    if (NMEA2000ParamsNextUpdate.IsTime() ) {
      NMEA2000ParamsNextUpdate.Inc(NMEA2000ParamsUpdatePeriod);
      ...
```
    

FastTimers.h
=========
```
#ifndef _FAST_TIMERS_H_
#define _FAST_TIMERS_H_
/*
FastTimers.h

Copyright (c) 2015-2021 Timo Lappalainen, Kave Oy, www.kave.fi

Permission is hereby granted, free of charge, to any person obtaining a copy of
this software and associated documentation files (the "Software"), to deal in
the Software without restriction, including without limitation the rights to use,
copy, modify, merge, publish, distribute, sublicense, and/or sell copies of the
Software, and to permit persons to whom the Software is furnished to do so,
subject to the following conditions:

The above copyright notice and this permission notice shall be included in all
copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED,
INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS FOR A
PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT
HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF
CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE
OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.

In loop one should test when something should be done next time
instead of any delay. Timer.h automizes that, but with it it is difficult
to have nested tests. FastTimers just packets testing:
unsigned long NextRunTime=5000;
  if ( millis()-NextRunTime<LONG_MAX ) { ... }

Use it simply by defining e.g. 
void loop() {
  static tmsScheduler SchedulerDebugPrint;
 ...
  if ( SchedulerDebugPrint.IsTime() ) {
    SchedulerDebugPrint.Inc(2000);
    Serial.println("Here we go again");
  }
 ...
[... 124 lines cut ...]
  bool IsDisabled() const { return NextTime==FastTimersDisabled; }
  bool IsInfinitive() const { return NextTime==FastTimersInfinitive; }
};

//------------------------------------------------------------------------------
class tusTimer {
protected:
  unsigned long StartTime;
public:
  tusTimer() { Start(); }
  void Start() {
    StartTime=micros();
    if ( StartTime>=FastTimersReservedStart ) StartTime=FastTimersReservedStart-1;
  }
  void Disable() { StartTime=FastTimersDisabled; }
  unsigned long Elapsed() const { return (StartTime==FastTimersDisabled?0:micros()-StartTime); }
  double SecondsElapsed() const { return (double)(Elapsed())/1000000UL; }
  bool HasElapsed(unsigned long TestElapsedTime) const {
     return Elapsed()>TestElapsedTime ;
  }
};

#endif

```

virtuvas, 2021-05-06:
thanks Timo for the comments, and especially for the timing code, will integrate it in all my projects!
turns out I had on my custom board one damaged teensy which is only good for testing code (except analogue stuff where ADC is off). something messed there, changed it with a brand new one, works fine!

now to answer your points/questions:
this is the HandleNMEA2000Msg routine, definitely haven't written it myself, probably took it off you?
For the record, I commented it out on my setup, everything still works, so will probably remove it completely.

```
void HandleNMEA2000Msg(const tN2kMsg &N2kMsg) {
  int iHandler;
  
  // Find handler
  for (iHandler=0; NMEA2000Handlers[iHandler].PGN!=0 && !(N2kMsg.PGN==NMEA2000Handlers[iHandler].PGN); iHandler++);
  
  if (NMEA2000Handlers[iHandler].PGN!=0) {
    NMEA2000Handlers[iHandler].Handler(N2kMsg); 
  }
}
```
```
const unsigned long ExtendedFastPacketMessage[] PROGMEM={130823L,0};
```
is the PGN  for the maretron EGT measurement. That's were it's needed. Haven't tried removing it to see if it still works, will do.
```NMEA2000.ExtendFastPacketMessages(ExtendedFastPacketMessage); ```

Regarding ADC and teensy temperature, what I'd really like is to just record in influx/grafana the inside temp of each teensy and compare and check for any silly mistakes in code or wiring that would rise the temperature. So not anything highly scientific or accurate needed. Will see how it goes with this one and if it's OK I'll leave it as is.

Finally my tasks are designed OK, I've timed them and do what they should within the timeframes set. I don't have delays anywhere, OneWire temperatures DS18B20 are all in asynchronous mode, so should be OK.

if your "tasks" has been designed right, it is enough to call NMEA2000.SendMsg(N2kMsg); once in loop. Tasks should be designed so that they do not spent long time.

You see in my loop I have:
```
    EGTParams();
    NMEA2000.ParseMessages();
    
    EngineDynamicParams();
    NMEA2000.ParseMessages();

    EngineRapidParams();
    NMEA2000.ParseMessages();
```
EGTParams is a routine that sends 4 different PGNs (for now at the same time intervals-will correct as needed):

```
      SetN2kTemperature(N2kMsg, 0xff, EngineInst, TempSource, CToKelvin(0.1*thermoTempC), N2kDoubleNA);
      NMEA2000.SendMsg(N2kMsg);
      
      SetN2kTemperatureExt(N2kMsg, 0xff, EngineInst, TempSource, CToKelvin(thermoTempC), N2kDoubleNA);
      NMEA2000.SendMsg(N2kMsg);

      SetN2kMaretronTempHR(N2kMsg, 0xff, EngineInst, TempSource, CToKelvin(thermoTempC), N2kDoubleNA);
      NMEA2000.SendMsg(N2kMsg);
      
        SetN2kTemperature(N2kMsg, 0xff, 138, N2kts_EngineRoomTemperature, CToKelvin(thermoTempEng));
        NMEA2000.SendMsg(N2kMsg);
```

if I remove the three NMEA2000.SendMsg(N2kMsg); lines and leave only the last, it will not update the  bus with the the first three PGNs only the last, code is really minimal, nothing fancy there!

Further and along the same lines, in my loop and under each subroutine I'm calling, is the ```NMEA2000.ParseMessages()``` of any use or shall I remove them? What is actually the role of this command?

cheers

V.

ttlappalainen, 2021-05-07:
Always try to make code as clean as possible.

If you do not handle any recived messages, you do not need function void HandleNMEA2000Msg(const tN2kMsg &N2kMsg), variable NMEA2000Handlers and call for NMEA2000.SetMsgHandler(HandleNMEA2000Msg); at all. Just remove them from code. You need them, if you would listen any message on the bus and do something with it. E.g. if you would make small simple display showing your tank capacisity, your would handle 127505, parse it and draw to display.

NMEA2000.ExtendFastPacketMessages(ExtendedFastPacketMessage); Is needed only for handling received fast packet messages right. Orignally I had only limited list of fast packet messages listed internally to handle them right. Later I have extended the list to cover all (I think) fast packet messages, so this is not needed and it either does not have any effect.

NMEA2000.SendMsg(N2kMsg); does the sending for build message, That you have to call after you have build each message as you have now.

NMEA2000.ParseMessages(); you can think as task. If you look Windows task list, it has a long list of tasks. In a deep level in OS, they work like "tasks" in loop. The main difference is that OS runs task1 while, then it saves its state (stack etc.) and goes to run task2. It first restores its state and runs it a while, saves its state and jumps to task3 etc. Teensy does not have multitasking so you have to write it by yourself. This is what I have in main loop - just list of tasks what is running and NMEA2000.ParseMessages(); is one of them. The difference for multitasking OS is that you have to take care yourself that task does not block for long time and it saves its state on exit. This is also what you are now doing on reading 1-wire. One good example for single task is updating display. There is lot of nice libraries, which unfortunately blocks loop time to time for too long time. In my own display task I have splitted it so that it does not draw everything in one call and finally call to UpdatDisplay(); task normally uses about 1 ms (on changing value on display - few us, if there is no change) and never more than 50 ms.

Multitasking is nice, but may become complex. If all your tasks does their own things, there is now problem. But if tasks need to share resources or data, you need to use locking and newbie will most likely cause deadlock for system. If you do not use locking, you will get unwanted results, since data may be changed partially when switching to other task happens. So in this point of view Arduinos simple loop is good for beginners.

So ParseMessages(); is task, which internally handles received messages and decides what to do with them. If some device e.g. requests your device product information, message comes to buffer, where ParseMessages reads and parse it. Then it noticed that other device has requested 126996. It then simply builds 126996 from information you have given in setup with SetProductInformation and sends it. If the request is not for any internal message, task calls ISORqstHandler, if you have registered it. So internally task builds received messages to N2kMsg and either handles them or call related registered call back - finally call back registered with SetMsgHandler. ParseMessages handles several NMEA2000 required things for you and you do not need to understand all its requirements. If your call for ParseMessages(); is delayed too much, internal handling will be delayed and e.g. important respond to requests like 60928 and 126996 will be too late and MFD:s will drop your device off. This is what happened with your first 1-wire task, which caused over 700 ms delay.

Then some improvements. With library you can write devices, which would pass NMEA2000 certification, but it can not do all automatically. Your device works fine as it is, but some small things could be added to make it certification ready:
- Respond to ISO request you have registered for TransmitMessages.
This is most important, since some device may request this data from you.
- Save and restore device source possibly changed by address claiming.
When address claiming happend, your device gets new source address, which you should use on next start of your device.
- Save and restore changed  device instances.
If you have conflicts of instances on the bus, some MFS:s provides way to change device instance. You should save that and restore it on device start.
- Save and restore changed installation descriptions.
This is important in complex systems and for professional installers. There are two  installation description fields, which can be changed by some NMEA2000 device/system. E.g. Actisence can do it and I am adding it to Simulator. With description you can e.g. tell where device is located.

Respond to ISO request you have registered for TransmitMessages.
For this you simply write routine 
```
bool ISORqstHandler(unsigned long RequestedPGN, unsigned char Requester, int DeviceIndex) {
  switch( RequestedPGN ) {
    case 130312UL: SendTemperatures(); return true;
    case <something else my device sends>: SendSomethingElse();  return true;
  }

 return false; // =I do not provide this PGN
};

```
then you register your handler with `NMEA2000.SetISORqstHandler(ISORqstHandler);`

Save and restore device source possibly changed by address claiming.
Add to your loop task e.g. CheckChangedNMEA2000Settings(); and write function:
```
void CheckChangedNMEA2000Settings() {
  if ( NMEA2000.ReadResetAddressChanged() ) {
[... cut ...]
  }
}
```
And on setup
```
void setup() {
...
  char buf[200];
  ReadInstallationDescription1(buf,200);
  NMEA2000.SetInstallationDescription1(buf);
  ReadInstallationDescription2(buf,200);
  NMEA2000.SetInstallationDescription2(buf);
...
}

```

Long story - I should copy this to manual. If you have any notes about text how you would understand it better, I would be pleased. You are not beginner with N2k, but not expert and manual should have instructions for more advanced and product developers.

virtuvas, 2021-05-07:
thanks for the detailed comments and suggestions Timo, only have time to respond to the first (simple!) ones, will read and digest the rest in the evening/tomorrow and come back.

**HandleNMEA2000Msg(const tN2kMsg &N2kMsg);**  I'm reading two custom PGNs (coming from the generator black box) as well as enginedynamic and 126208 (for this I have to start a new issue during the w/e). So I AM parsing data from 4 pgns, and acting upon receiving - could be setting variables, could be activating parts of other code doing something else, or simply used to process some data and be sent to the bus as part of another PGN. So looks like I have to keep it, right?

**NMEA2000.ExtendFastPacketMessages(ExtendedFastPacketMessage);** where is this list of messages you maintain? Is the Maretron PGN (PGN130823) you helped me setup in there? if so, fine I'll remove that bit of code. 
IIRC the other two Maretron PGNs (PGN65286 & PGN65287) are not fastpacket so not listed here anyway.

**NMEA2000.SendMsg(N2kMsg);**  OK, I got that, it's rather simple After every SetN2k* line you NEED  a SendMsg line!

the rest need a lot of studying, will come back!

cheers

V.

ttlappalainen, 2021-05-07:
For fast packets resolving there are different calls. Messages are listed in NMEA2000.cpp, see IsFastPacketSystemMessage, IsMandatoryFastPacketMessage and IsDefaultFastPacketMessage. But there is also IsProprietaryFastPacketMessage, which should cover all proprietary range fast packet messages anyway. The one stupid thing in NMEA2000 is that there is no defined range or any way to know is message fast packet or not. They just need to be listed. Luckily proprietary messages ha defined range.

virtuvas, 2021-05-08:
Ok, slowly working my way to implementing the rest of the things you mentioned. 
Found a number of errors/forgotten to register things in   NMEA2000.ExtendTransmitMessages(TransmitMessages);
 (but surprisingly still worked!)

Regarding **ISORqstHandler** I have a slight complication in that some Temp related PGNs are sent from different subroutines within my code (code is close to 200kb, so I've split it up in IDE in a dozen subs). I guess the listing below is OK? It compiles but haven't tested on board yet will do later on tonight:

```
bool ISORqstHandler(unsigned long RequestedPGN, unsigned char Requester, int DeviceIndex) {
  switch( RequestedPGN ) {
    case  65525L: EngineStartStop(); return true;
    case 127488L: EngineRapidParams(); return true;
    case 127489L: EngineDynamicParams(); return true;
    case 127493L: TransmissionParams(); return true;
//    case 127497L: TripParams(); return true;      //fuel consumption tests, not ready yet.
    case 127501L: Switching(); return true;
    case 127505L: BlackWaterFluidLevel(); FreshWaterFluidLevel(); DieselFluidLevel(); return true;
    case 128777L: AutoAnchor(); return true;        
    case 130306L: NASAWind(); return true;  
    case 130311L: NASAWind(); Environmental(); return true;      
    case 130312L: EGTParams(); Environmental(); loop(); return true;        
    case 130314L: EngineDynamicParams(); return true;        
    case 130316L: EGTParams(); return true;        
    case 130576L: SmallCraftParams(); return true;        
    case 130823L: EGTParams(); return true;        

  }
  return false; // =I do not provide this PGN
};
```
I know, I'll get rid of 130311, but wont change much as it will move to 312 or 314 (got to check what my GMI and Maretron supports...

timing routine implemented, works fine, have to wait for 50days to see the results :-D 

cheers

V.

ttlappalainen, 2021-05-10:
Why you call EngineDynamicParams 130314L? That is just pressure.
Do not call  loop(); for 130312L. You are already on loops NMEA2000.ParseMessages callback and now you call again loop, which contains NMEA2000.ParseMessages So you may generate some kind of infinite loop or stack overflow.

I use normally different routines collecting data and building messages to be sent. So they are like independent tasks. Normally it is also good to make small modules instead of some huge modules. I rarely have more than 500 lines per module. Some complex are up to 1000 lines.

Have you skipped Arduino IDE and moved to e.g. Visual Studio Code and PlatformIO.

ttlappalainen, 2021-05-12:
I prefere to split long code to small logical modules. Arduino IDE is good for beginners, but as far as you start to do more work, it would be better to change to something else. The one biggest difference is that in Ardun IDE you can have e.g. function without defining it first in other .ino or below the function call. Normally in c/c++ you have to always define funtions/variables before you can call use them. That is what header files are for. In header (.h) file there are all definitions, what you can use from module. Actually you can use nearly anything inside module, which has not been define as static.

It is OK to call several functions for single PGN request. When you get request to e.g. 130312, you should respond with all instances you normally send. But you should not send any other messages. Anyway I againg prefer to have separate functions for reading data and sending them. Building message to be sent from simple values is fast process, so call back will be shor. But if you start to read e.g. 1-wire temperature on call back it may take time. It will most probably work, it it works on loop, but it is not good practise.

virtuvas, 2021-05-12:
> So ParseMessages(); is task, which internally handles received messages and decides what to do with them. If some device e.g. requests your device product information, message comes to buffer, where ParseMessages reads and parse it. Then it noticed that other device has requested 126996. It then simply builds 126996 from information you have given in setup with SetProductInformation and sends it. If the request is not for any internal message, task calls ISORqstHandler, if you have registered it. So internally task builds received messages to N2kMsg and either handles them or call related registered call back - finally call back registered with SetMsgHandler. ParseMessages handles several NMEA2000 required things for you and you do not need to understand all its requirements. If your call for ParseMessages(); is delayed too much, internal handling will be delayed and e.g. important respond to requests like 60928 and 126996 will be too late and MFD:s will drop your device off. This is what happened with your first 1-wire task, which caused over 700 ms delay.
> 
> Then some improvements. With library you can write devices, which would pass NMEA2000 certification, but it can not do all automatically. Your device works fine as it is, but some small things could be added to make it certification ready:
> 
> * Respond to ISO request you have registered for TransmitMessages.
>   This is most important, since some device may request this data from you.
> * Save and restore device source possibly changed by address claiming.
>   When address claiming happend, your device gets new source address, which you should use on next start of your device.
> * Save and restore changed  device instances.
>   If you have conflicts of instances on the bus, some MFS:s provides way to change device instance. You should save that and restore it on device start.
> * Save and restore changed installation descriptions.
>   This is important in complex systems and for professional installers. There are two  installation description fields, which can be changed by some NMEA2000 device/system. E.g. Actisence can do it and I am adding it to Simulator. With description you can e.g. tell where device is located.
>
> Long story - I should copy this to manual. If you have any notes about text how you would understand it better, I would be pleased. You are not beginner with N2k, but not expert and manual should have instructions for more advanced and product developers.

Timo, have been thinking, editing and trying to understand your comments (it's not v.easy at points)

I can understand that it's impossible to have a fully documented text explaining how things work to great detail. OTOH, having to reply to everyone's Qs is not very productive either. At least I hope most ppl have a good look at the open and mainly closed issues for replies to their problems, I do!

Now, you do have simple examples which I've tried initially, fine, give you an idea and help you v.quickly to do something and test if things work in terms of h/w, libs, wiring, etc.

What I think is missing is ONE LARGE COMPLEX example which integrates all the bits that you've posted above.
Say, a hypothetical project where you read:

* an analogue pin for service/starter voltage
* a  VDO sender for oil pressure
* a NMEA0183 serial line for depth or wind
* a One-Wire for temperature
* at the same time read one PGN from the bus

and get all these pumped in the bus in 3-4 PGNs

this way you handle a bit of everything N0183 &N2K input, analogue, digital and export PGNs

If you do so with some explanation on the various bits of code (two three lines comments before each chunk of code would be enough) then someone could pick that up and know what s/he is doing and not end up where I am that I'm copying code without (at points) really knowing why I'm doing it. OK, you said so, I believe you :-) but I cannot debug something like that, so I have to come back to you...

What do you think?
OK, maybe the N0183 reading is a bit too advanced but I think it's intriguing and something many of us have in their wishlist when slowly moving away from N0183 without the funds to throw away everything in one go (or like in my case decide to buy a NASA 0183 wind instrument because it's 200euro cheaper than the cheapest N2K one and with an 232 chip get it the N2K bus...)

cheers

V.

ttlappalainen, 2021-05-12:
I have planned to make some simple example like TemperatureMonitor as ExtentedTemperatureMonitor with things containing NMEA 2000 requirements. Unfortunately time is limited and for hobbyist those are not so critical, since device will work. Also those who has used library for commercial product, they have been more experienced about NMEA2000 and already knew what to search. Also I wrote those for you, but then I also have part, which I may add to document and use with extended example. Also by answering for different questions I may get better vision, what I should add to document. Maybe some day...

I do not think one large example is good idea. It is easier to get into system with small parts.

virtuvas, 2021-05-12:
at least a COMPLETE example with all the bits that you have recommended to me over the last posts would help for sure!
would probably also help me understand what I did and now after adding

```NMEA2000.SetISORqstHandler(ISORqstHandler);``` 
and it's routine as suggested,

```NMEA2000.SetMsgHandler(HandleNMEA2000Msg);```
causes the device to not appear on the bus and stops it from broadcasting after power cycle of the system!

ttlappalainen, 2021-05-13:
ISORqstHandler will be called, when some device on bus requests your device to send some PGN. As I mentioned, request respond is NMEA 2000 requirement.

HandleNMEA2000Msg will be called to all other messages coming from bus so that you can handle them, if it is your interest.

All callbacks as above (there are also others, which I have not mentioned) will be called when you call NMEA2000.ParseMessages(); So it is important that you do not call infinite loop. That is normally easiest to manage, if all your "tasks" does just their own thing. Bad example:
```
void HandleTemp() {
   ReadTemp();
   NMEA2000.ParseMessages(); 
   if ( TempScheduler.IsTime() ) {
     TempScheduler.Inc(TempPeriod);
     tN2kMsg N2kMsg;
     SetN2kTemperature(...);
     NMEA2000.SendMsg(...);
...
}

void ISORqstHandler(...) {
... case 130312: HandleTemp(); break;
}

void loop() {
  ...
  NMEA2000.ParseMessages();
}
 
``` 
Now on above sample we call NMEA2000.ParseMessages(); just for sure so that delays as small enough. Now on loop call to NMEA2000.ParseMessages(); we may do call back to ISORqstHandler on request, which then may call HandleTemp(); and again NMEA2000.ParseMessages(); Also that would not actually respond to request, since TempScheduler.IsTime is not necerrarily true. This does not yet necessary cause infinite loop, but is risk for troubles with stack etc. If all things are splitted to own tasks, it is easier to handle.
```
void SendTemp() {
   tN2kMsg N2kMsg;
   SetN2kTemperature(...);
   NMEA2000.SendMsg(...);
}

void ReadTemp() {
  // Just read temp to some variable.
}

void ReadInputs() {
...
   if ( TempReadScheduler.IsTime() ) {
     TempReadScheduler.Inc(TempReadPeriod);
     ReadTemp();
   }
...
}

void SendMessages() {
...
   if ( TempScheduler.IsTime() ) {
     TempScheduler.Inc(TempPeriod);
     SendTemp();
   }
   if ( EngineDynamicScheduler.IsTime() ) {
     EngineDynamicScheduler.Inc(TempPeriod);
     SendEngineDynamic();
   }
...
}

void ISORqstHandler(...) {
... case 130312: SendTemp(); break;
}

void loop() {
  ...
  NMEA2000.ParseMessages();
  SendMessages();
  ReadInputs();
  ...
}

```
On above "tasks" has been splitted to small atomic functions, which can be called alone. Also if device/OS would have real multitasking, You could just start those three calls in loops as own tasks. Naturally there should be some locking for thread safety, but that is an other story.

In your case I expect that some of your function in ISORqstHandler callback may do something causing finally stack overflow or infinite loop. If you instead just do simple message build from variables and send it, it is clean. In principle it is nice to have e.g. ReadTemp routine, which would update temperature from input (or 1-wire) and return current temp and you could use in building message:
SetN2kTemperature(...ReadTemp()...);
But if callbacks will be used, you have to keep that in mind.

If you simply comment your `NMEA2000.SetISORqstHandler(ISORqstHandler);` , your code may work again. Then you can start to split code.