ttlappalainen/NMEA2000 #220: This is probably not a an issue just lack of competence

h2omatt, 2021-05-30:
I am trying to learn about CAN Bus and the NMEA 2000 standard. I have 2 Teensy 4.1 with two SN65HVD230 Transceivers. I am trying to use the Kave Oy simulator to send data to one of the Teensy4.1 then listen on the other teensy4.1 using the TeensyActisenseListenerSender code below. Any tips for Debugging would be helpful.

#include <NMEA2000.h>
#include <N2kMessages.h>
#include <NMEA2000_Teensyx.h>

tNMEA2000_Teensyx NMEA2000;

void setup() {
  Serial.begin(115200); // Data from PC
  Serial1.begin(115200); // Data to PC
  NMEA2000.SetForwardStream(&Serial1); 
  NMEA2000.SetMode(tNMEA2000::N2km_ListenAndSend);
  //NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show in clear text
  NMEA2000.Open();

  // I had problem to use same Serial stream for reading and sending.
  // It worked for a while, but then stopped.
  //ActisenseReader.SetReadStream(&Serial);
  //ActisenseReader.SetMsgHandler(HandleStreamN2kMsg); 
}

void HandleStreamN2kMsg(const tN2kMsg &N2kMsg) {
  N2kMsg.Print(&Serial);
  NMEA2000.SendMsg(N2kMsg,-1);
}

void loop() {
  NMEA2000.ParseMessages();
  //ActisenseReader.ParseMessages();
}

h2omatt, 2021-05-31:
Also i do have some  MCP2551 on order as they seem to be preferred transceivers.

ttlappalainen, 2021-05-31:
Just use example ActisenseListenerSender. It works with Teensy 4. TeensyActisenseListenerSender was for special case, where you have an different port for data input and output. In your code you have commented out data parsing for USB side.

I prefer to use MCP2562 or ISO1050, when isolated connection should be used. MCP2551 is 5V chip and Teensy 4 is not 5V tolerant.

h2omatt, 2021-05-31:
Thanks for the fast response! Ahh good catch I just ordered MCP2551 to have them on hand. I have some MCP2562 on order now too!! 

I did the your temperature monitor example using the NMEA Reader. unfortunately I don't have a real NMEA devices to play with in front of me so I'm trying to use the NMEA simulator and a small bread board CANBUS. 

As of right now I have ActisenseListenerSender flashed on both Teensy and simulator running on one and the NMEA Reader on the second but with no data shown on the NMEA Reader. still using the SN65HVD230 as shown in the image. 

[image]

// Demo: NMEA2000 library. Bus listener and sender. 
//   Sends all bus data to serial in Actisense format.
//   Send all data received from serial in Actisense format to the N2kBus.
//   Use this e.g. with NMEA Simulator (see. http://www.kave.fi/Apps/index.html) to send simulated data to the bus.
//   I have plans to add receiving functionality and data forwarding to NMEA Simulator. Meanwhile you can define
//   other stream to different port so that you can send data with NMEA Simulator and listen it on other port with 
//   Actisense NMEA Reader.

#define N2k_CAN_INT_PIN 21
#include <Arduino.h>
#include <N2kMsg.h>
#include <NMEA2000.h>
#include <NMEA2000_CAN.h>
#include <ActisenseReader.h>

tActisenseReader ActisenseReader;

// Define READ_STREAM to port, where you write data from PC e.g. with NMEA Simulator.
#define READ_STREAM Serial       
// Define ForwardStream to port, what you listen on PC side. On Arduino Due you can use e.g. SerialUSB
#define FORWARD_STREAM Serial    

Stream *ReadStream=&READ_STREAM;
Stream *ForwardStream=&FORWARD_STREAM;

void setup() {
  // Define buffers big enough
  NMEA2000.SetN2kCANSendFrameBufSize(150);
  NMEA2000.SetN2kCANReceiveFrameBufSize(150);
  
  if (ReadStream!=ForwardStream) READ_STREAM.begin(115200);
  FORWARD_STREAM.begin(115200);
  NMEA2000.SetForwardStream(ForwardStream); 
  NMEA2000.SetMode(tNMEA2000::N2km_ListenAndSend);
  //NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show bus data in clear text
  if (ReadStream==ForwardStream) NMEA2000.SetForwardOwnMessages(false); // If streams are same, do not echo own messages.
  //NMEA2000.EnableForward(false);
  NMEA2000.Open();

  // I originally had problem to use same Serial stream for reading and sending.
  // It worked for a while, but then stopped. Later it started to work.
  ActisenseReader.SetReadStream(ReadStream);
  ActisenseReader.SetDefaultSource(75);
  ActisenseReader.SetMsgHandler(HandleStreamN2kMsg); 
}

void HandleStreamN2kMsg(const tN2kMsg &N2kMsg) {
 // N2kMsg.Print(&Serial);
  NMEA2000.SendMsg(N2kMsg,-1);
}

void loop() {
  NMEA2000.ParseMessages();
  ActisenseReader.ParseMessages();
}

ttlappalainen, 2021-05-31:
What is R2 for on your module? They may be termination resistors. So do you have totally 4 resistors now?

h2omatt, 2021-05-31:
Yeah the look like terminating resistors to me... I removed my external wire and wired them directly together but still no luck. going to try a few more things but I do have some MCP2562 on the way to test too!

Thanks again for your support.

h2omatt, 2021-05-31:
I tried your MessageSender.INO file and all I get is PNG send failure. do you think that is do to the transceivers module I choose? if so hopefully its fixed when I get the new ones in. 

[image]

ttlappalainen, 2021-06-01:
That means that CAN controller does not communicate with bus. So -
- CAN tranceiver has been connected wrong
- CAN bus is blocking communication   --  no termination resistors, too many or broken device on bus.
- CAN tranceiver is dead

h2omatt, 2021-06-03:
I got some MCP2562 transceiver in but still no luck. I updated my crappy schematic for reference. Is there an issue with the fact I'm running a teensy 4.1?

https://www.digikey.com/en/products/detail/microchip-technology/MCP2562-E-P/4079966?utm_adgroup=Integrated%20Circuits&utm_source=google&utm_medium=cpc&utm_campaign=Dynamic%20Search_EN_RLSA_Buyers&utm_term=&utm_content=Integrated%20Circuits&gclid=Cj0KCQjw2NyFBhDoARIsAMtHtZ5pKNwbT1OR70DtOW2FS2I3mjRXc96d4iaSxv44297vIPij7h8ZLgQaAgpwEALw_wcB

[image]

ttlappalainen, 2021-06-03:
I have tested with 4.0, but it should be same.

On MCP2562 Vio must be connected to 3.3 V and  STBY to GND

h2omatt, 2021-06-03:
I did supply the Vio with 3.3v but did not ground the STBY. testing now to see if I can get something working. what do you think is a good first example test my benchtop network with?

bigalnz, 2021-09-18:
@h2omatt did you ever get this to go? I have same hardware.

h2omatt, 2021-09-18:
@bigalnz no I haven't. unfortunately had to put it aside for a little bit. I really want to learn more and how it works. my personal email is matt.colvin26@gmail.com I would love to connect and see what you have working. maybe google hangout if your available.

ttlappalainen, 2021-09-19:
The connection to MCP2562 on picture is wrong. You must connect:
- Vdd to 5 V
- Vio to 3.3 V
- STBY to GND

bigalnz, 2021-09-19:
I am using the board matt showed in his first picture:
[image]

I am expecting however to have to connect the GND (Black?) wire from the NMEA2000 cable to GND on the Teensy4.1?

ttlappalainen, 2021-09-19:
Read my document https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/Connecting_hardware_to_NMEA2000.pdf - on chapter 7 there are most common hardware errors.

Note that you have r1 and r2 on that board. Do you know meaning of them? If they are terminal resistors, you may have worng termination on your bus.

bigalnz, 2021-09-19:
Hi Timo. I haven't wired anything up yet. I want to make sure ground wire goes to the correct place. Chapter 7 doesn't tell me where the ground wire goes.

bigalnz, 2021-09-19:
The resistors measure 9.8K Ohm and 120 Ohm.

ttlappalainen, 2021-09-19:
As in document: False or unconnected termination resistors. Measure that unpowered NMEA2000 bus has 60 ohm resistance between CAN-h and CAN-L.

bigalnz, 2021-09-19:
It has 120Ohm between high and low.

ttlappalainen, 2021-09-19:
You must measure from fully installed bus and get 60 ohm:
- False or unconnected termination resistors. Measure that unpowered NMEA2000 bus has 60 ohm resistance between CAN-h and CAN-L.

bigalnz, 2021-09-19:
Ok. So connect my Teensy via the NMEA2000 cable to the unpowered bus, and them measure resistance between CAN-H and L on the SN65HVD230 board?

ttlappalainen, 2021-09-19:
I prefer to study electrical schemas and NMEA2000 bus. Now you are asking wire by wire how should make connections. You could also check other connection documents under library documents. There is also one drawing with SN65HVD230.

h2omatt, 2021-09-19:
> The connection to MCP2562 on picture is wrong. You must connect:
> 
> * Vdd to 5 V
> * Vio to 3.3 V
> * STBY to GND

@ttlappalainen hmmmm I am surprised I might have wired the MCP2562 wrong. I am setting up a test today.

h2omatt, 2021-09-19:
WOW.... Vdd=5v.... that was my problem the whole time... and that was should of been the easy part. @ttlappalainen thank you for you patience. I am looking forward playing with this on my boat.

[image]