ttlappalainen/NMEA2000 #221: Arduino DUE with MCP 2515 receive issue

Gcaravell0, 2021-05-31:
Hi to everybody and tanks a lot for this magnific library!!
I'm replaicing my iTC-5 device with an arduino DUE. With this opportunity I also want to export some data to my PC via serial interface. I performed different communication tests with two arduino, as shown below.
[image]
[image]
[image]

Observing the oscilloscope I'm able to see that there is a message in the bus. I am also sure that the transimitting node is working because when I disconnect one of the two bus I see a message sending error...
However, trying to use the examples "Datadisplay 2" or "Actisense Listener" I cannot see any message on the listener node. I'm also not able to see any message if I use the MCP CAN receive example. :(
Obviously I tryed to invert the two arduino devices but nothing changes.
I attached the following preamble to the code:
_

#include <Arduino.h>
#define N2k_SPI_CS_PIN 53    // Pin for SPI select for mcp_can
#define N2k_CAN_INT_PIN 2   // Interrupt pin for mcp_can
#define USE_MCP_CAN_CLOCK_SET 8  // Uncomment this, if your mcp_can shield has 8MHz 
#define USE_N2K_CAN 1 // Force mcp_can
#include <NMEA2000_CAN.h>  // This will automatically choose right CAN library and create suitable NMEA2000 object
#include <SPI.h>
#include <mcp_can.h>
#include <NMEA2000_mcp.h>
#include <N2kMessagesEnumToStr.h>
#include <N2kMaretron.h>
#include <N2kMessages.h>
#include <math.h>   // these are necessary for my code
#include <DueTimer.h>

_
Someone could tell me what the mistake could be?
Thank you in advance for your time and consideration

Giuseppe

ttlappalainen, 2021-05-31:
At least you have wired int pin to 21 but define it to 2.

Do not forget terminaltion resistors.

Why you do not use DUEs internal CAN?

DUE is big and old board. Teensy 3.2 and up would be better.

BerndCirotzki, 2021-06-18:
Hello Gcaravell0,
maybe I can help, ... I had the same Problem. I think it is nor an issure from the NMEA2000 library.
The issue is in the Library "CAN_BUS_Shield-master". I thing you are using it. Issure withe the mcp2515 library.
Try to add "#define DEBUG_RXANY 1" in the File "mcp_can_dfs.h" then it works. I don't know why, but it was so !

Greeting
Bernd

BerndCirotzki, 2021-06-18:
... and ... I forgot, ... do not use the MCP2515 Interrups !
Do NOT define this : " #define N2k_CAN_INT_PIN"

ttlappalainen, 2021-06-18:
Why not use MCP2515 interrupts? Library is very unefficient without interrupts. You should use #define N2k_CAN_INT_PIN xx, where xx is pin where interrupt line has been wired. You should only leave interrupt undefined, when you test and can not get it running. When it works without interrupt, you should define interrupt pin and check that it works with interrupt. If not, then there is problem with wireing or pin definition.

BerndCirotzki, 2021-06-18:
... I have had some Problems, so I disabled it and have not enabled since long time. I think it works with Interrupts.
In my Config, Im using a lot Interrupts at same time. ... getting Wind speed data, sending Seatalk(1) Messages etc ...
In this case, very mutch thanks for your "super" Librarys !!

BerndCirotzki, 2021-06-20:
So, I tested it again. when setting  #define N2k_CAN_INT_PIN xx the Ardurino Mega2560 makes a reboot every 3 to 5 Seconds.
when not defining it, it works without Problems. I don't konw why. that was the reason I swiched it off.

ttlappalainen, 2021-06-21:
Did you use mcp_can from my git.

BerndCirotzki, 2021-06-21:
Yes, I am using your Git library mcp_can.

 

And tested with 

#define DEBUG_RXANY 1  set ,

and with

//#define DEBUG_RXANY 1  not set

 

. with Interrups activ . reboots, without Interrupts activ everything works.

 

My used librarys:

NMEA2000 in Version 4.13.1

NMEA0183-master in Version 1.7.1

EEPROM in Version 2.0

SPI in Version 1.0

CAN_BUS_Shield-master    (from your Git)

NMEA2000_mcp-master in Version 1.1.2

 

PinChangeInterrupt-1.2.8 in Version 1.2.8  for this :

pinMode(26, INPUT_PULLUP); // Digitalpin 13

attachPinChangeInterrupt(7, ReceiveWindSpeedImpuls, RISING);

 

And Timer 5 (OCR5A, and ICP5) for sending and receiving Seatalk with ICP and
timer Interrupts

 

. My NMEA2000 opening :

 

#define USE_MCP_CAN_CLOCK_SET 8 

.

NMEA2000.SetForwardSystemMessages(false);

NMEA2000.SetMode(tNMEA2000::N2km_NodeOnly,25);

NMEA2000.EnableForward(false);

NMEA2000.Open();

issue (#221)

 

<https://github.com/ttlappalainen/NMEA2000/issues/221#issuecomment-864793046
3R2RANCNFSM452TLVZA> .
<https://github.com/notifications/beacon/AHU3PHGDWT4EZZ6QRGYQ2VDTT3R2RA5CNFS
M452TLVZKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGOF3DVQ
.gif>

ttlappalainen, 2021-06-21:
Arduino Mega has rather small RAM and it may run out of it. Have you tested some simple example with interrupt - e.g. Temperature monitor or DataDisplay2. I have tested those with Mega.

BerndCirotzki, 2021-06-21:
… memory is not the Problem. But I know what you mean.

 

I am saving it like this :

 

static const tNMEA0183Handler NMEA0183Handlers[] PROGMEM ….

 

for (iHandler=0; (char*)pgm_read_word(&NMEA0183Handlers[iHandler].Code) != 0 && !NMEA0183Msg.IsMessageCode((char*)pgm_read_word(&NMEA0183Handlers[iHandler].Code)); iHandler++);

  if ((char*)pgm_read_word(&NMEA0183Handlers[iHandler].Code)!=0) {

     Handler = pgm_read_ptr(&NMEA0183Handlers[iHandler].Handler);

…

And 

Instead of 

This :   Serial.print(„Balbal“); …

I do it like this :

char Textbuffer[xx];

EEPROM.get(880,TextBuffer);

Serial.print(TextBuffer);

 

So Sketch compile looks good: ( German output)

 

Der Sketch verwendet 190556 Bytes (75%) des Programmspeicherplatzes. Das Maximum sind 253952 Bytes.

Globale Variablen verwenden 2281 Bytes (27%) des dynamischen Speichers, 5911 Bytes für lokale Variablen verbleiben. Das Maximum sind 8192 Bytes.

BerndCirotzki, 2021-06-21:
You are so right !! 

Memory !  I took a look at your Code and saw  your „Memory- eater“ 

 … N2kCANMsgBuf = new tN2kCANMsg[MaxN2kCANMsgs];

 

I add 

 

NMEA2000.SetN2kCANMsgBufSize(2); 

 

And now it works with Interrupts !!!!!

 

thanks

ttlappalainen, 2021-06-21:
There is other memory eaters too - you can find them on document. But e.g. NMEA2000 requires that you should be able to handle 4 fast packet simultaneously. I do not prefer to use original Arduino devices at all. Teensy or ESP32 has lot more memory. Also they eat less current than e.g. Due.