ttlappalainen/NMEA2000 #222: How to port away from Arduino?

peterantypas, 2021-06-05:
Hi Timo,

I am the creator of MAIANA, the open source AIS transponder here: https://github.com/peterantypas/maiana

With the core of the design done, I want to start looking at different adapters for connecting MAIANA to other systems, and N2K is definitely important. For this purpose, I'm planning to design a custom board and use your library with STM32+Eclipse as I am not into the Arduino ecosystem.

I looked at the different implementations and it seems to me that each "driver" class overrides the following 4 methods at a minimum:

- CANSendFrame
- CANOpen
- CANGetFrame
- InitCANFrameBuffers

In addition, I noticed that some implementations have a dependency on the Arduino debug output and Serial class. Is there any other important dependency that I've missed?

Thanks,
Peter

ttlappalainen, 2021-06-05:
That is correct. But note that library has been ported also for RPi, whic does not have Serial class. Library actually uses N2kStream class, which you need to port for STM. Check e.g. example NMEA2000ToNMEA0183, which I tested also on RPi.

The other proble is with CAN drivers. Currently library supports Teensy 3.2, 3.5, 3.6, 4.0, 4.1, ESP32, MCP2515 and some others. And for all those I had to fix driver to support NMEA2000. Main problem is that NMEA2000 uses fast packet, which means that long data will be sent on several separated frames. And those frames has to be in order. And there come problem. CAN controllers has normally multiple "Mailboxes" ( e.g. Teensy 3.2 has 16) and default CAN drivers just uses next free. This may cause that fast packet 1. frame would be written to Mailbox 7 and 2. frame to Mailbox 2. But Mailbox 2 has higher priority than 7 so it will be sent first and order has been broken. So please study first how STM driver behaves, otherwise you will have problems.

peterantypas, 2021-06-05:
Thanks for the quick response! I'm going to start digging further into ST's CAN implementation.

peterantypas, 2021-06-17:
I have decided to use MCP2515 + line driver instead. The MCU will be STM32L422 which runs at 80MHz. Do you think that's good enough for a node that mostly sends AIS messages? It will be decoding NMEA0183 messages from a UART and bridging them to N2K. It won't be doing anything else. Also, which library do you think I should use as a starting point?

ttlappalainen, 2021-06-18:
Arduino Mega with 8k memory is enought to do rather many things, if one write memory saving code. If you have >16k, should be possible to rathe complex things.

Under my Git there is also NMEA0183 library, which has same logic as NMEA2000. Check example NMEA0183ToN2k under NMEA0183 library.

ronzeiller already made NMEA0183->NMEA2000 AIS conversion code. Check https://github.com/ronzeiller/NMEA0183-AIS

peterantypas, 2021-06-18:
Sorry, my question was actually about MCP2515. Which library should I start porting to STM32? mcp_can? and then maybe NMEA2000_mcp? Just trying to understand the architecture.

ttlappalainen, 2021-06-18:
Yes, mcp_can and NMEA2000_mcp both from my git. But how much different STM32 is? Doesn't they compile?

peterantypas, 2021-06-18:
No, not right out of the box. I'm porting from Arduino to either Eclipse or STM32CubeIDE using ST's HAL plus my own BSP abstractions. I'll have to change the SPI layer for sure, and also interrupt handling. I'll figure out the rest along the way. Thanks!

peterantypas, 2021-06-30:
Here is what I've done so far in my main(). Does this look reasonable for a node that only transmits certain messages? Is there anything obvious missing? Thanks!

    bsp_init();
  
    static const unsigned long TransmitMessages[] = {129038, 129039, 12979, 129809, 129810, 0};
  
  
    tNMEA2000 *n2k = new tNMEA2000_mcp();
    n2k->SetMode(tNMEA2000::N2km_NodeOnly);
  
    n2k->SetProductInformation("",          // Manufacturer's Model serial code
                               1,           // Manufacturer's product code
                               "MAIANA",    // Manufacturer's Model ID
                               "1.0.1",     // Manufacturer's Software version code
                               "1.0");      // Manufacturer's Model version
  
    n2k->SetDeviceInformation(999,        // Serial number -- we don't have one
                              195,        // AIS
                              60,         // Navigation
                              2047);      // Just an unassigned value
  
    n2k->ExtendReceiveMessages({0});
    n2k->ExtendTransmitMessages(TransmitMessages);
    n2k->Open();
  
    while (1)
      {
        // TODO: Dispatch our own message queue here
  
        // This is the NMEA2000 processing loop
        n2k->ParseMessages();
  
        // We have Systick at a minimum, so this is 1ms at most
        __WFI();
      }

ttlappalainen, 2021-06-30:
- Manufactures core must be <2047. Use e.g. 2046, which is free and I have used for library.
- You do not need to call n2k->ExtendReceiveMessages({0});

closed 2021-08-21 by peterantypas