ttlappalainen/NMEA2000 #224: Basic test with teensy 4.0 failing

wifijt, 2021-06-27:
I just bought a pair of teensy 4.0 and Waveshare SN65HVD230 CAN boards. I am using the message sender and datadisplay2 examples but compiled with the teensyx library like this

DataDisplay2
`#include <NMEA2000.h>
#include <NMEA2000_Teensyx.h>
#include <N2kMessages.h>
#include <N2kMessagesEnumToStr.h>
tNMEA2000_Teensyx NMEA2000;` 

Message sender
'#include <NMEA2000.h>
#include <NMEA2000_Teensyx.h>
#include <N2kMessages.h>
#include <N2kMaretron.h>
tNMEA2000_Teensyx NMEA2000;'

I have tx connected to pin 3 and rx connected to pin 4 (not sure if this is correct given the default pins on the board seems to be rx on 13 and tx on 11)

They compile and load fine but the message sender throws 
'PGN 127250 send failed' messages 

and the data display shows no messages

What else do I need to do to get this working?

ttlappalainen, 2021-06-27:
If you get sent failed, device does not communicate with BUS. Possible errors are:
- missing teminal resistors
- invalid bus connection
- invalid tranceiver connection
- broken tranceiver

Where di you got pins 3/4? According docs Teensy 4.0 has LRCLK2 on pin 3 and BLCLK2 on pin 4. Default CAN 1 pins are 23 CRX1 and 22 CTX1

Also you do not necessarily need to force Teensyx library. Just with #include <NMEA2000_CAN.h> selects it automatically according to board.

wifijt, 2021-06-27:
I was digging through docs on the teensy 3.x and saw  pin 3 & 4

switching to the 22 and 23 didn't fix the issue. 

Not sure where to look from here

ttlappalainen, 2021-06-28:
Possible cause list was on my previous mail.