ttlappalainen/NMEA2000 #226: How to add support for PGN 129041?

peterantypas, 2021-07-08:
AIS message 21 (ATON report) is quite frequent near shore, so I'd like to add this to the library. Do you have any knowledge of it? I think I have enough resources to reverse engineer it if necessary.

ttlappalainen, 2021-07-09:
Do you mean 129041? For that I have information.

peterantypas, 2021-07-09:
Yes, I want to translate AIS message 21 to PGN 129041.

peterantypas, 2021-07-09:
I believe I figured it out. Will test the logic over the next few days and send you a PR.

ttlappalainen, 2021-07-09:
Check also PR https://github.com/ttlappalainen/NMEA2000/pull/214 what I complained about parameters at end so that you do not make same mistake.

MENIER, 2021-11-08:
There is this documentation to decode:
`   <PGNInfo>
      <PGN>129041</PGN>
      <Id>aisAidsToNavigationAtonReport</Id>
      <Description>AIS Aids to Navigation (AtoN) Report</Description>
      <Type>Fast</Type>
      <Complete>true</Complete>
      <Length>60</Length>
      <RepeatingFields>0</RepeatingFields>
      <Fields>
        <Field>
          <Order>1</Order>
          <Id>messageId</Id>
          <Name>Message ID</Name>
          <BitLength>6</BitLength>
          <BitOffset>0</BitOffset>
          <BitStart>0</BitStart>
          <Signed>false</Signed>
        </Field>
        <Field>
          <Order>2</Order>
          <Id>repeatIndicator</Id>
          <Name>Repeat Indicator</Name>
          <BitLength>2</BitLength>
          <BitOffset>6</BitOffset>
          <BitStart>6</BitStart>
                 <Type>Lookup table</Type>
          <Signed>false</Signed>
          <EnumValues>
            <EnumPair Value='0' Name='Initial' />
            <EnumPair Value='1' Name='First retransmission' />
            <EnumPair Value='2' Name='Second retransmission' />
            <EnumPair Value='3' Name='Final retransmission' />
          </EnumValues>
        </Field>
        <Field>
          <Order>3</Order>
          <Id>userId</Id>
          <Name>User ID</Name>
          <BitLength>32</BitLength>
          <BitOffset>8</BitOffset>
          <BitStart>0</BitStart>
          <Units>MMSI</Units>
                 <Type>Integer</Type>
                 <Resolution>1</Resolution>
          <Signed>false</Signed>
        </Field>
        <Field>
          <Order>4</Order>
          <Id>longitude</Id>
          <Name>Longitude</Name>
          <BitLength>32</BitLength>
          <BitOffset>40</BitOffset>
          <BitStart>0</BitStart>
          <Units>deg</Units>
                 <Type>Longitude</Type>
                 <Resolution>0.0000001</Resolution>
          <Signed>true</Signed>
        </Field>
        <Field>
          <Order>5</Order>
          <Id>latitude</Id>
          <Name>Latitude</Name>
          <BitLength>32</BitLength>
          <BitOffset>72</BitOffset>
          <BitStart>0</BitStart>
          <Units>deg</Units>
                 <Type>Latitude</Type>
                 <Resolution>0.0000001</Resolution>
          <Signed>true</Signed>
[... 249 lines cut ...]
          </EnumValues>
        </Field>
        <Field>
          <Order>22</Order>
          <Id>reserved</Id>
          <Name>Reserved</Name>
          <BitLength>3</BitLength>
          <BitOffset>205</BitOffset>
          <BitStart>5</BitStart>
                 <Type>Binary data</Type>
          <Signed>false</Signed>
        </Field>
        <Field>
          <Order>23</Order>
          <Id>atonName</Id>
          <Name>AtoN Name</Name>
          <BitLength>272</BitLength>
          <BitOffset>208</BitOffset>
          <BitStart>0</BitStart>
                 <Type>ASCII or UNICODE string starting with length and control byte</Type>
          <Signed>false</Signed>
        </Field>
      </Fields>
    </PGNInfo>
    <PGNInfo>`