ttlappalainen/NMEA2000 #227: AIS Transceiver Information should not be hard-coded in PGNs 129038, 129039 and others

peterantypas, 2021-07-11:
In some messages, the AIS Transceiver Information field is hard-coded as 0xff, which creates problems downstream.

Specifically for AIS, without this information, a display or gateway may not know whether to treat the message as a received packet or as a transmitted packet (AIVDM vs AIVDO in NMEA0183 terms).

I have no choice but to modify this interface in my own copy of the library, because it breaks at least one gateway device that I'm testing with. My implementation of PGN129041 (AtoN) does carry the AIS Transceiver Information field, so it renders correctly on the other side.

It looks like I'm going to have  to do a proper fork here ...

ttlappalainen, 2021-07-12:
You can add AIS Transceiver Information to function parameter list at end and set it to default 0xff. Then it does not affect to any existing code. Other way is to make totally new interface by using struct as parameter. Then you can also fill struct with constructor.

ttlappalainen, 2021-08-24:
I closed your PR. I had to update N2k library, so please reload it and duo your updates. And please do only necessary changes and not full formatting. It will be easier to follow/compare changes, if there are only necessary changes on code.