ttlappalainen/NMEA2000 #228: Best Micrcontroller to use

bigalnz, 2021-08-02:
The little documentation I can find seems to be several years old.

In the absence of a detailed guide is the best supported hardware to buy the Arduino Uno R3 or?

Combined with one of these : https://www.ebay.com/itm/263690841938?hash=item3d6532ef52:g:2R8AAOSw94Ba-6me

Will this setup with appropriate cable let me do everything a Actisense NGT-1 would do? (N2k reader/writer)

Thanks

ttlappalainen, 2021-08-02:
I prefer Teensy 3.2 ... 4.1 or ESP32, if you  need WiFi. I have not used much Arduinos for long time and do not  know does they have any good boards for NMEA 2000.

Forget everything smaller than Arduino DUE. Mega is just on the limit, but both are physically big and eats lot power compared to above.

bigalnz, 2021-08-02:
Hi TImo,

Thanks for the quick reply.

Maybe something like this : https://www.pjrc.com/store/teensy41.html

Your code will run on the above board too?

Thanks

Al

ttlappalainen, 2021-08-02:
I have tested on on Teensy 4.0, but it has same CAN controllor, so should not be any problem. You need then NMEA2000 and NMEA2000_TeensyX libraries

bigalnz, 2021-08-02:
Do you think it will do the same as : https://actisense.com/products/ngt-1-nmea-2000-to-pc-interface/

ttlappalainen, 2021-08-02:
Hw is better. Depends of sw. There is example ActisenseListenerSender, which sends and receive data. Then you can use NMEA Simulator http://www.kave.fi/Apps/ with it.

bigalnz, 2021-08-02:
You mean NGT-1 hardware is better than Teensy4.1?

ttlappalainen, 2021-08-02:
Teensy has better hw

bigalnz, 2021-08-03:
Would teensy4 do? 4.1 is sold out everywhere.

ttlappalainen, 2021-08-04:
Yes. It just does not have flash card. All Teensy 3.2 and up will run ActisenseListenerSender and much more.

bigalnz, 2021-09-02:
Hi Timo,

My board (Teensy 4.1) arrived today.

I downloaded NMEA2000-master and NMEA2000_Teensy added libs to Arduino IDE,
but then what do I do :-0

Cheers

Al

ttlappalainen, 2021-09-02:
You need NMEA2000 and NMEA2000_TeensyX for Teensy 4.0 and 4.1. You can test it with example ActisenseListenerSender and NMEA Simulator (www.kave.fi/Apps), if you have parts to connect your Teensy to the NMEA2000 bus.

Your question is bit funny "but then what do I do". How I would know that? I have no idea of what you have and what you can do and what is your goal, but definetely I'll not guid you how to start from scratch. In the library document there are some very basic samples to start with library and Arduino environment.  Please read those. Either I do not know do yo even have any basic knowledge of electronics or how to connect to NMEA2000 bus.

bigalnz, 2021-09-02:
Hi TImo,

Duh - sorry about that. I have some programming experience in Java. I have
uploaded the blinking light examples as a test.

My aim is to create a programmer for my Actisense EMU-1 (in other words
create a poor mans NGT-1).

So I presume:

1. Load the libs that you detailed above
2. Open the ActisenseReaderSender code. Compile and upload.
3. Wire it all up?

I have not got far enough yet to work out how to connect these two together:

[image: image.png]

Then the pins to connect the NMEA 2000 cable to.

Might make a video about it when I am all done :-)

Cheers

Al
SV Sunrise [NZ]

ttlappalainen, 2021-09-02:
I do not see the picture, but I hope you have tranceiver chip like MCP2562 between Teensy Rx/Tx and CAN-H/CAN-L.

bigalnz, 2021-09-02:
This link?

https://photos.app.goo.gl/WpyZW14dTzCuF96Z9

bigalnz, 2021-09-02:
The board that I got by the way uses Texas Instruments SN65HVD230.

ttlappalainen, 2021-09-02:
OK. Good luck! If you read through other issues and forum https://forum.arduino.cc/t/nmea-2000-shield/51012/1102 you will find that you are not first one having problems with that chip. At least yo hould find out the meaning of R1 and R2. Note that in CAN BUS there should be exactly 2 120 ohm termination resistors.

bigalnz, 2021-09-02:
Ok I buy another one then. They are cheap.

bigalnz, 2021-09-02:
Ok I ordered one with MCP2515 like this :
https://www.ebay.com/itm/324720229995?hash=item4b9ad57e6b:g:m4IAAOSwngpg9QD3

Probably cant do much till it arrives.

What are the screw terminals on it for?

Thanks again

Al

ttlappalainen, 2021-09-03:
Missed again. I have to write some hw instructions to library documents...

That board has CAN tranceiver, but Teensies already have better one. You need simple MCP2562  do inisolated or ISO1050 for isolated connection. For MCP2562 there is document https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/ArduinoDUE_CAN_with_MCP2562.pdf for DUE connection. MCP2562 connection to Teensies or ESP32 is similar.

bigalnz, 2021-09-03:
Ok, now I am even more confused.

I need a MCP2512 controller not transceiver - is that what you mean?

Is a transceiver different to a controller?

Sigh!

ttlappalainen, 2021-09-03:
No. With Teensies or ESP32 you need unisolated MCP2562 or isolated ISO1050 tranceiver. Those boards has build in CAN controller.

Controller means that it takes care of bus arbitration. Tranceiver is just "level shifter" and converts 3.3 V Tx/Rx to CAN bus differential levels.

ttlappalainen, 2021-09-05:
Hi all,

Could you make me favour and read and comment document https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/Connecting_hardware_to_NMEA2000.pdf

bigalnz, 2021-09-05:
Happy to help. Do you have it in a format I can edit or markup?

ttlappalainen, 2021-09-05:
You can add notes to .pdf.

bigalnz, 2021-09-05:
Hi Timo,

What would you like to know?

From the perspective of a lay person who is new to this, I think this
document would help, but would not have taken me all the way.

What did you find confusing I hear you ask?

*Canbus Controller/Transceiver*
I did not realise that the controller and transceiver are two different
functions, and that the Teensy has a controller built in, but not a
transceiver. One must be careful of buying boards on eBay as is you search
MCP2515 you get boards that do both - for Duo and Teensy 4.1 - this is NOT
what you want.
Insert link to boards

*Buying Canbus Transceiver*
I have still not found a MCP2515 transceiver - only boards that do both the
transceiver and controller. Can you include link and or picture to
examples?

*Pictures*
I would be inclined to include some pictures of a wired up teensy to
transceiver.

I would also think about maybe a getting start quick guide - there is a lot
of background in the document you have done.

I plan to make a tutorial once I get mine going, but it is a project to
acheive a specific purpose (create a NGT1 to program my EMU1)

Does this make sense?

Al

ttlappalainen, 2021-09-05:
I would like to know does document help beginner as you are:)

*Canbus Controller/Transceiver*
Document should explain what you need. It is problematic to put any links to eBay or any shop, since they arechanging all the time. Point is that I give information what you need and then you find it.

*Buying Canbus Transceiver*
There is no MCP2515 tranceiver. MCP2515 is CAN controller. MCP2551 is tranceiver, but it is only 5V. Thats why I'll prefer MCP2562, since that works with any IO level.

*Pictures*
I expect people should be able to read schematics. Have to think about that.

Also I see that written background is important to understand. With quick start guide I should go directly to the isolated solution to be safe. I tried to keep background as short as possible. This is a bit different than just making solution to blink led. 

If there would be enough market, I would make some own shield.

bigalnz, 2021-09-05:
Right - so even in your email I have learnt a lot more

MCP2515 = CAN Controller
MCP2551 = CAN Transceiver (5v only)
MCP2562 = CAN Transceiver (3.3v - preferred)

I understand what you are saying about links. Pictures would do it.

It feels a bit like you are saying to people *"here is how it all works and
what you need to consider, which parts you get are up to you"*. I am just
thinking some people may prefer a precise list of known compatible parts?

I *still* cant find a MCP2562 board??? Got any links? I will have another
look at the manual tomorrow, but it may make more sense, once I have got
mine going, then I can document it.

I plan on a youtube style tutorial.

AG

ttlappalainen, 2021-09-05:
I did not either found "board" with MCP2562. When I started, I did not search board for MCP2515 with MCP2551. I ordered parts and soldered my own and did not had any problems. Thats also why I had no problems with MCP2562 - just ordered chip and soldered own board for it.

Problem is that all those CAN shields are really CAN shields and not for NMEA2000. Your SN65HVD230 may work, if you remove those resistors on it.

ttlappalainen, 2021-09-05:
Now I found one: https://copperhilltech.com/isolated-can-bus-breakout-board-5-vdc/

ttlappalainen, 2021-09-05:
One can get ISO1050 chip in DIP as low as 3.60 $. Would save Teensy board, by learning soldering. Or actually you can wire DIP package to test board.

bigalnz, 2021-09-05:
Yes - original here :

https://www.skpang.co.uk/collections/can-bus-boards/products/can-bus-can-fd-breakout-board-5v-supply-and-3-3v-logic

The have designed some basic boards on EasyEDA and had them
printed....soooo cheap.

Something I might do in future for MCP2562.

bigalnz, 2021-09-05:
I might need to remove resistors, or definitely will?

ttlappalainen, 2021-09-05:
You will need. As in document, bus must have exactly 2 termination, which I expect you already have on your boat. If you connect this, you will have three. Also that has been centers to gnd with two 60 ohm. This is not used on NMEA2000.

So just remove both and if you make test bench do it as in my document.

ttlappalainen, 2021-09-05:
My first Teensy 3.2 + MCP2562 + PSU + fuse + 1-wire took less space than that CAN-Bus CAN FD Breakout Board.

ttlappalainen, 2021-09-05:
One more hint: You do not need to quote my previous answer.

bigalnz, 2021-09-05:
Hi Tmio

I was checking if it was a must or a maybe. Nothing else I have read take
about removing them.

It sounds like a must.

Thanks

Al

bigalnz, 2021-09-06:
Do I need to bridge the pads once the resistors are removed?

ttlappalainen, 2021-09-06:
No. But as I said you need termination resistors in both end of backbone as in my document chapter 6.

bigalnz, 2021-09-06:
The network on the boat has those blue raymarine terminators on the end of
the backbone. This is the same thing right?

So I remove the resistors but don't close the circuit with a piece of wire?

Cheers

Al

ttlappalainen, 2021-09-06:
No. That would shortcut CAN-L and CAN-H

bigalnz, 2021-09-18:
So I got the cable. It has 
CAN-H
CAN-L
and Red and Black wires which appear to be 

[image]

Can the Black go to ground on the Teensy?
Where should the Red go?

Thanks

Al

ttlappalainen, 2021-09-19:
Read my document https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/Connecting_hardware_to_NMEA2000.pdf
On chapter 5 there is connection examples. If you feed your device with USB, you leave red 12 V unconnected, but you have risk on groundloop.

bigalnz, 2021-09-19:
The issue with the docs is it relates only to the MCP2562 - which cant be brought as just a transceiver - well I think there was one place in the UK - but by and large SN65HVD230 is most common.

When you say you can power your "device" from NMEA2000 bus - do you mean connecting the Red and Black wires to the Teensy4.1? (If so which pins?)

I also have no idea if the SN65HVD230 is isolated or unisolated transceiver?

ttlappalainen, 2021-09-19:
I do not want to advertise SN65HVD230, since there has been so much problems with that. It is unisolated. In document there is clear schema that if you want to power device from bus, you need DC-DC converter. Teensy boards can be powered with 5V only by cutting USP power pad as shown in schema. 4.1 has similar pad.

bigalnz, 2021-09-19:
Hi Timo, I see. But if I type in MCP2562 into ebay or google only the chips
themselves come up not a board.
I would intend to power my board with power from the USB bus if this is
possible? And if that is possible should the ground wire be connected to
the Teensy?

ttlappalainen, 2021-09-19:
Then you connect ground to Teensy, but this is risky to create ground loop as I have explained in document. If you are just testing on table with few devices, it does not matter, but in boat it may cause problems.

bigalnz, 2021-09-19:
I will run laptop off it's internal battery so no other path to boat
ground. Does that mitigate the loop risk?

mrbubble62, 2021-09-19:
> I will run laptop off it's internal battery so no other path to boat ground. Does that mitigate the loop risk?
> […](#)
That will work, just don't try charging the laptop.

ttlappalainen, 2021-09-19:
Actually laptop powersupplies are isolated. But if you connect some unisolated sensors to board,...

bigalnz, 2021-09-20:
In the meantime (until I get to test this on boat) I am looking at ordering MCP2562 and designing a board to mount it. Does this look right?

[image]

ttlappalainen, 2021-09-20:
Yes, but if you make a board, why not to connect STBY to ground and add screw terminal for NMEA2000 connection as in some of my documents.

bigalnz, 2021-09-20:
You mean connect stby to ground on the teensy (and use screw in terminals)? I don't mind screw in terminals but that is just a part I solder to the board, the board remains unchanged or am I missing something?

ttlappalainen, 2021-09-21:
I mean that if you make own board for MCP2562, it would be best to make more improved on NMEA2000 side. Add pads for 5 pin screw terminal for NMEA2000. You could also have optional place for DC-DC converter so that you could use it also as powering from bus side. Connect STBY to ground directly or use resistor. Add also capacitor between Vss and Vdd. See my document chapter 5.2 Teensy 3.2, simple unisolated. 

Screw terminal type is "21.95MV/5-E Terminal Block, Vertical, Header, 5.08 mm, 5 Ways, 20 A, 320 V, Through Hole Vertical" See e.g. on Farnell: https://fi.farnell.com/imo-precision-controls/21-95mv-5-e/terminal-block-header-5pos-th/dp/2517881?ost=2517881
This connector type is used in DeviceNet.

Then you would have board, which can be used with any board having internal CAN (Teensy, ESP32). If you do not wnat to use screw terminal, you just solder wire directly to the screw terminal pads. Or optionally you can solder DC-DC and use it as bus powered board as soon as other side is isolated (optoisolation or just simple isolated sensors).