ttlappalainen/NMEA2000 #229: Mega2560 w/ MCP2515 module, receive not working unless primed with another sketch

arduinomnomnom, 2021-08-08:
I have a problem where the only way that I can get two Mega2560s to talk to one another (TemperatureMonitor -> DataDisplay2) is by "priming" the external MCP2515 module with a different sketch/library before loading the DataDisplay2 one.

1. I have two Mega2560 boards each connected to an external MCP2515 module (https://www.amazon.co.uk/AZDelivery-MCP2515-CAN-Module-Parent/dp/B086XGHW25)
2. Interrupts enabled, all wiring double-checked and terminating resistors 120R both ends.
3. Both powered via USB connected to a PC to watch the serial outputs.
4. Sender board is running the TemperatureMonitor sketch. Running on it's own, the serial output shows messages followed by "Send Failed" (as expected).
5. Receiver board is running the DataDisplay2 sketch. Serial output shows "CAN device ready Running...", **but no further output**.
6. When receiver board is running, the sender serial output no longer shows the "Send Failed" messages (great).

The only way so far that I have managed to get the receiver board to work (i.e., start showing the sent messages) is to load another sketch using the library from https://github.com/autowp/arduino-mcp2515, then reload the DataDisplay2 sketch again. The serial output then shows the temperature messages.

I can reset the Mega2560 and it restarts and continues to work, but if I unplug it from USB and replug it, it goes back to the "Running..." stage with no further output.

The MCP2515 sketch I run is:

```
#include <mcp2515.h>
MCP2515 mcp2515(53);  // CS pin
void setup() {
  Serial.begin(115200);
  mcp2515.reset();
  mcp2515.setBitrate(CAN_500KBPS);  // speed here does not matter; DataDisplay2 works fine later
  mcp2515.setNormalMode();
  Serial.println("------- CAN Read ----------");
}
void loop() {
}
```

Grateful for any suggestions how to troubleshoot this. 

Note that in the DataDisplay2 sketch, I have experimented with and without defining the interrupt, as well as playing around with the buffer sizes. The above procedure works regardless.

ttlappalainen, 2021-08-08:
Do you use mcp_can from my git https://github.com/ttlappalainen/CAN_BUS_Shield?

I did not had this kind of problem with Mega board. Either I do not use Arduino boards anymore except sometimes for testing. Tensy 3.2 and up or ESP32 are much better.

arduinomnomnom, 2021-08-09:
Yes, I have used the libraries given in the README, and I'll keep the other boards you recommend in mind; I have an ESP32 at hand.

In any case, I think I have found the solution to the issue with my MCP2515 external module; it seems to need the masks and filters cleared on startup. I saw this in the MCP2515 library reset function, so I have modified my DataDisplay2 sketch as follows:

```
MCP_CAN CAN(N2k_SPI_CS_PIN); 

void setup() {
    Serial.begin(115200);
    while (!Serial);
    OutputStream = &Serial;
    
    CAN.begin(CAN_125KBPS);
    CAN.init_Filt(0, 0, 0);
    CAN.init_Filt(1, 1, 0);
    CAN.init_Filt(2, 0, 0);
    CAN.init_Filt(3, 0, 0);
    CAN.init_Filt(4, 0, 0);
    CAN.init_Filt(5, 0, 0);
    CAN.init_Mask(0, 1, 0);                         
    CAN.init_Mask(1, 1, 0);

    NMEA2000.SetForwardStream(OutputStream);
    NMEA2000.SetForwardType(tNMEA2000::fwdt_Text);
    NMEA2000.EnableForward(true);
    NMEA2000.SetMsgHandler(HandleNMEA2000Msg);
    // NMEA2000.SetN2kCANMsgBufSize(2);
    // NMEA2000.SetN2kCANSendFrameBufSize(50);
    // NMEA2000.SetN2kCANReceiveFrameBufSize(50);
    NMEA2000.Open();
    OutputStream->println("Running...");
}
```
The above was derived from the RESET function in mcp2515.cpp file of the arduino-mcp2515 library I mentioned above:
```
    // clear filters and masks
    // do not filter any standard frames for RXF0 used by RXB0
    // do not filter any extended frames for RXF1 used by RXB1
    RXF filters[] = {RXF0, RXF1, RXF2, RXF3, RXF4, RXF5};
    for (int i=0; i<6; i++) {
        bool ext = (i == 1);
        ERROR result = setFilter(filters[i], ext, 0);
        if (result != ERROR_OK) {
            return result;
        }
    }

    MASK masks[] = {MASK0, MASK1};
    for (int i=0; i<2; i++) {
        ERROR result = setFilterMask(masks[i], true, 0);
        if (result != ERROR_OK) {
            return result;
        }
    }
```
Now, the DataDisplay2 sketch works even after a power cycle. Excellent!

Thanks for your fantastic libraries. I'm off to my boat to see if I can get speed and depth data off of my NMEA2000 network. If so, I'll be able to build a temporary replacement for a failed IS40 display.

Greetings from Singapore,
Patrick

arduinomnomnom, 2021-08-09:
(I have updated the title of this issue, in case anyone else using these particular MCP2515 modules runs into this problem.)

ttlappalainen, 2021-08-09:
It must be related to your module reset, since you are the first having this problem.

arduinomnomnom, 2021-08-09:
Tested on my NMEA2000 network and the device is pulling in the data I need. Thank again!

closed 2021-08-09 by arduinomnomnom