ttlappalainen/NMEA2000 #230: BatteryMonitor product information format

cth000182, 2021-08-13:
Is the format incorrect for the product information listed in the BatteryMonitor.ino example?

`const tNMEA2000::tProductInformation BatteryMonitorProductInformation PROGMEM={
                                       1300,                        // N2kVersion
                                       100,                         // Manufacturer's product code
                                       "Simple battery monitor",    // Manufacturer's Model ID
                                       "1.1.0.14 (2017-06-11)",     // Manufacturer's Software version code
                                       "1.1.0.0 (2017-06-11)",      // Manufacturer's Model version
                                       "00000001",                  // Manufacturer's Model serial code
                                       0,                           // SertificationLevel
                                       1                            // LoadEquivalency
                                      };`

From NMEA2000.h
` void SetProductInformation(const char *_ModelSerialCode, // Default="00000001". Max 32 chars. Manufacturer's Model serial code
                               unsigned short _ProductCode=0xffff,  // Default=666. Manufacturer's product code
                               const char *_ModelID=0, // Default="Arduino N2k->PC". Max 33 chars. Manufacturer's  Model ID
                               const char *_SwCode=0, // Default="1.0.0.0". Max 40 chars. Manufacturer's software version code
                               const char *_ModelVersion=0, // Default="1.0.0". Max 24 chars. Manufacturer's Model version
                               unsigned char _LoadEquivalency=0xff,  // Default=1. x * 50 mA
                               unsigned short _N2kVersion=0xffff, // Default=2101
                               unsigned char _CertificationLevel=0xff, // Default=1 
                               int iDev=0
                               );`

ttlappalainen, 2021-08-15:
You are comparing tNMEA2000::tProductInformation structure initialization and tNMEA2000::SetProductInformation function parameters, which are different. If you use structure, you set it simply with overloaded SetProductInformation function by providing only structure as parameter.

cth000182, 2021-08-16:
Thanks for the heads up. It was a long day of troubleshooting...

closed 2021-08-16 by cth000182