ttlappalainen/NMEA2000 #231: Migration from Arduino-Framework to PlatformIO on VisualStudioCode

JoergLaub, 2021-08-18:
I have a large project implemented with Arduino-Framework (1.8.13), also with many N2K-components - it works fine :-)
But now it's time to upgrade the framework from Aruduino to PlatformIO (my favourite for IoT).
All other funktions with PlatformIO works fine, but no more CAN/N2K-messages are received (the N2K-libraries are the same).
Has someone already used PlatformIO with NMEA2000 or a idea?

ttlappalainen, 2021-08-18:
I use and several others. Take care that you have all N2k related libraries from my git under you project lib path. So e.g. if you use e.g. Teensy , you should have NMEA2000, NMEA2000_Teensy, FlexCAN libraries from my git under project lib path.

JoergLaub, 2021-08-18:
I use a Arduino-DUE and these libs:
[image]

ttlappalainen, 2021-08-18:
I have not used DUE with PlatformIO, but do not see why it would not work if due_can-master is downloaded from my git. 

You could take e.g. DataDisplay2 example and make that for PlatformIO and check does that show any bus messages. And take care you have bus connected right and termination resistors connected.

JoergLaub, 2021-08-18:
The DUE worked since 5 years with many N2K-components in the boat - the bus is working fine.
With the same libs on Arduino-framework build und uploaded on board ist works - with PIO build und upload no message received.

There are many #ifdef for Arduinos in the due_can.h/.cpp files for initializing CAN-bus such are:

[image]

and so i think there is a context.

I stay at home and have no N2K-components for testing here, but at next onboard i will test DUE pure with DataDisplay2 example.

If there someone with DUE and N2K with PIO - please get an info?

JoergLaub, 2023-05-05:
Solved :-)

A copy&paste error - I forgot to delete 'void' after paste the funtion definition in my code :-(

void myfunction()
....
    void initNMEA2000(); // Bus init
....

Now all works fine with DUE.

closed 2023-05-05 by JoergLaub