ttlappalainen/NMEA2000 #233: Wind Data

pkdjsolo, 2021-08-30:
Hi,

Can the library be used to convert NMEA 2000 wind data to clear text?  

Is the wind data PGN already in the list?

Thanks

Paul

ttlappalainen, 2021-08-31:
With simple search "Wind" on N2kMessages.h you will find:
```
//*****************************************************************************
// Wind Speed
// Input:
//  - SID                   Sequence ID. If your device is e.g. boat speed and wind at same time, you can set same SID for different messages
//                          to indicate that they are measured at same time.
//  - WindSpeed             Measured wind speed in m/s
//  - WindAngle             Measured wind angle in radians. If you have value in degrees, use function DegToRad(myval) in call.
//  - WindReference         Wind reference, see definition of tN2kWindReference
// Output:
//  - N2kMsg                NMEA2000 message ready to be send.
void SetN2kPGN130306(tN2kMsg &N2kMsg, unsigned char SID, double WindSpeed, double WindAngle, tN2kWindReference WindReference);

inline void SetN2kWindSpeed(tN2kMsg &N2kMsg, unsigned char SID, double WindSpeed, double WindAngle, tN2kWindReference WindReference) {
  SetN2kPGN130306(N2kMsg,SID,WindSpeed,WindAngle,WindReference);
}

bool ParseN2kPGN130306(const tN2kMsg &N2kMsg, unsigned char &SID, double &WindSpeed, double &WindAngle, tN2kWindReference &WindReference);

inline bool ParseN2kWindSpeed(const tN2kMsg &N2kMsg, unsigned char &SID, double &WindSpeed, double &WindAngle, tN2kWindReference &WindReference) {
  return ParseN2kPGN130306(N2kMsg,SID,WindSpeed,WindAngle,WindReference);
}

```
Then you just handle read values as you like. E.G.
Serial.print(msToKnots(WindSpeed)); Serial.print(" kn"); Serial.print(RadToDeg(WindAngle)); Serial.print(" deg");

pkdjsolo, 2021-08-31:
Fantastic.  Thank you for such a quick response.  Just waiting for a can bus interface module and an Arduino Mega to arrive and will give it a go.

Thank you

Paul

ttlappalainen, 2021-08-31:
OK.

At what point in documents or anywhere I should add that I prefer to use either Teensy 3.2 and up or ESP32 boards? Arduino Mega can just handle small projects. Arduino DUE is better. Both anyway are slower, has less memory and eats more power than Teensy or ESP32.

pkdjsolo, 2021-08-31:
Your documentation is spot on regards preferred boards. Was going to use Mega as I have used before but will use DUE instead.

Its my lack of reading your documentation that was the problem not your documentation.

Really appreciate your help and super fast responses.

Regards

Paul

ttlappalainen, 2021-08-31:
As I said I prefer to use Teensy 3.2 ... 4.1 or ESP32. Teensy 3.2 is smaller sized and faster than DUE. You just need MCP2562 or ISO1050 can tranceiver.

pkdjsolo, 2021-08-31:
Many thanks, will look at teensy 3.2 or esp32

Thanks

Paul

ttlappalainen, 2021-08-31:
This is easy, but very expensive: https://copperhilltech.com/teensy-3-2-with-can-bus-breakout-board/

Also note that if you do not use isolated tranceiver, you need to have your board near of NMEA2000 power feed to avoid ground loops.

pkdjsolo, 2021-08-31:
Thank you again, your support is fantastic.  ESP32 ordered.  Will let you know how I get on.

Paul

pkdjsolo, 2021-09-22:
Hi Again,

I am using a Arduino Due and MCP2562 transceiver and have got the actisense and display examples working but am struggling to display wind data despite the information you have previously provided due to my lack of Arduino programming skills.

Could you explain how to modify the data display example to include wind speed/direction. I've tried modifying but just get zero readings for speed and direction.  I have my anemometer connected to an NMEA 2000 backbone with a plotter which is displaying the wind information so I know the anemomneter is working OK.

Really appreciate your help

Paul

ttlappalainen, 2021-09-22:
You write your listener function similar as WaterDepth
```
void WindSpeed(const tN2kMsg &N2kMsg) {
    unsigned char SID;
    double WindSpeed;
    double WindAngle;
    tN2kWindReference WindReference;

    if (ParseN2kWindSpeed(N2kMsg,SID,WindSpeed,WindAngle,WindReference) ) {
        Serial.print("Wind speed: ");
        Serial.print(WindSpeed);
        Serial.print("m/s, angle: ");
        Serial.print(RadToDeg(WindAngle));
        Serial.print(" deg, reference: ");
        Serial.println(uint8_t(WindReference));
    }
}
```

Then modify:
```
void WindSpeed(const tN2kMsg &N2kMsg);

tNMEA2000Handler NMEA2000Handlers[]={
  {130306L,&WindSpeed},
  {128267L,&WaterDepth},
  {0,0}
};
```

pkdjsolo, 2021-09-22:
Timo you are a star.  

Thankyou so much for your help.  

Your support is fantastic in terms of content and speed of response.  Have used the modifications you kindly provided and have exactly the output I required and can now progress with my project.

Many many thanks

Paul

newbezz, 2021-09-23:
Hello, 

I try to follow the immense work you continue to do by expanding your system and also supporting all it-fumblers such as myself. The post below reminds me of an aborted effort I ventured into some time ago: The NMEA2000 rules do not allow devises to transmit their info on true wind speed&angle. It wants each device to calculate this based on info on apparent measures of wind and water speed. 
Calculating true wind can be done both roughly using the basic date and more refined by also taking into consideration info such boat heel and deviation. Have you ventured into this type of calculations? Github has several offerings, by laziness dictates me to ask you before I get back into the coding. 
Best regards,
Jorgen Christensen, Denmark.

ttlappalainen, 2021-09-23:
Never seen that rule. Where it has been defined? If I have device measuring apparent wind and either using data from bus or having itself necessary data, why it could not send information to the bus as calculated. Sounds like NMEA2000 logic to have setting to provide true wind and then not allow to use it. I wonder why, but not the only stupid definition on N2k.

Calculation is simple vector calculation and I do it on NMEA Simulator (without heel), where I then provide true wind information to the bus. Maybe should remove that.

Lynda96, 2022-03-07:
Hi 
I am working on communication with anenometer sensor and NucleoBoard H743ZI2 , this communication with NMEA2000 , I just wanna read the wind data(direction,speed) from the anenometer sensor and i read this data with NucleoBoard , I am coding with mbed studio OS ,so i wanna know if i need NMEA2000_CAN.h librairy ,or how can I modify  it in this case ?
Thank you

ttlappalainen, 2022-03-08:
I ca not answer for that without fully investigating your board and I do not have time for that. The board seem to contain internal CAN controller. If you are using MBED, it may be supported by MBED driver https://github.com/thomasonw/NMEA2000_mbed.
Module NMEA2000_CAN.h idea is to automatically select driver depending of processor or environment. It simply include necessary libraries and is easy way for beginners with commonly supported CAN controllers. MBED is listed as supported, but if you read thomasonw comment about MBED you find that it may not work.

I have never tested MBED and I think thomasonw gave up, since there was no support or interest from MBED side to fix MBED CAN drivers.

In generally there are different ways to select driver:
1. Just include NMEA2000_CAN.h so driver and NMEA2000 object will be created according to processor or environment:
`#include <NMEA2000_CAN.h>`
Note that there may be extra settings needed like SPI_CS wire for mcp_can etc.
2. Include NMEA2000_CAN.h, but force specific driver to be used. Driver lit is in the beginning of the NMEA2000_CAN.h
```
#define USE_N2K_CAN 6  // for use with MBED (ARM) systems
#include <NMEA2000_CAN.h>
```
3. Define everything by yourself
```
#include <N2kMsg.h>
#include <NMEA2000.h>
#include <NMEA2000_mbed.h>       // https://github.com/thomasonw/NMEA2000_mbed
tNMEA2000 &NMEA2000=*(new tNMEA2000_mbed());
tmbedStream serStream;

```