ttlappalainen/NMEA2000 #234: running on a pro16MHzatmega328

ieb, 2021-09-12:
Hi,

Building .pio/build/pro16MHzatmega328/firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.4% (used 377 bytes from 2048 bytes)
Flash: [========= ]  89.4% (used 27458 bytes from 30720 bytes)

I have this library running on a 5V Pro Mini 16MHz + MCP2515, with the recommended memory optimisations, however I had to patch NMEA2000-library  to get it to compile. Details here https://github.com/ieb/EngineManagement/blob/main/src/NMEA2000-library.patch. Please Ignore the prints of failed packets, I need that for feedback but don't want the full PGN output on the serial.

It may still crash loop when plugged into a live NMEA2000 bus or queried for device info, fingers crossed.

Why a 328 and not a Teensy ? 
I had a bag of them, they have enough ADC's onboard, and it draws 20mA running (not relevant for the EngineMonitor).

Also, I found the SPI requesting 10MHz resulting in 7MHz CLK on a Pro min produced an ugly square wave so reduced that to 1MHz. It could have been capacitance in my scope leads. at 1MHz, so far it seems stable.

The projects started as a J1939 device to drive Volvo Penta instruments and there is a collection of documentation on the PGN/SPI's used with the code as abandoned in https://github.com/ieb/EngineManagement/blob/main/src/main_deprecated.cpp and J1939 library in https://github.com/ieb/EngineManagement/tree/main/libarchive/ARD1939 (not mine).  I think we talked about J1939 once before and I promised to get back with info on success or fail. It was abandoned when I found the internals of my VP Tacho had a blown CAN interface inside the custom chip (lightning strike), in spite of surge protection.

As always the NMEA2000 lib is a pleasure to use. Thank you.

ttlappalainen, 2021-09-12:
As quick look I found only one necessary change: test on line 2103 ...!N2kMsg.IsTPMessage()... But in case of no TP support, whole handler must be disabled, since according to docs support for that should be only by TP broadcast. You do not need to change ForwardStream, since you can set it to Serial.

If you drop Teensy speed to 24 MHz, it draws only 13 mA from 12V with good DC-DC converter. I had Temperature using 1-wire with Teesny 24 MHz.

ieb, 2021-09-13:
Thanks for the info on the Teensey. 
Will give them a go when the pro mini bag empties.

> ... whole handler must be disabled
Good point, didn't  spot that, fixed.

RAM:   [==        ]  18.4% (used 377 bytes from 2048 bytes)
Flash: [========= ]  88.3% (used 27122 bytes from 30720 bytes)

Thanks for checking the patch and the advice.

closed 2021-09-13 by ieb