ttlappalainen/NMEA2000 #236: Problem compiling example ActisenseListenerSender on Arduino due board

Tur-Thure, 2021-09-24:
I try to compile example ActisenseListenerSender in NMEA2000 downloaded from Github. And I get the following error below. What am I doing wrong?

C:\Users\ULundqvist\Documents\Arduino\libraries\NMEA2000_due\NMEA2000_due.cpp: In member function 'virtual bool tNMEA2000_due::CANOpen()':
C:\Users\ULundqvist\Documents\Arduino\libraries\NMEA2000_due\NMEA2000_due.cpp:87:39: error: 'class CANRaw' has no member named 'getNumRxBoxes'
for (mailbox = 0; mailbox < CANbus->getNumRxBoxes()-1; mailbox++) {
^
C:\Users\ULundqvist\Documents\Arduino\libraries\NMEA2000_due\NMEA2000_due.cpp:91:28: error: 'class CANRaw' has no member named 'getNumRxBoxes'
for (; mailbox < CANbus->getNumRxBoxes(); mailbox++) {
^
exit status 1
Error compiling for board Arduino Due (Native USB Port).

ttlappalainen, 2021-09-25:
You have wrong due_can library. Remove it and download it from my git and install.

Tur-Thure, 2021-09-25:
Thx, now the compilation work :)