ttlappalainen/NMEA2000 #237: Problems with simple example to use NMEA reader on example TemperatureMonitor

Tur-Thure, 2021-09-27:
[image]

Hi I successfulle uploaded the TemperatureMonitor example after chaning to:

  Serial.begin(115200);
  NMEA2000.SetForwardStream(&Serial);

and 

//NMEA2000.EnableForward(false);

I downloaded NMEA reader according to instructions and get the attached error messages timeout and can't see any data. I tried both with the old and the new version. I selected the same com port in NMEA reader as in IDE. The Serial has not been opend in IDE.

What can be wrong?

Tur-Thure, 2021-09-27:
I also tried to uncomment the line below, and check with the Arduino serial monitor but could not see any data 

 // If you want to use simple ascii monitor like Arduino Serial Monitor, uncomment next line
  // NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show in clear text. Leave uncommented for default Actisense format.

Tur-Thure, 2021-09-28:
Below is what I get when I compile. I think it looks ok:

Sketch uses 37684 bytes (7%) of program storage space. Maximum is 524288 bytes.
Atmel SMART device 0x285e0a60 found
Erase flash
done in 0.040 seconds

Write 39904 bytes to flash (156 pages)
[==============================] 100% (156/156 pages)
done in 7.536 seconds

Verify 39904 bytes of flash
[==============================] 100% (156/156 pages)
Verify successful
done in 4.224 seconds
Set boot flash true
CPU reset.

ttlappalainen, 2021-09-28:
// If you also want to see all traffic on the bus use N2km_ListenAndNode instead of N2km_NodeOnly below
So uncomment and modify line
NMEA2000.SetMode(tNMEA2000::N2km_ListenAndNode,22);

Tur-Thure, 2021-09-28:
I Only have the PC and the Arduino Due connected. I uncomment and changed the line as follows: 

 // If you also want to see all traffic on the bus use N2km_ListenAndNode instead of N2km_NodeOnly below
  NMEA2000.SetMode(tNMEA2000::N2km_ListenAndNode,22);
  //NMEA2000.SetDebugMode(tNMEA2000::dm_Actisense); // Uncomment this, so you can test code without CAN bus chips on Arduino Mega
  //NMEA2000.EnableForward(false); // Disable all msg forwarding to USB (=Serial)
  // Here we tell library, which PGNs we transmit
  NMEA2000.ExtendTransmitMessages(TransmitMessages);
  NMEA2000.Open();

Still no data in NMEA reader and no data in IDE serial monitor with 

// If you want to use simple ascii monitor like Arduino Serial Monitor, uncomment next line
 NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show in clear text. Leave uncommented for default Actisense format.

Tur-Thure, 2021-09-29:
Some progress. :)
I connected to arduino native port and uploaded the sketch. then I moved the physical USB - cable to the programming port and now I get some data. Should native port be used for flashing? How do I know if serial data is sent on native port or programming port?

ttlappalainen, 2021-09-30:
Check from DUE documentation. If I remember right native port is SerialUSB and programming port is Serial.