ttlappalainen/NMEA2000 #238: Problem to get data from NMEA2000 Bus

Tur-Thure, 2021-10-03:
I have an old Raymarine E120 but it works fine. In the menu I found setting for "AIS 38400" so I selected baudrate to 38400
I can´t read any data either via actisense NMEA reader nor via arduino serial monitor.
I just get the following on the srceen.

[image]

I checked the unpowered NMEA network and measured 60 Ohm

I also checked the CAN H and CAN L with oscilloscope and could find a signal of 1,5 V
 I use the sn65 hvd 230 tranceiver.
I will try with the MCP 2562 tranceiver instead.

The led on the on the board only blinks with 1 Hz so I guess there is no reception of data. Could the baudrate be wrong or something else? The NMEA 2000 network on the boat works because I can see data coming from other devices to the E120 to be displayed.

Tur-Thure, 2021-10-03:
I changed to the MCP 2562 and now it works like a dream. :) :)
🥇

Tur-Thure, 2021-10-03:
The NMEA reader do not still work. Is there some setting that I missed?

[image]

I pushed "Update from hardware twice"

The serial monitor shows the following
[image]

Tur-Thure, 2021-10-03:
I Comment the line
// NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show in clear text

And then I get this info

[image]

Tur-Thure, 2021-10-03:
How do I easiest isolate a variable like engine speed starboard engine and put it into a variable that I can use for a calculation in my arduino sketch?

ttlappalainen, 2021-10-04:
Check example DataDisplay2. It has several parsing samples.

Tur-Thure, 2021-10-10:
Hi Timo,

Thank you for all your support. Now I able to get the data on the serial port. However, I would like to send a variable to another serial port like:

Serial1.print (EngineSpeed);

But when I try to compile I get an error saying EngineSpeed do not exist.

Is there someway I can isolate? 
EngineSpeed
FuelRate
EngineCoolantTemp
EngineBoostPressure
SOG

Seams like this variables are not available outside it's function

[image]

Tur-Thure, 2021-10-10:
Sorry to bother you with this question. I just defined a global variable and now it works :)

Lynda96, 2022-03-07:
Hello 
I dont understand what is this function  NMEA2000.ParseMessages(); for ?
Thank you

ttlappalainen, 2022-03-08:
There are e.g., example DataDisplay2 example how you handle received messages with callbacks.

Library has been designed to work also in single thread systems so something must take care of library routines. NMEA2000.ParseMessages() is for that purpose. In single thread system you must call NMEA2000.ParseMessages() in loop and take care that any other task you do in loop does not cause long pause for loop. Max loop time should be kept <50 ms and that only randomly.

And if you have multi thread system, you can have one thread calling NMEA2000.ParseMessages(); naturally you have to the take care data locking during callbacks.

Lynda96, 2022-03-11:
Hi,
Thank you  your support is fantastic.Thank you for such a quick response,
But as you said this code can't help me for my project cause I am working with an Mbed Board Nucleo Board (Stm32) ,
Thank you for all

ttlappalainen, 2022-03-11:
Library core itself works any system (or can be fixed to work). You need inherited driver for each different CAN controller and for MBED there is one as I pointed. You can also write all by yourself if you like, but unfortunately it does not help, since it does not fix main problem with MBED CAN library, which does not support forcing NMEA2000 fastpacket frames to be sent in order.