ttlappalainen/NMEA2000 #241: PGN127488 RPM setting when engine OFF ( N/A or 0)?

virtuvas, 2021-11-03:
morning Timo,

I notice that in my code I have:

```
    if (engineRPM == 0 ) 
      engineRPM =  N2kDoubleNA; 
    else if (engineRPM > 0)
      do whatever I have to do...
```

as a result when engine is off, tacho needle on plotter and GMI10s is missing - which is right.
Now, before I remove that if and get the needle back at 0, is there possibly a reason I've done it?
I know sounds silly asking you, but I cannot think of a reason myself and wonder if you suggested it 3-4yrs ago and I've forgotten 🙄 

cheers

V.

ttlappalainen, 2021-11-03:
Can not remember, I may have succested it. On the other hand this is not comple truth. Your engine RPM may be 0, but switch is on. In that case you should show 0 RPM. When your switch is off, you could set NA.

virtuvas, 2021-11-03:
thanks, 
well a bit more complicated as I'm talking about the generator and that's fully controlled by the teensy (even the starter via a SSR) so there's no "key" so to speak. Further, I'm logging EGT, coolant temp, fuel pressure etc in influxDB for debugging and testing various issues, so RPM could well be 0 all the time. May follow your suggestion on the main engines.

second Q, you've mentioned/suggested before to make a move to PlatformIO IDE.
Other than the struggle to learning another interface, will I have any real benefit? Will my teensy code run faster or something?
Most applications I've done, I don't care, but on the stabilisers with the MPU9250 data I would like some speed improvement.

thanks as ever!

V.

ttlappalainen, 2021-11-03:
Just use same code for both, but wire "key" always on for generator. Or what ever method.

Actually Vscode (Visual Studio Code) IDE with Platform IO extension. There is no speed benefit on changing to that, but whole programming is more professional. Easier to handle modules and you can have own libraries for each project. This is important since sometimes everybody does not take care of backward compatibility of libraries.

One "drawback" is that then you need to be more strict with c/c++ forward definition rule. In Arduino IDE you can have:
void setup() {
...
 InitMyStuff();
...
}

void InitMyStuff() {...}

But in real c/c++ programming above will not compile with error ...InitMyStuff has not been defined. So you have to either move your routine above setup or have forward definition
void InitMyStuff();
above setup. Normally this has been done so that you have your init on own module code on Init.cpp and definitions on Init.h as you have seen on libraries. So in a way it is "drawback", but when you have already passed beginner state it will teach you "better" programming.

virtuvas, 2021-11-04:
thanks Timo will do so,

re PlatformIO I hope you don't work in a marketing post, you definitely didn't "sell" that to me 😄 
if we end up with another lockdown, I'll make the move, if not, got plenty of things to keep me busy over the winter!

cheers

V.

ttlappalainen, 2021-11-05:
Happily I am not, but still would lik to sell it for you. It has lot of other good things:
- Syntax highlight
- Syntax checking. It will show lot ot basic errors without compiling
- Better window handling
- Better search
- etc,

I should first move back to Arduino IDE to tell more.

virtuvas, 2021-11-05:
hm, syntax highlight and checking without having to compile to find a missing bracket is enough of a reason to change!
will look at it when the weather turns colder 😃 

thanks

V.

closed 2021-11-05 by virtuvas