ttlappalainen/NMEA2000 #242: NMEA2000 to NMEA183 AIS AIVDM CODING

MENIER, 2021-11-08:
First great tanks! Its run perfectly with my ESP32....
My VHF RO6800(AIS) is NMEA2000 is there a code source library to SET AIS ( !AIVDM) NMEA183 Sentences? (PGN 129038,129039,129040,129041,129792 to AIVDM )?

MENIER, 2021-11-22:
It's done!; [(https://github.com/ronzeiller/NMEA0183-AIS)]

ttlappalainen, 2021-11-22:
Great. What develop environment you used and what device (ESP32?)? Could this be included to NMEA0183 library? NMEA0183 library works with small Arduino devices. If this does not, it could have ifdef to exclude those out or ifdef to include tested systems.

You defined MIT license for your project - right? Would be more clear to have in definition:
License
MIT License

Permission is hereby...

MENIER, 2021-11-26:
It's not my project(a fork for me...) it's Ronnie Zeiller's project .

ronzeiller, 2022-04-30:
Hi Timo,
updated license to your recommendations.

We talked about this project in 2019 here: https://github.com/ttlappalainen/NMEA2000/issues/154
It is working for me on Teensy 3.6 and ESP32, but as it is based on your library it should work on same devices as yours.
Develop environment is PlatformIO but should work in Arduino as well.

Feel free to include it to your NMEA0183 library. Would be nicer than an extra piece of software to integrate.