ttlappalainen/NMEA2000 #243: Two CAN controller with one ESP32

buhhe, 2021-11-16:
Hello Timo,
at first lots of thank for you sharing this great result of your work and for supporting learners like me and many other.
I have worked on coupling the J1939 CAN bus of my VP engine to my N2K network. My solution consists of two ESP32 boards with one CAN controller each. One ESP32 reads the J1939 datagrams and sends it via ESP_now using an interim data format. The other ESP32 receives the data and sends it to the N2K bus. Everything works fine.
But to reduce complexity it woud be helpful to hande two CAN controllers with only one ESP32 board. So I could reduce the coupling to only one ESP32 and could get rid of the ESP_now communication. Is that possible with your library?
Thanks in advance
Burkhard

ttlappalainen, 2021-11-17:
Should be. You need to add another CAN for ESP32 by using MCP2551 controller and use mcp_can. I prefer to use that on  J1939 side. I prefer to use ISO1050 isolated CAN tranceiver at least on NMEA2000 side to avoid grounding both CAN together.

Another option is to use Teensy 3.5, 3.6, 4.0 or 4.1. They have 2 CAN controllers build in.

buhhe, 2021-11-29:
Great, thank you. That will let me go further.

closed 2021-11-29 by buhhe

wjquigs11, 2023-09-08:
@buhhe were you successful? I'm struggling with the same issue (2 CAN bus on one ESP32).

buhhe, 2023-09-09:
Yes. you can find my solution here: https://github.com/buhhe/VolvoPenta-N2K_InterfaceVon meinem Dingens gesendet.

wjquigs11, 2023-09-11:
Thank you, your code is excellent and I will use it for my application. I only need to parse wind PGN and only need to get one value (apparent wind speed) so it will be ganz einfach!
Did you consider using mcp_can for the Volvo CAN side instead of parsing the messages?

buhhe, 2023-09-11:
Thanks for the feedback. I’m not shure if I understand your question in the right way. Since the VP engine’s CanBus  don’t use N2K PGN’s but only J1939 ones I did use MCP_CAN on that side.

So I could use Timo’s fantastic libaries on N2K side to assemble and send the appropriate PGN’s.

If you read N2K messasges and send N2K messages why can’t you connect your two CanBusses directly to each other?

wjquigs11, 2023-09-11:
I understand. I assumed those PGNs were N2K and not J1939 but it makes sense.
The reason I cannot connect the busses is because I am correcting data from the wind transducer on my rotating mast. I use a position sensor to adjust the Wind PGN and re-transmit using Timo's library. The issue is that my wind display (Garmin) has no capability to user-select the data source, so it follows whatever is on the bus. If I put the original data and the corrected data on the same bus, the display gets very confused.

buhhe, 2023-09-11:
Ok, sounds logical, nice project. 

 

Do you know this: https://open-boat-projects.org/en/

There are many people who develop electronics and software for their boats and helping each other. I'm sure they would be interested in your project.

wjquigs11, 2023-09-12:
Yes, I've explored that site and have found the examples from AK-Homberger particularly useful. Thanks for your help!

ttlappalainen, 2023-09-12:
On this https://github.com/ttlappalainen/NMEA2000/issues/343#issuecomment-1711046836 response I gave you rather exact instructions how you need to do it. You do not need to learn how to parse wind message from cratch bare can data. You can actually take example WindMonitor and first test that with Garmin to see that it works at all. Then add wind side CAN handler to code.