ttlappalainen/NMEA2000 #244: Add a preprocessor macro for disabling N2K fast packet support

mairas, 2021-12-16:
Given that this library is stable, quite complete, and has excellent hardware support (and also because it's poison I know...), I'd be interested in utilizing it for prototyping with other J1939/ISO11783-3 based protocols than NMEA 2000.

For that purpose, I would be interested in adding a preprocessor macro for optionally disabling N2K fast packet support. Is this something you'd like to consider?

What I would do is:

- Have a macro `N2K_NO_FAST_PACKET_SUPPORT`
- If the macro is defined:
  - never set `FastPacket` to true in `tNMEA2000:CheckKnownMessage()`
  - Make `IsFastPacket()` always return false
  - Disable the fast packet code block in `tNMEA2000::SendMsg()`

ttlappalainen, 2021-12-16:
I should have originally write library a bit different way - write a library J1939 and inherit NMEA2000 from it. Using NMEA2000 library for J1939 is not probably good idea. J1939 uses TP (Transport Protocol), which I have not seen to be used on NMEA2000. It is requirement and library does it but only up to 233 byte messages instead of nearly 2k bytes. So you may end up to that problem.

Also you should define out all NMEA2000 related things like Group function support (PGN 126208) Heartbeat (PGN 126993), proprietary message handling, configuration information (126998), product information (126996). Lot of code would disappear. Maybe that was original reason write directly NMEA2000. It did not had originally support for TP, which I added due to certification requirements.

With defines you mentioned you can test and use library, but I do not see sense to bring it to library. Definition would drop it to J1939 level, but all object naming refers to NMEA2000.

mairas, 2021-12-16:
There seemed to be defines for disabling most of the other N2K specific features, right? But true, I hadn't noticed the TP message maximum length limitation. That doesn't sound too difficult to patch either, though. Your comments on the object naming etc. do stand, though, and I can relate to your other concerns, too. I'll just do a fork of my own for my prototyping needs, then.

Digressing a bit, but I would assume the manufacturer specific firmware updating over N2K would be implemented with TP messages. So, if you'd want to flash some hacked firmware to your Raymarine autopilot, you'd need TP (if my assumption is correct, that is). :-)

ttlappalainen, 2021-12-16:
Those defines removed some memory consuming features so that one could use library in with smaller memory. On own boat it does not matter that much is library certified or not. I have planned to add one slot for big TP message for receiving. In requirement devices should be able to catch 4 fast packet and 1 TP message at same time and it is not too bad to have one TP, since controllers like Teensy and ESP32 has rather much memory.

I have not checked how does they upgrade fw. I have seen also proprietary fast packet to be used to transfer data.

mairas, 2021-12-16:
Just checked; NMEA2000 requires concurrent reception of 1 SF, 2 FP, 1 TP BAM and 1 TP RTS/CTS. But of course, N2K TPs need to be only as long as the maximum expected FP. (I had actually missed that part, but it's indeed quite clearly stated in the spec.) So, in principle, two TP buffers would be required. But for the beefier MCUs that still doesn't matter.

Anyway; I'll close this issue.

closed 2021-12-16 by mairas