ttlappalainen/NMEA2000 #251: Library calls message handlers for messages not intended for specific node

garaminowicz, 2022-02-22:
It seems there if function `bool tNMEA2000::HandleReceivedMessage(unsigned char Destination)` that checks if incoming message is intended for specific node, but it isn't called anywhere. It's also not possible to call from outside, as it is protected.

Is it expected behavior and message destination should be checked in user code?

ttlappalainen, 2022-02-22:
That had role on my design on very beginning, but has not been used later. I should clean it from code.

ttlappalainen, 2022-02-22:
Library catches all messages and will make call back for all new messages. User code can then handle messages of interest. Normally all data messages are sent as broadcast (255). Only special requests are sent as destination addressed. Library internally takes care of basic requests like requests for ISO address claim, product information and configuration information. User must take care of request for messages user code is transmitting.

garaminowicz, 2022-02-22:
It's clear now - I will compare it in message handler.

closed 2022-02-22 by garaminowicz