ttlappalainen/NMEA2000 #253: NMEA Simulator

usauerbrey, 2022-02-22:
May be a little of topic, but ist there any way to get the non-free version without the hardware. What would it cost?

"Try with my NMEA Simulator (http://www.kave.fi/Apps/NMEA-Simulator/NMEA-Simulator.7z). It shows on message console all messages sequentially and on message list it separates messages with different wind reference. Unfortunally free version does not show message data parsed, but you will at least see that messages are there."

Thanks Uwe

denravonska, 2022-02-22:
Maybe it's not applicable for your needs, but this is the simulator I use (same?): https://github.com/panaaj/nmeasimulator

usauerbrey, 2022-02-22:
Thanks for the hint, I will take a look to it.
Currently I use Timo's Simulator and would be happy to buy the enhanced version, but not spending $500 and getting a HW, I dont need ;-)

ttlappalainen, 2022-02-22:
Contact me direcly with email.

ttlappalainen, 2022-02-23:
As far as I understood the simulator you mentioned does not generate NMEA 2000 messages - "NMEA0183 and Signal K data streams".

My NMEA Simulator does NMEA0183 and NMEA2000. NMEA0183 is simple, so there is not much to develop on that side. The mot development has been done on NMEA2000 side. With proper hw NMEA Simulator also simulates devices on the bus. Currently it also does address claiming and message requests for simulated devices. Simulator is not only Simulator. It has also good analyzing tools on it and is far more better that Actisense NMEA reader. With free version you test nearly all features and also see parsed messages for first 100 messages. Several commercial device developers uses my Simulator - also two big ones.

As Simulator manual says, you can build your own box and I can provide unlock code for you for full features. The difference is that code is fixed to single computer while it moves with box.