ttlappalainen/NMEA2000 #255: mcp2515 with esp

ersingencturk, 2022-03-05:
installed the library working great with 2 mcp2515 modules working with each other in an isolated setup. I can run temperature example from one device and can see the messages from other. However when it comes to actually conncetion to a nmea2000 network it doesn't work. traffic stops on the network. also tried actisense example to read with nmea reader but no luck. what could be the issue ? my module has 8mhz crystal, with a jumper to add terminatior resistors. (leaving them out in the real network)

Module that I use Features:
This is a brand new MCP2515 CAN bus module TJA1050 receiver SPI protocol

Specifications:
Support CAN V2.0B specification, the communication speed 1Mb / S
0 to 8-byte data field
Three frames - standard, expand the frame and remote
DC5V power supply module & SPI interface protocol control
120 ohm termination resistors, Impedance matching, ensure drive capacity, long-distance data transmission against signal radiation

https://www.aliexpress.com/item/1005001621818243.html?spm=a2g0o.detail.1000014.1.67074265yAfAJf&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40000.267768.0&scm_id=1007.40000.267768.0&scm-url=1007.40000.267768.0&pvid=766ac9af-b78d-4b68-a5ac-0c7be5abb1a1&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40000.267768.0,pvid:766ac9af-b78d-4b68-a5ac-0c7be5abb1a1,tpp_buckets:668%232846%238111%23412&pdp_ext_f=%257B%2522sku_id%2522%253A%252212000016846296965%2522%252C%2522sceneId%2522%253A%252230050%2522%257D&pdp_pi=-1%253B32.99%253B-1%253B-1%2540salePrice%253BTRY%253Brecommend-recommend

ersingencturk, 2022-03-05:
i solved the issue by adding this:

```
#define N2k_SPI_CS_PIN 15    // Pin for SPI select for mcp_can
#define N2k_CAN_INT_PIN 2 
#define USE_MCP_CAN_CLOCK_SET 8
```

before

#include <NMEA2000_CAN.h>

make sure you have right pinout's for your board connection

closed 2022-03-05 by ersingencturk

thomasnisole, 2022-03-22:
Hello @ersingencturk ,

I tested with same configuration the MessageSender example with the DataDisplay2 example. I used esp32 38 Pins and CAN shields like that : 
[image]
but i can't view any messages in second program because the first one can't send any message, i have always PGN XXXXX send failed in monitor. Is it possible for you to share your two configurations (without nmea200 network and with) of wiring and your two programs ?

Thank you in advance.

Thomas.

ttlappalainen, 2022-03-23:
The examples works with ESP32 for 100% If you get "sent failed" message, you have problem with NMEA2000 bus or interrupt if mcp_can has been used. Since you use ESP32, but you have connected MCP2515 module to it, you may also have problem that you have not forced MCP2515 module to be used instead of internal CAN controller. Examples uses as default ESP32 own controller.  Read carefully document https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/Connecting_hardware_to_NMEA2000.pdf

What is the point of using that module with ESP32? ESP32 has build in CAN controller and is easier to use just with tranceiver chip like MCP2562 or ISO1050.

ersingencturk, 2022-03-23:
hi thomas as timo explained it might be a problem with how you used esp32 as it has internal can and yet I'm going to try it with a transreciever only . I used esp8266, that's why I needed this whole module.still if you need wiring of it I used like this:

[image]

later I changed the int pin from gpio2 (d4) to gpio4 (d2) since it turned out that pin was also used for internal led on the board that I used so the configuration of mine is updated to:

#define N2k_SPI_CS_PIN 15    // Pin for SPI select for mcp_can
#define N2k_CAN_INT_PIN 4
#define USE_MCP_CAN_CLOCK_SET 8

hope this helps.

thomasnisole, 2022-03-23:
Hello @ttlappalainen , @ersingencturk ,

First of all, thank you for your replies.
For information, I try to reproduce this remote control [remote control ](https://github.com/AK-Homberger/ESP32-Evo-Remote-Pilot-NMEA2000) but having 3 shields and 3 ESP32, I would like, as long as I do, not to have to buy another element, that's why I don't do exactly what you have write in your documentation @ttlappalainen . So I connected the shields well and add a resistor jumper at the ends, and I modified the programs in this way :
```
#include <Arduino.h>
#define USE_N2K_CAN USE_N2K_MCP_CAN
#define N2k_SPI_CS_PIN 34    // Pin for SPI select for mcp_can
#define N2k_CAN_INT_PIN 35    // Interrupt pin for mcp_can
#define USE_MCP_CAN_CLOCK_SET 8  // Uncomment this, if your mcp_can shield has 8MHz chrystal
//#define ESP32_CAN_TX_PIN GPIO_NUM_16 // Uncomment this and set right CAN TX pin definition, if you use ESP32 and do not have TX on default IO 16
//#define ESP32_CAN_RX_PIN GPIO_NUM_17 // Uncomment this and set right CAN RX pin definition, if you use ESP32 and do not have RX on default IO 4
//#define NMEA2000_ARDUINO_DUE_CAN_BUS tNMEA2000_due::CANDevice1    // Uncomment this, if you want to use CAN bus 1 instead of 0 for Arduino DUE
#include <NMEA2000_CAN.h>
#include <N2kMessages.h>
#include <N2kMaretron.h>
```
I went to the NMEA2000_CAN.h code to see how to force the use of the CAN shield and i had this line `#define USE_N2K_CAN USE_N2K_MCP_CAN` but now I have this message: CAN device failed to open.

@ttlappalainen , if I understood your documentation and your comment correctly, i necessarily need a chip to use the internal CAN bus of the ESP32 ?

Thank you, Thomas.

ttlappalainen, 2022-03-23:
At beginning of NMEA2000_CAN.h there are instructions:
```
  selected on Arduino development environment. You can still force
  library by adding one of next defines before including NMEA2000_CAN.h:
  #define USE_N2K_CAN 1  // for use with SPI and MCP2515 can bus controller
  #define USE_N2K_CAN 2  // for use with due based CAN
```
You can not use definition `#define USE_N2K_CAN USE_N2K_MCP_CAN` since USE_N2K_MCP_CAN has not been defined at that point. It happens to work since mcp_can is under else after all other has failed.

You get "CAN device failed to open" since ESP can not communicate with your module. Note that AK-Homberger has used just simple tranceiver - not MCP2515. So most probably you have wired your module wrong. Check your ESP32 module pinout where it has SPI pins. And remember define pins right. E.g., #define N2k_SPI_CS_PIN GPIO_NUM_5

thomasnisole, 2022-03-23:
Thank you guys ! So, i rework my #define, and i have that :
```
#include <Arduino.h>
#define USE_N2K_CAN 1
#define N2k_SPI_CS_PIN GPIO_NUM_5    // Pin for SPI select for mcp_can
#define N2k_CAN_INT_PIN GPIO_NUM_35    // Interrupt pin for mcp_can
#define USE_MCP_CAN_CLOCK_SET 8  // Uncomment this, if your mcp_can shield has 8MHz chrystal
#include <NMEA2000_CAN.h>
#include <N2kMessages.h>
#include <N2kMaretron.h>
```
The two programs can open the CAN device, but i always have the  `PGN XXX send failed`.

I'll keep looking why the two ESP32s can't communicate, I come back to you !

Thank you.

ttlappalainen, 2022-03-24:
For sure two ESP32 can communicate. I use them and they are used also on some commercial products. 

You may have problem with interrupt line or bus connection. Read the document I pointed again.

And I still do not understand why to use MCP2515 module instead of ESP32 internal CAN with just tranceiver.

thomasnisole, 2022-03-24:
Hello @ttlappalainen, i use thé shield because it is unused in my garage 😊, but if it still does not work, i buy tranceivers. This [tranceiver ](https://www.ebay.fr/itm/294171691651?_trkparms=ispr%3D1&hash=item447dffca83:g:3z4AAOSwdgtgvIhJ&amdata=enc%3AAQAGAAAA4CikGEIe1AZPNlKIb%2FHsQIJx7QndlSPd90kzesxTW3ZcqQWnlinJMEEZbLiPkSRkXOWLEVjEDJd6fGZfewzPb7mIzBQBFAKuQRjoyBQghr9Vjrjjt3ZgF4UjFEJqq3aU60tCslCL41YVKO8qYbVnsD%2BbGcHLKnU%2Fe9IL365YNZP61R3r24aAmLQBha9dVPYduRnM0X%2BO%2Fl9i4%2BI5hQpBSpZgIOr6aLwCJVd8C4cIQA%2BGJ0dqDYqgxWFh7I0gd9iG8vuErviFyd0Wt4AeaJTJmIUXC6XLH4s6%2BKBn9tfNO%2F0f%7Ctkp%3ABFBMxIDTq_df)  can work with esp32 ?

ttlappalainen, 2022-03-24:
Maybe. I have never used and there is several post of problems with that model. At least you have to remove those resistors on module. Read also carefully from document about difference of unisolated and isolated tranceivers.