ttlappalainen/NMEA2000 #256: how to get data from NMEA2000

Lynda96, 2022-03-07:
I didn't find the definition of NMEA2000.ParseMessages() function ,i dont understand what is it for ?
Can I have more informations please?
the goal of my project is to get wind data from NMEA2000 and read it with nucleoBoard 
Thank you

ttlappalainen, 2022-03-08:
There are e.g., example DataDisplay2 example how you handle received messages with callbacks.

Library has been designed to work also in single thread systems so something must take care of library routines. NMEA2000.ParseMessages() is for that purpose. In single thread system you must call NMEA2000.ParseMessages() in loop and take care that any other task you do in loop does not cause long pause for loop. Max loop time should be kept <50 ms and that only randomly.

And if you have multi thread system, you can have one thread calling NMEA2000.ParseMessages(); naturally you have to the take care data locking during callbacks.

Lynda96, 2022-03-08:
Thanks for your answer 
I have another question please ,the function SendN2kWind() is for sending data to NMEA2000, so if we need receive only data from NMEA2000 , we don't have to write this, isn't it ? 

void SendN2kWind() {
  static unsigned long WindUpdated=millis();
  tN2kMsg N2kMsg;

  if ( WindUpdated+WindUpdatePeriod<millis() ) {
    SetN2kWindSpeed(N2kMsg, 1, ReadWindSpeed(), ReadWindAngle(),N2kWind_Apprent);
    WindUpdated=millis();
    NMEA2000.SendMsg(N2kMsg);
  }
}

ttlappalainen, 2022-03-08:
If you do not sent wind data to the bus you do not need SendN2kWind() function.

You need N2kMessages.h module, if you use any definitions or functions of it. That module contains also parser functions for most common messages. But if you do not use them, you do not need to include it, that is normal c/c++ rule. On the otherhand even if you include it, but do not use anything from it, compiler optimizes everything out and does not unnecessarily increase your code size.