ttlappalainen/NMEA2000 #259: wind speed

Lynda96, 2022-03-14:


Lynda96, 2022-03-14:
Hi,
I have a question about windMonitor example , in this code the goal is to send data to NMEA2000 ,this data is PGN,priority,wind speed, wind angle ,wind apparent , 

void SendN2kWind() {
  static unsigned long WindUpdated=millis();
  tN2kMsg N2kMsg;

  if ( WindUpdated+WindUpdatePeriod<millis() ) {
    SetN2kWindSpeed(N2kMsg, 1, ReadWindSpeed(), ReadWindAngle(),N2kWind_Apprent);
    WindUpdated=millis();
    NMEA2000.SendMsg(N2kMsg);
  }
}
so the question is how you mesure the wind speed and the wind angle cause i find this function 
double ReadWindSpeed() {
  return 10.3; // Read here the wind speed e.g. from analog input
}
which means the wind speed is 10.3  ,it's a value that you wrote it ,it is not mesured with a sensor can you explain me ? ,please

ttlappalainen, 2022-03-14:
The example is for how to use NMEA2000 library. You can write double ReadWindSpeed(); function as you like to get real wind measurement.

Lynda96, 2022-03-14:
Thank you 
So in this case you don't use sensor for reading windspeed ?