ttlappalainen/NMEA2000 #260: connection hardware

Lynda96, 2022-03-14:
Hi,
I have a question about  connection hardware  ,Can I read wind data from NMEA2000 by doing this connection ?,
[image]
 I mean ,can I read data from sensor directly because on DataDispaly2 example the are two Arduino boards connected to NMEA2000 one for sending and the second one for receveing what the one has sent

ttlappalainen, 2022-03-14:
I do not know what you mean with "read data wind from sensor directly". You need microcontroller on both sides. One runs WindMonitor example, which you develop to read sensor. Other runs DataDisplay example, which you modify to show wind data on display.

Lynda96, 2022-03-14:
ok now I understand so what i did in the picture is false , I need to connect the sensor to microcontroller for mesuring wind speed for example  and after that I send data to NMEA2000 and receive this data with other microcontroller if i want isn't it?

ttlappalainen, 2022-03-15:
Yes. You must think NMEA2000 as ethernet - it is a bus to transfer data between devices, which must be cabable to communicate throught the bus.