ttlappalainen/NMEA2000 #262: NMEA2000 Messages

Lynda96, 2022-03-30:
Hi
I am looking for the complete document that determines the data field for each NMEA message as in the picture ,please 
[image]
 Thank you

ttlappalainen, 2022-03-31:
Only way to get complete document is to buy it from NMEA organization. Total price with memebership is somewhere aroung 1500 $. This does not naturally contain any proprietary messages.
From N2kMessages module you can read from functions most common message content.
You can also find information in structured format of data content from https://github.com/canboat/canboat/blob/master/analyzer/pgn.h.

Lynda96, 2022-04-21:
Hi 
thank you for your answer ,i didn't find the format of data that i am loking for on the link that you sent to me 
Actually i wanna decode message PGN 127250 (Vessel Heading)

ttlappalainen, 2022-04-21:
There is ready function ParseN2kPGN127250 (or alias ParseN2kHeading) in N2kMessages module

Lynda96, 2022-04-21:
Ok great ,thank you so much

MarcoVg1980, 2022-10-19:
Hi, 
I am new oh GitHub and sorry if i am not in right group. I would like know if somebody can help me with PGN 127250 HDG

From simulator i read tthe value is HDG=294.660° but i don't know how to find this value... 

Here the data read, I think that the value is in N.2 bytes E3 and C8 but how can to find 294.660 ?

FF E3 C8 00 00 00 00 FC 

Thanks

ttlappalainen, 2022-10-20:
There is ready function ParseN2kPGN127250 (or alias ParseN2kHeading) in N2kMessages module. There is example of using parser on example DataDisplay2 line 241: void Heading(const tN2kMsg &N2kMsg)

MarcoVg1980, 2022-10-20:
Hi Timo,
Thanks for your feedback. Could you e plani me better? Whee. Where can find this software ?

Thanks

ttlappalainen, 2022-10-20:
How about looking from Examples folder under this library? There is also Documents folder and specially I prefer you to read [NMEA2000_library_reference.pdf](https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/NMEA2000_library_reference.pdf) and [Connecting_hardware_to_NMEA2000.pdf](https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/Connecting_hardware_to_NMEA2000.pdf)