ttlappalainen/NMEA2000 #263: Exception/Panic with MCP2515 when NGT-1-USB open with NMEA Reader

mfoolio, 2022-04-05:
Great library, have made many useful devices.

Hardware is d1 mini (ESP-8266EX) with an MCP2515_CAN module from eBay, modified to use 3.3V logic with separate 5V for the bus. Works great in ListenOnly and in ListenAndNode so long as an NGT-1-USB is not active on the CANBUS.

Experience an immediate, persistent crash mode to anything other than ListenOnly (N2km_ListenAndNode) when an NGT-1-USB is on the CANBUS and open with Actisense NMEA Reader. Does not crash if NMEA Reader is not open, able to send/receive PGNs without issue.

Serial Monitor shows:
---
CAN device ready
Start address claim for device 0

Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
---

Is there a known issue with this mix of hardware? Or some kind of race condition with NMEA Reader's bus inquisition?

ESP crash decoder output attached. Appreciate any guidance!

Thanks!

--
[espCrashNGT1USB.txt](https://github.com/ttlappalainen/NMEA2000/files/8414027/espCrashNGT1USB.txt)
-

ttlappalainen, 2022-04-05:
Library does not make much difference, which hw you use. 

But something strange happens in your code. It call tNMEA2000::HandleCommandedAddress and I have never seen that in real life. Only with test tools.

mfoolio, 2022-04-05:
I have reproduced the behaviour with the example sketch "ActisenseListener". Seems to be related to receiving PGN 59904. Maybe something I've fiddled incorrectly in definitions? Any pointers to trace through the stack? 

* Example ActisenseListener sketch, I added the defines for pins on my hardware:
---
#define USE_N2K_CAN 1 // Use MCP2515 SPI method

// VCC(3.3V), GND, SCK=SCL, SDI=SDA, CS=CSB, MISO=SDO
#define BMP_SCK  (7) // SCK 
#define BMP_MISO (9) // SDO = MISO
#define BMP_MOSI (11) // SDI = SDA = MOSI
#define BMP_CS   (12) // CS = CSB

//#define N2k_CAN_INT_PIN D2 // INT pin for MCP_2515
#define N2k_SPI_CS_PIN D8 // CS pin for MCP_2515
#define USE_MCP_CAN_CLOCK_SET 8 // 8Mhz crystal
#define MCP_CAN_RX_BUFFER_SIZE 50 // receive buffer size
---

Then set for plain text output and ListenAndNode mode:

NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); NMEA2000.SetMode(tNMEA2000::N2km_ListenAndNode);

---
Address 1 is the NGT-1
Address 15 is the MCP2515
---

Serial Monitor output:

CAN device ready
Start address claim for device 0
334 : Pri:6 PGN:60928 Source:15 Dest:255 Len:8 Data:1,0,C0,FF,0,82,32,C0
335 : Pri:7 PGN:59904 Source:1 Dest:15 Len:3 Data:16,F0,1
454 : Pri:7 PGN:59904 Source:1 Dest:15 Len:3 Data:14,F0,1
585 : Pri:7 PGN:126993 Source:15 Dest:255 Len:8 Data:60,EA,0,FF,FF,FF,FF,FF
1080 : Pri:7 PGN:59904 Source:1 Dest:15 Len:3 Data:14,F0,1
1081 : Pri:6 PGN:126996 Source:15 Dest:255 Len:134 Data:35,8,9A,2,41,72,64,75,69,6E,6F,20,4E,32,6B,2D,3E,50,43,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,31,2E,30,2E,30,2E,30,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,31,2E,30,2E,30,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,30,30,30,30,30,30,30,31,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,FF,0,1
1267 : Pri:7 PGN:59904 Source:1 Dest:15 Len:3 Data:16,F0,1

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

ttlappalainen, 2022-04-06:
If you do not enable interrupt, rx buffer size does not have any effect. If you enable interrupt, default size is 50.

Responding internally to 59904 is very normal operation in library. I do not expect any problem with that, since then there would be hundreds problematic devices.

I am afraid that your device runs out of stack or the whole memory. Have you checked how much RAM you have after all initialization? And have you checked your free stack size?

mfoolio, 2022-04-06:
Agree 59904 should be very common, hence why I went for a dead simple program to try to isolate the issue. I've tried both with and without interrupt pin enabled, exact same behaviour.

I don't believe we're getting anywhere near SRAM exhaustion. At runtime at the end of setup(), ESP.getFreeHeap() reports: Free Memory: 45104

Compile time reports:

---
Executable segment sizes:
ICACHE : 32768           - flash instruction cache 
IROM   : 266376          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27693   / 32768 - code in IRAM          (IRAM_ATTR, ISRs...) 
DATA   : 1520  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 1212  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25984 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 296801 bytes (28%) of program storage space. Maximum is 1044464 bytes.
Global variables use 28716 bytes (35%) of dynamic memory, leaving 53204 bytes for local variables. Maximum is 81920 bytes.
---

Anything else to check to try to isolate the issue?

ttlappalainen, 2022-04-07:
You have to also run stack max usage test on loop.

void MemoryCheck() {
  static UBaseType_t uxHighWaterMarkSmallest=-1;

  UBaseType_t uxHighWaterMark = uxTaskGetStackHighWaterMark(NULL);
  if ( uxHighWaterMark<uxHighWaterMarkSmallest) {
    uxHighWaterMarkSmallest=uxHighWaterMark;
    Serial.printf("Stack margin: %lu\n",(long unsigned int)uxHighWaterMark);
  }

  static uint32_t SmallestHeapFree=-1;
  uint32_t CurrentHeapFree=ESP.getFreeHeap();
  if ( CurrentHeapFree<SmallestHeapFree ) {
    SmallestHeapFree=CurrentHeapFree;
    Serial.printf("Heap free: %lu\n",(long unsigned int)CurrentHeapFree);
  }
}

mfoolio, 2022-04-07:
OK, on esp8266 the FreeRTOS calls are not available to track the stack. 

ESP.getFreeContStack() in loop() reports the stack high water mark (initialized at 4096). In ListenAndNode it stays at 3228 without NMEA Reader connected. When NMEA Reader connects to the NGT1 stack goes down to 2892, which is the only value reported before crash. I also bumped the stack allocation up to 6144 with no change in behaviour. In ListenOnly mode stack settles in at 3392.

---
StartSetupStack: 3868
CAN device ready
Start address claim for device 0
Free Memory: 45072
EndSetupStack: 3228
stack size 2892
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

ttlappalainen, 2022-04-07:
Sorry, no more ideas. Go to ESP32 or Teensy and use MCP2562 or ISO1050 tranceiver.

mfoolio, 2022-04-07:
Thanks. Reworked with ESP32-S2 in place of the d1 mini and crashing is resolved. For benefit of future readers, esp8266, at least in LOLIN d1 mini v3.0 configuration, has unknown interactions/issues and is not recommended.

closed 2022-04-07 by mfoolio