ttlappalainen/NMEA2000 #268: temperature monitor example with b&g mfd device

ersingencturk, 2022-05-17:
temperature monitor example works well but I've a problem with it. when you turn on the nmea2k network , b&g mfd can't see the device. however if i turn on the device first and then later plug in temperature monitor example it works. also from the devices list i can't list the temperature monitor ProductInformation and deviceinformation. what might be the problem ? is temperature monitor sensor example is fully functional with real devices ? (i never had problems when testing between 2 instances of same code running locally)

ttlappalainen, 2022-05-18:
Example is fully functional. It should not matter in which order you turn on devices. Sounds like your device would not get any request to handle and respond with product information. When you turn on device it will immediately send address claim to the bus.  Enable forwarding on examle and check do you get any data from bus.
  Serial.begin(115200);
  NMEA2000.SetForwardStream(&Serial);
  // If you want to use simple ascii monitor like Arduino Serial Monitor, uncomment next line
  NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show in clear text. Leave uncommented for default Actisense format.
  // If you also want to see all traffic on the bus use N2km_ListenAndNode instead of N2km_NodeOnly below
  NMEA2000.SetMode(tNMEA2000::N2km_ListenAndNode,22);

ttlappalainen, 2022-06-30:
1. Do you use example untouched or do you read somehow real temperature?
2. Does your Rx line work? If your device sends data to the bus, but does not reveive, it will never send its information to the bus. N2k devices works so that  they just informs it self to the bus and then others request more information, if they are interested. So if your Rx does not work, it will not get requests and will not respond to them. Library does that automatically, when hw is right. If you have settings as message above 18.5, you should see all bus data on serial.
3. Also you may that problem, if you loop is blocking too long. You should call NMEA2000.ParseMessages continuously and have maximum occasional delay <50 ms. If you use for temperature reading something like DallasTemperature for sure it will not work.

Broadcasting device info is not proper way to do it and you should not do it. Library handles that on request.

ersingencturk, 2022-07-01:
> 1. Do you use example untouched or do you read somehow real temperature?

yes i read real temperature, will switch untouched too and report back

> 2. Does your Rx line work? If your device sends data to the bus, but does not reveive, it will never send its information to the bus. N2k devices works so that  they just informs it self to the bus [...]

yes it does. interesting bit is if i connect my device to the network after mfd boots everything works. this problem happen if i power network with my device and all others at the same time

> 
> 3. Also you may that problem, if you loop is blocking too long. You should call NMEA2000.ParseMessages continuously and have maximum occasional delay <50 ms. If you use for temperature reading something like DallasTemperature for sure it will not work.

will test this too thank you
> 
> 
> 

> Broadcasting device info is not proper way to do it and you should not do it. Library handles that on request.

i understand but is there a way to force it so i can narrow if is related to problem also ?

ersingencturk, 2022-07-01:
i can confirm example didn't work as well, only things i had to change in the example was adding cs pin , int pin and 8mhz clock setting.

when i say didn't work, it doesn't work with a real mfd design, i can read it's data from another device. not sure why the mfd don't recognize and list under devices my device.

there is a switch in the boat when i turn, it power ups mfd , nmea network and my device .. when i boot all them up it doenst 'work however when the network and mfd is on, if i manully power up only my device it works...

ersingencturk, 2022-07-01:
i made a workaround for the problem with a delay of 60 secs before any nmea library code, now it works with mfd.

still can't see device information that i do register with:

 ```
NMEA2000.SetProductInformation("00000001",` // Manufacturer's Model serial code
                                 100, // Manufacturer's product code
                                 "Simple temp monitor",  // Manufacturer's Model ID
                                 "1.1.0.21 (2016-12-31)",  // Manufacturer's Software version code
                                 "1.1.0.0 (2016-12-31)" // Manufacturer's Model version
                                 );
  // Set device information
  NMEA2000.SetDeviceInformation(112233, // Unique number. Use e.g. Serial number.
                                130, // Device function=Temperature. See codes on http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                75, // Device class=Sensor Communication Interface. See codes on  http://www.nmea.org/Assets/20120726%20nmea%202000%20class%20&%20function%20codes%20v%202.00.pdf
                                2040 // Just choosen free from code list on http://www.nmea.org/Assets/20121020%20nmea%202000%20registration%20list.pdf                               
                               );

```

it looks like this has a problem with my network.

here is a screenshot of how the device looks: (other devices are appear with their names)
[image]

ttlappalainen, 2022-07-02:
Please remove all start delays. Set forwarding as mentioned before. Also set SetForwardSystemMessages and SetForwardOwnMessages. Then turn on your network and send data from your USB.