ttlappalainen/NMEA2000 #272: Starting project....

markingle, 2022-06-29:
What is the best CAN transceiver to purchase for an ESP32?

ttlappalainen, 2022-06-29:
Have you read the document https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/Connecting_hardware_to_NMEA2000.pdf chapter "5 Schematics sample"? You need to decide do you need isolation or not. See chapter "4.2 CAN Tranceiver"

I can not say what is the best. I have been using MCP2562 and ISO1050 and e.g. https://acrossoceansystems.com/ has used them without problems.  Problems with SN65HVD230 may have been caused by users. It is 3.3V only and can not be used with e.g. Teensy 3.2.

markingle, 2022-06-29:
Hi @ttlappalainen !  Yes I have the hardware document printed out.  When google searching for MCP2562 and ISO1050 boards I dont find any.  I would like to use isolated so ISO1050 type transceiver is my target.  Are you saying that I need to design my own pcb with a CAN transceiver?

ttlappalainen, 2022-06-30:
ISO1050 was available also on DIP package, which is easy to put on empty board. I do not situation now - there has been problems with chips availability. 

If you design own board, you can also use ISO1042BDWVR.