ttlappalainen/NMEA2000 #276: Suggestion: Tags and Releases

speters, 2022-07-27:
Thx again for your outstanding libraries!

To make it easier to stay up-to-date, it would be great if you could use _git tag_s to annotate version info and _Github releases_ for your library projects.

Git tags annotated with your version info makes it far easier to spot differences between versions and to check out a specific version.
Currently, checking out a different version involves digging through the libraries.* files history to search for the commit which matches a version, then checking out this commit.

Platformio library registry also relies on these tags (see those invalid "Release Notes" links on https://registry.platformio.org/libraries/ttlappalainen/NMEA2000-library/versions).

Additionally, adding this as a "Github" Release would make it possible to get infos about new versions automatically ("Watch" feature set to notify on "Releases").

ttlappalainen, 2022-07-28:
Thanks for the note. If you have simple instructions what to do, please make short list with dashes. I currently do not have time to start to google how it should be done.

speters, 2022-07-28:
1. Add new version numbers to _libraries.*_ files
2. Commit your code to your git repository (`git commit -am "my commit message`)
3. Create an annotated tag with this new version number `git tag -a v4.17.x -m "new version 4.17.x info message"`
 The _v_ letter in front of your version number is common practice for git tag names (some say it just makes filtering easier)
4. Push your changes and tags to Github `git push --tags`
5. On GitHub repository main page follow [Create a new release](https://github.com/ttlappalainen/NMEA2000/releases/new) link in right column (link only visible if you have adequate rights to do so for this repository).
    - Choose tag with the version number you want to offer as a "Release"
    - Give it a title
    - Add some release notes and/or press "Generate release notes" button to try to get a Changelog from commit messages
    - Click "Publish release" button

6. Optionally add tags for old releases, too
   - List commits where version string was changed `git log -L /version/:library.properties --pretty=oneline`
   - `git tag -a vOLDVERSIONSTRING COMMITHASH ...` for each version

<hr>
PS:
  Being lazy, some shell commands might suffice. But better double-check, as this is not failsafe:

``` shell
git log -L /version/:library.properties --pretty=format:"commit %H" -U1 \
  | sed -ne '/^commit [0-9a-f]*$/p;/+version/p' | tr  '\n' '#' \
  | sed -n -e 's/commit \([a-f0-9]*\) commit [a-f0-9]* +version/ commit \1 +version/g;s/#+version=/ v/g;s/#commit/\ncommit/g;s/#$/\n/g;p' \
  | sed -ne '/ v/{s/commit \([^ ]*\) v\(.*\)/git tag -a v\2 \1 -m "Version \2"/g;p}'  
```

```
git tag -a v4.17.4 c2de650e9abe943da7c87f6a6c03796c4f7b3234 -m "Version 4.17.4"
git tag -a v4.17.3 ad30dced133cf7063b97aaa9ea05e434912e9100 -m "Version 4.17.3"
git tag -a v4.17.2 5a09ab1b0038a0107f9ce2018cc290cf6821ccf5 -m "Version 4.17.2"
git tag -a v4.17.1 a75f6369a25e70b3ef67b5949d63555e3acf3894 -m "Version 4.17.1"
git tag -a v4.16.2 94c9bd5e1256ad8e4b3b8f7bd3f69339b2645b26 -m "Version 4.16.2"
git tag -a v4.16.1 f49a25ae7fd2ab5f5efcf95c78cecf9395c961cb -m "Version 4.16.1"
git tag -a v4.15.1 8ad0a240a0cc0ca8d2b5e2bde43afb660027bdd6 -m "Version 4.15.1"
git tag -a v4.14.2 db22adbb3fec182ecae2ae8bc816378ac43d7fc3 -m "Version 4.14.2"
git tag -a v4.14.1 2d30ceaec8615ef1ad858df9b9be984c44633ef3 -m "Version 4.14.1"
git tag -a v4.13.2 863770b6d60bc9668ed224364285e189a195257c -m "Version 4.13.2"
git tag -a v4.13.1 dcb940735948df26da9bf0f1637ae789ca4ce6e2 -m "Version 4.13.1"
git tag -a v4.12.1 0c64d3ee22ffc33f7e26f27c0f122d8ad3cd88b5 -m "Version 4.12.1"
git tag -a v4.11.1 fd89a54faf9ee6d1f26ea9caec6a7c5a22b59660 -m "Version 4.11.1"
git tag -a v4.10.1 6a70953c9090353dd1d417c86ae1f55a2523ac8e -m "Version 4.10.1"
git tag -a v4.9.1 015a93a8a51f712814e12a3ba157ed1800abe9d3 -m "Version 4.9.1"
git tag -a v4.8.1 8e8971d9f569238b13d70977667ae5054c97055c -m "Version 4.8.1"
git tag -a v4.7.1 226a58052fe7db63a7f2402d20163bafd22f149b -m "Version 4.7.1"
git tag -a v4.7.0 6a91163a6248f91b897606efa5aee798ca95b696 -m "Version 4.7.0"
git tag -a v4.6.0 074069176dd41481691b2cbceedd4d7b98b56d23 -m "Version 4.6.0"
git tag -a v4.5.8 17cede2defb15f5bf5b69f5bcdb550d841cb75cf -m "Version 4.5.8"
git tag -a v4.5.7 14320ee6ebd3d0370b33eb2117cc95e4a2f2ee81 -m "Version 4.5.7"
git tag -a v4.5.4 f6aefe1eb0d45f7d3750a1022b3ebe820a5503ba -m "Version 4.5.4"
git tag -a v4.5.3 c94620afee7386ddff5bb1d3a74615c8193d5e43 -m "Version 4.5.3"
git tag -a v4.5.2 d0d2010c4d7bc956e6cbbac142c3f37ac36358e4 -m "Version 4.5.2"
git tag -a v4.5.0 db494e431be21f6e51fdbd95cb3a2cb390d04306 -m "Version 4.5.0"
git tag -a v4.4.2 7ef4d533abb04587c6436426d31c319ead1fd51d -m "Version 4.4.2"
git tag -a v4.4.1 ecd56bcf57f90b860759626b9a248d9e05e7ba50 -m "Version 4.4.1"
git tag -a v4.4.0 6e3595f775f8faf34d896da7094de7821f9c9cc4 -m "Version 4.4.0"
git tag -a v4.3.5 b5426b3ad177928d5c23018e2ac82c70455e9b95 -m "Version 4.3.5"
git tag -a v4.3.4 10c8e3301af7fe29d04a0613cacbd06f2c2c111d -m "Version 4.3.4"
git tag -a v4.3.2 8f14fb8893add24ab647c5824e415714b7297cdb -m "Version 4.3.2"
git tag -a v4.3.1 6f59e03f562c3f5d534e56a0ebe8a60401d41354 -m "Version 4.3.1"
git tag -a v1.1.141 fee8d977b2e42468f7ca583101889efaba1b44d6 -m "Version 1.1.141"
git tag -a v1.0.1 3d5a30c08012d8172b031382b759a17d217d4234 -m "Version 1.0.1"
git tag -a v1.0.0 3e85eff18fc47def6f248ce0ea4dbe1373d2f472 -m "Version 1.0.0"
```

ttlappalainen, 2022-07-29:
- I have version number on libraries.* files
- I use Windowns and GitHub Desktop. Any idea how to do it there?

speters, 2022-08-18:
Please excuse my delayed reply due to vacation.

Having `git` installed is a [prerequisite for GitHub Desktop](https://docs.github.com/en/desktop/installing-and-configuring-github-desktop/installing-and-authenticating-to-github-desktop/setting-up-github-desktop#part-3-configuring-git).
So `git`command should be available from the [commandline shell of your Windows](https://support.microsoft.com/fi-fi/windows/powershell-korvaa-komentorivin-fdb690cf-876c-d866-2124-21b6fb29a45f).

The commands shown above show a programatical way of looking on which commit the version number has been changed in _library.properties_, then using this commit hash to create a list of git commands for doing an annotated git tag.

As your library is the base for many other products, you better double check manually (check out a commit from the list above, validate, then tag manually).

For starters, or if unsure, just git tag the latest release with the version info and please keep on doing so on each subsequent release. There is no need to do all at once or in a chronological order, as the annotated tag references a specific commit, not a point in time.
For the tags appearing on GitHub, you must _git push --tags_ (there might be a config option in GitHub Desktop to do so on each push) else they will stay local.

ttlappalainen, 2022-10-01:
I still do not get it. I list changes on Readme.adoc file. Version info message would be often very long and not formatted like with adoc. 

The next update will have lot of changes around, so there is no sense for anybody check changes through files - users should trust my changes on the list and test against them.

Would you like to get new version for testing before publish?

ba05, 2025-04-11:
Anyone know how to implement this? Would be nice to have GitHub tags/releases implemented then can get email updates on a new release.