ttlappalainen/NMEA2000 #277: PGN 126464 causing exception

dmarc1234, 2022-08-11:
Hello and thank you for a great library.

I am running the library on an ESP32 dev module and setting up multiple devices (battery, motor and GNSS) to talk to a Raymarine Axiom 7 plotter on my boat. Measurements are collected by the ESP32 and sent to the plotter for display.

The problem I am having is that PGN 126464 is causing a stack smashing exception when I first power-up the plotter which is caused by the length of the 126464 frame. I have enabled the debug and attached and as you can see the 126464 PGN frame length is 1390 which is way above what I believe is the maximum length.

Can anyone throw any light on this problem ?
[PGN126464 exception error.txt](https://github.com/ttlappalainen/NMEA2000/files/9308026/PGN126464.exception.error.txt)

dmarc1234, 2022-08-11:
I believe if you look at the address claims earlier on in the log that 22~25 are the individual devices setup on the ESP32 i.e. battery, motor etc. so the 126464 is in response to the 59904 from the plotter at address 0.

The back-trace from the exception looks like this:
Decoding stack results
0x4008c630: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008c861: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400e75bc: __stack_chk_fail at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/stack_check.c line 36
0x400d6a62: tNMEA2000::SendTxPGNList(unsigned char, int, bool) at ../../driver/src/NMEA2000.cpp line 1783
0x400d6ea1: tNMEA2000::RespondISORequest(tN2kMsg const&, unsigned long, int) at ../../driver/src/NMEA2000.cpp line 1930
0x400d6f65: tNMEA2000::HandleISORequest(tN2kMsg const&) at ../../driver/src/NMEA2000.c

ttlappalainen, 2022-08-11:
Building Tx/Rx list does not have checking for buffer limitation. This works normally, if there are too many Tx/Rx messages defined.
- Check that you buffer provided for ExtendTransmitMessages does contain 0 as last item. The last resonable PGN is 14,F2,1 = 127508. After those there is no sense on PGN numbers.
- Check that your buffer provided for ExtendTransmitMessages does not contain too many PGNs.

dmarc1234, 2022-08-12:
This is what I setup for the messages:

ttlappalainen, 2022-08-12:
You should have e.g.,
const unsigned long BatteryMonitorTransmitMessages[] PROGMEM={127504L,127506L,127507L,127508L,0};

ttlappalainen, 2022-08-12:
on
const unsigned long MotionTransmitMessages[] PROGMEM={127257L,0};
there are unnecessary 0:s

dmarc1234, 2022-08-12:
Looks like we have got this working now by making sure that there are 0 values at the end of each const allocation.

Thank you for your help.

closed 2022-08-12 by dmarc1234