ttlappalainen/NMEA2000 #278: Simple need to save data to global variables

beefy23, 2022-08-17:
First of all thanks for all the hard work you must have put into developing these libraries.

I have an Evinrude Etec engine that I wish to read engine parameters from. I simply want to get each individual engine parameter and save it to a global variable so that I can send it via serial to a custom programmable display.

Using the Data Display 2 and taking this function as an example:

void EngineRapid(const tN2kMsg &N2kMsg) 
{
    unsigned char EngineInstance;
    double EngineSpeed;
       
    if (ParseN2kEngineParamRapid(N2kMsg,EngineInstance,EngineSpeed) ) 
	{
      PrintLabelValWithConversionCheckUnDef("Engine rapid params: ",EngineInstance,0,true);
      PrintLabelValWithConversionCheckUnDef("  RPM: ",EngineSpeed,0,true);     
    } 
	
	else 
	{
      OutputStream->print("Failed to parse PGN: "); OutputStream->println(N2kMsg.PGN);
    }
}

Using only the engine speed parameter as an example, could I simply change the above to:

void EngineRapid(const tN2kMsg &N2kMsg) 
{
    unsigned char EngineInstance;
    double EngineSpeed;
       
    if (ParseN2kEngineParamRapid(N2kMsg,EngineInstance,EngineSpeed) ) 
	{
     myGlobalVariable = EngineSpeed:  
    } 
	
	else 
	{
      // ........................
    }
}

ttlappalainen, 2022-08-17:
Yes, this is simple way and works if you have one engine and one device providing RPM information PGN (most cases).

beefy23, 2022-08-17:
Thanks very much Timo.

The reason I'm wanting to do this rather than simply sending basic NMEA 2000 data, is so that I can do various checks and computations on the data, and give alarms/warnings of anything unusual.

I'm quite excited about having such engine monitoring on my boat, and it will be thanks to your great libraries, so once again thanks very much.

beefy23, 2023-04-15:
Hi Timo,

I've only programmed in C so I'm a little confused with how to achieve the following:

In Data Display 2 example, regarding this function:

void EngineDynamicParameters(const tN2kMsg &N2kMsg) 
{
    unsigned char EngineInstance;
    double EngineOilPress;
    double EngineOilTemp;
    double EngineCoolantTemp;
    double AltenatorVoltage;
    double FuelRate;
    double EngineHours;
    double EngineCoolantPress;
    double EngineFuelPress; 
    int8_t EngineLoad;
    int8_t EngineTorque;
    **tN2kEngineDiscreteStatus1 Status1;
    tN2kEngineDiscreteStatus2 Status2;**

I want to put "Status1" and "Status2" into my own global variables. 

For something like AlternatorVoltage I would simply create my own double variable like "double beefy23_variable" and write "beefy23_variable = AlternatorVoltage".

However because Status1 and Status2 are not declared variables like all the others above them, I don't know how to do that.
I know that Status1 has 16 bits and Status2 has 8 bits, and each bit represents a true/false engine condition.
Once I can get Status1 and Status2 into my own variables I can mask the individual bits to find their state.

Thanks,

Keith

ttlappalainen, 2023-04-15:
What do you mean that "Status1 and Status2 are not declared variables like all the others "? They are variables of type tN2kEngineDiscreteStatus1 and tN2kEngineDiscreteStatus2.

Like you have global beefy23_variable you can also declare globals:
tN2kEngineDiscreteStatus1 MyStatus1;
tN2kEngineDiscreteStatus2 MyStatus2;
and then do setting
MyStatus1=Status1;
MyStatus2=Status2;

Follow the declaration for tN2kEngineDiscreteStatus1 (synonyme for tN2kDD206) and you will find that it is union of uint16_t or bit sruct definition. So you can handle them as uint16_t or use e.g., if ( Status1.OverTemperature ) {...}

beefy23, 2023-04-15:
Sorry Timo, I hope you can excuse my ignorance.

My limit programming experience is with C (with AVR microcontroller) only, not C++, and to be honest I do not yet understand your statement:
**_"They are variables of type tN2kEngineDiscreteStatus1 and tN2kEngineDiscreteStatus2"_**

I only understand the basics in programming and when it comes to variables I know the typical ones like uint8_t, uint16_t, char, bool, int, etc, etc. However I don't know what the a "variable of type tN2kEngineDiscreteStatus1" means.

I will look for the declaration and try and get an understanding of it. 

However, I think you've given me enough information for me to accomplish what I want to do.

Again, sorry for my lack of knowledge and I appreciate your help.

Keith

ttlappalainen, 2023-04-15:
uint8_t is neither primary type. It is normally declared with typedef as unsigned char. uint16_t is declared in esp32 environment unsigned short. Both declaratons depends of environment. So  tN2kEngineDiscreteStatus1 is not more than new type declaration. In C++ it can be done with literal "using" instead of old C style typedef. I think that using is more logic since you declare type like setting:
`using tN2kEngineDiscreteStatus1 = tN2kDD206;`
Easier to read: tN2kEngineDiscreteStatus1 is same as tN2kDD206

beefy23, 2023-04-15:
Thanks very much for the explanation Timo. I have a lot to learn.