ttlappalainen/NMEA2000 #281: Project on a twin motor boat nmea 2000

MaxESP, 2022-08-25:
Dear M Timo

Thanks again for  your Work that huge . I am a domotic developper with all integrate solution with hard and software , and i would be glade to have your advise about my future project.

I project to buy a cruiser boat with twin volvo penta motor  witch have nmea 2000  output via EVC Volvo Penta Electronic Vessel Control

Before i dive in it does your nmea library  lib will read the 2 engines  data to serial like your example DataDisplay2 example
because PGN header are not the same 

[image]

thanks a lot

ttlappalainen, 2022-08-29:
The library can read anything from NMEA 2000 bus. Engine PGNs are 127488 and 127489, where engines are separated by first field - "Engine instance".

MaxESP, 2022-09-07:
Thank you very much

I am working on both side electronic and display .
i have make an electronic compass fully tilt compensed 

[image]

ttlappalainen, 2022-09-07:
What display/driver/library you use?

I have used for NaN values background color yellow to make it remarkable. Also I turn indicator some degree outside of minimum value.

MaxESP, 2022-09-07:
It's an amazing lib https://github.com/HanSolo/SteelSeries-Canvas.

I have allready fix it in a electron nodejs app with 3 serial port read data.
The unique problem is to fix the timming in reading port in javascript vs nmea 2000 incoming timming  because i will have a lot of port 
minimum:
1 gps
2 compass + imu
3 motors nmea from volvo.

I have made some performance test need a windows plateform pc with  good perf . My problem now is to find a pc plotter , i have find one good is i boating but iven if i have buy the spain charts their time validity is not clear.

To go faster can you tell me more about transceivers and nmea2000 .(sorry if some stupid questions)

1) the nmea2000 bus voltage  high and low voltage are :
CAN Bus uses a Drive Voltage: High; 2.75v to 4.5 volts, Low; 0.5 to 2.25 volts, Differential 1.5v to 3.0 volts.

2) the 12 volt in the nme2000 bus is for provinding DC supply to devices that they don't have a dc power?

3)  this module https://www.az-delivery.de/en/products/mcp2515-can-bus-modul
can be used as a reveiver or transmiter data on the nme2000 bus.?

4) if you want some help for your display i can make you the interface depending of your config

ttlappalainen, 2022-09-07:
Have you checked OpenCPN? It is very good nav. sw and now they have good vector charts available by oCharts.

On N2k bus you need CAN controller and tranceiver. If you going to use PC, better to buy Actisense NGT-1. If you are going to use some MCU I prefer to go ESP32 or Teensy 3.x, which has build in CAN controller. Read my document https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/Connecting_hardware_to_NMEA2000.pdf. I also prefer to use isolated tranceiver.

MaxESP, 2022-09-14:
Thanks for your help the project go well

1 have a question about can view why your batterie sketch work with actisence reader soft  and not canview
[image]
I have quite finish my heading + imu sensor with auto calibrate function

ttlappalainen, 2022-09-14:
What is canview? What do you mean that it does not work with canview?

MaxESP, 2022-09-15:
Sorry https://www.yachtd.com/products/can_view.html

ttlappalainen, 2022-09-16:
Do you mean forwarded messages? If you enable message forwarding on library, it send them out to serial in Actisence format. I am not sure does can view support that format you should it be switched to that. YDNU has also sends Actisense format, if enabled.

MaxESP, 2022-09-16:
So if i understand well your PNG library  is native nmea 2000 them output serial format depend of back end decoder in our case Actisence. that's why Canview log can't decode out serial .

Here is the nme2000 central commande i am working on 

[image]

ttlappalainen, 2022-09-16:
Yes, library uses native CAN connection. There is no standard way to provide CAN or NMEA2000 data in serial. I use Actisense format.

I wonder why do you use MCP2515 modules? ESP32 has internal CAN controller and requires only tranceiver e.g., ISO1050. If you would like to have 2 NMEA2000 connections, you can do it with one ESP32 by using internal CAN controller for one and MCP2515 for other. Depending of your system you may need to have both isolated. In this case you can not use ready MCP2515 modules. Instead you have to build your MCP2515 by your own and use isolated tranceiver.

Your current setup does not have isolation.

ttlappalainen, 2022-09-19:
You need to have bot tranceiver and power isolted. Other option is to use isolated tranceivers and have own dc-dc converter for them on bus side and then own dc-dc converter for other parts - so you should have 3 DC-DC converters. By just using isolated converter does not do isolation.

ttlappalainen, 2022-09-19:
You must use ISO1050 tranceivers. Not TJA1050, which is not isolated.

MaxESP, 2022-09-19:
But if i   implement B0505s-1w Conversor Dc Dc Isolador

MaxESP, 2022-09-19:
ok i understand the iso must be between can and tx rx side

MaxESP, 2022-09-21:
here my design mcp2515 + iso1050
[image]

ttlappalainen, 2022-09-21:
VCC1 and VCC2 max voltage is 6V, so you need regulator to drop bus voltage to 5V for bus side. On other side you need also use 5V, since MCP2515 is 5V only.

MaxESP, 2022-09-21:
yes you are right..... the regulator is on the the first pcb with the power isolator  . I have to test the direct esp32 can controler too.

MaxESP, 2022-09-21:
What do you think is better put a stand alone supply for esp32 or convert the 12 volt from nmea and convert dcdc with float isolation

ttlappalainen, 2022-09-21:
If you power it from bus, you need isolated DC-DC converter. It is OK. But there is no clear answer. If you just make bus to bus converter, it may be easiest to power it from one bus. Then it can locate anywhere, where you have busses. One problem is that avoiding all EMI from your device to bus is not that easy and specilly because you have there DC-DC converter taking power from bus. Designing good PCB to protect from EMI needs training and one problem is that you do not have devices to test it. If you just use linear regulator for powering tranceivers and DC-DC on other side you are more safe.

MaxESP, 2022-12-26:
Hello Timo I wish you a happy new year and give you some new about my project

1 i have implement serial read from gps nmea 0183 and display data on main screen ok good flow
2 i have implement  aviation grade compass with tilt  ok good flow
3 before diving in your lib i have made some test on mcp2515 with esp 32 with this lib https://github.com/autowp/arduino-mcp2515 test seems to be good . i was scrared because i have read on the web that mcp2515 is not compatible with esp32.
i have tried /[Seeed_Arduino_CAN](https://github.com/Seeed-Studio/Seeed_Arduino_CAN)
is working as well
4 my final CAN NMEA 2000 shield look like this 

[image]
5 About my config  mcp2515 with my esp32  I HAVE TO USE   #define USE_N2K_CAN 1   ? Tell me if wrong
6 improve front app 
[image]

ttlappalainen, 2022-12-26:
1. Why you do not use ESP32 internal CAN. That requires only tranceiver chip.
2. In library documents it should clearly define that with mcp you should use version under my git https://github.com/ttlappalainen/CAN_BUS_Shield . At least other realeases of earlier CAN libraries did not handle NMEA2000 fast packet sending properly. I have not checked the latest situation.

MaxESP, 2022-12-27:
Hello Timo  thanks again 
the couple mcp2515 + esp32 work good no problem for the first test 
i have follow the chat https://github.com/ttlappalainen/NMEA2000_mcp/issues/5

I am currious in witch field do you have made changes on https://github.com/Seeed-Studio/Seeed_Arduino_CAN to make mcp can work
[image]

MaxESP, 2022-12-27:
If there is some improvement to work on can bus shield tell me

ttlappalainen, 2022-12-27:
You can copy both and compare.

The problem is that other libraries mixes fastpacket frames on sending time to time. You will not see problem if you only receive data. Except your device may time to time drop from other devices list, if they do not get respond properly for request.

MaxESP, 2022-12-28:
where do  you set clock 

#define USE_MCP_CAN_CLOCK_SET 16

don't work

MaxESP, 2022-12-28:
I am really confused about clock set 
In the couple scketch message send and display 2 when i try to change clock there no effect even if i change to 16 mhz my mcp2515 are 8 mhz they continue to work.

I have 2 mcp2515 with 16 Mhz and i wanted to test them but not working strange
i have try in the header
#define USE_MCP_CAN_CLOCK_SET 16 and #define MCP_CAN_CLOCK_SET MCP_16MHz

no effect

ttlappalainen, 2022-12-28:
#define USE_MCP_CAN_CLOCK_SET 16 is default setting and so does not actually have any effect. If you use #define USE_MCP_CAN_CLOCK_SET 8 it has effect and stops your devices working.

you must have definition before include. E.g.,
#define USE_MCP_CAN_CLOCK_SET 8
#include <NMEA2000_CAN.h>

MaxESP, 2022-12-28:
thank a lot i have a curious issue with 16mhz cristal they seems to be defective because i have replace by 8 mhz on my definive board photo up and it's work 
[image]
Would you be at the boat show 2023 dusseldorf I would be please to invite you for drink or diner

MaxESP, 2022-12-28:
your comment help me a lot isolation work good too the iso1050 is feeded with 12 volt with the step down and the esp32 by pc .

i have made a small mistake on the 120 resistor jumper is not well placed but it ok

MaxESP, 2022-12-28:
I going to test my board to send

ttlappalainen, 2022-12-28:
I did always use 16MHz chrystal with MCP2515.

Normally there is never terminator resistor on NMEA2000 devices. Terminators are on main bus only.

MaxESP, 2022-12-29:
yes you are right it's just for debug