ttlappalainen/NMEA2000 #287: CAN ID Confusion

matthias-mw, 2022-09-11:
Hi Timo,

first of all thanks for the great work. I am just getting started with the lib and question about the canID's.

I am using an ESP32 board with an MCHP2551 transceiver and an Microchip CAN Analyser (CANdevStudio and SavvyCAN) to analyze the CAN communication. I am working an linux with socketcan.

I just used your example "TemperatureMonitor" and it worked just out of the box. Excellent Work.

While analyzing the CAN-BUS I find three Messages:
` can0  15FD0816   [8]  01 01 04 5B B3 FF FF FF`
` can0  15FD0716   [8]  01 C4 BF B3 FF 7F FF FF`
` can0  15FD0616   [8]  01 C1 70 FF FF FF FF FF`

so I just grabed a PGN dbc-file
`:~$ candump can0 | cantools decode pgns_new.dbc
  can0  15FD0816   [8]  01 01 04 3B C9 FF FF FF :: Unknown frame id 368904214 (0x15fd0816)`

the readout from the esp32 shows 
`6995513 : Pri:5 PGN:130312 Source:22 Dest:255 Len:8 Data:1,1,4,13,BF,FF,FF,FF`
`6995513 : Pri:5 PGN:130311 Source:22 Dest:255 Len:8 Data:1,C4,77,BF,FF,7F,FF,FF`
`6995524 : Pri:5 PGN:130310 Source:22 Dest:255 Len:8 Data:1,C1,70,FF,FF,FF,FF,FF`

In the dbc-File the PGN 130311 has an ID: 2583496702
`BO_ 2583496702 PGN_130311_environmentalParamete: 8 Vector__XXX
 SG_ atmosphericPressure : 48|16@1+ (1,0) [0|0] "hPa" Vector__XXX
 SG_ humidity : 32|16@1+ (0.004,0) [0|0] "%" Vector__XXX
 SG_ temperature : 16|16@1+ (0.01,0) [0|0] "K" Vector__XXX
 SG_ humiditySource : 14|2@1+ (1,0) [0|0] "" Vector__XXX
 SG_ temperatureSource : 8|6@1+ (1,0) [0|0] "" Vector__XXX
 SG_ sid : 0|8@1+ (1,0) [0|0] "" Vector__XXX
`

To my knowledge die ID 2583496702 is an extended Address (as expected for NMEA2000) aud die ESP32 sends out an Std CANID 368904214 (0x15FD0816).

Do I have to specify something to send out ext. CAN IDs

Thanks

Matthias

ttlappalainen, 2022-09-11:
You can not analyze it in that way. CAN id is not constant since it includes also source, destination and priority. Your dbc definition id 99FD07FE defines PGN 130311, pri:6, source:254, dest:255. CAN id 15FD0716 sent by example defines PGN:130311, pri:5, source:22, dest:255.

I prefer to purchase my NMEA Simulator box for analyzing NMEA2000 bus.

matthias-mw, 2022-09-13:
Hi Timo,

understood. Thanks for the support. Then I can now dig deeper into the NMEA2000 Standard and start the project...

matthias-mw, 2022-10-01:
Hi Timo,

I just worked a bit with the library and it is really working well. While trying to understand it better I do more comments for my self inside the code. While starting that I was wondering if  you would be interested to have have a full doxygen docu provided for your library. So I would spend some more time in the next weeks an provide a full documentation....
Please let me know if you are interested

Matthias

ttlappalainen, 2022-10-01:
Sure it is working well - it has been used in several commercial products.

Sure I am interested, but it is damn big job for both. You need to write and I have to check. But I prefer to do it through email and so that I do final PR. You will get your name somewhere.

matthias-mw, 2022-10-01:
I will start working on it in the next weeks an over the winter. Do you have an EMail for me?

ttlappalainen, 2022-10-01:
Here are instructions https://webapps.stackexchange.com/questions/29197/any-way-to-contact-a-user-on-github

ttlappalainen, 2022-10-01:
Did you find it. Yours is not yet available, since you have not done some events.

Have you read the documents https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/Connecting_hardware_to_NMEA2000.pdf and https://github.com/ttlappalainen/NMEA2000/blob/master/Documents/NMEA2000_library_reference.pdf ?

Should also think how those could be merged. There should be information on generated document, which is not in any .h or .cpp files also with pictures. I used doxygen several years ago, but then I had problems to generate full document so I came up simple solution just to use Word. I have not followed how doxygen has developed since those times - maybe it now has way to include other things.