ttlappalainen/NMEA2000 #289: Bilge Pump Monitor

MrMount11, 2022-10-01:
Hi Timo,

Your work on the N2k libraries is awesome... Thankyou.

I am working on a node to monitor bilge pump status and I would like to use similar functionality like Mareton proprietary PGN 130836 "Switch Status Counter" & 130837 "Switch Status Timer".  Can support for these be added to the library or do I have to create my own proprietary PGN's? 

Thanks, Rich

ttlappalainen, 2022-10-01:
Since they are Maretron proprietary, they can be added to N2kMaretron module, if you have full description with all field data types, lenghts and multipliers for them.

MrMount11, 2022-10-01:
Great.. I will research the field definitions and get back to you.

Thanks, Rich

MrMount11, 2022-10-02:
Hi Timo,  

I think I have found the full definitions of the two PGN's that I would like to add.  I found them in Canboat,  another great work!.  I will forward them shortly,  however I was wondering if I can add them myself?  In accordance with your reference manual, to do this I would fork the repository?  If all goes well can you then merge this back in to the main branch?  Or should I do it another way?   I am decent with C++ but not very familiar with GitHub. 

Thanks, Rich

ttlappalainen, 2022-10-03:
If you do not have any Maretron device to prove messages, do it just for your own.

MrMount11, 2022-10-03:
Yes, that makes sense.   Thanks for all your help, Rich

closed 2022-10-03 by MrMount11

virtuvas, 2023-02-05:
@MrMount11  did you actually implement the Maretron Switch status PGNs? I'm about to implement a bilge monitor and I do have a DSM250 to test it. If you have anything you'd like to share, let me know.

cheers

V.

ttlappalainen, 2023-02-06:
If you can make simple board with example ActisenseListenerSender, you can test it with my NMEA Simulator. It can now simulate both control and "relays" side. Works fine with e.g., Yacht Devices relay boxes.

MrMount11, 2023-02-06:
Hi Vassilis,

Yes, I did complete my project but I did not use the Mareton PGN's.  At
Timo's suggestion, since my device will only be installed on my boat, I
just used some other unknown or other proprietary PGN's which will likely
never conflict on my network.

Rich