ttlappalainen/NMEA2000 #291: AIS N2K Msg problem with Class A ships

usauerbrey, 2022-10-25:
Hi Timo

As you know, I am working with you library since several years and I am very happy with it. Thanks again for your great work 😉

I have the following situation/problem. On my boat I have all instuments, MFDs, etc connected with N2K, e.g. Raymarine Seatalk NG.
All but not the AIS transmitter. It works with NMEA0183 and the same is true for my Windows based Navigation program.

I am using a Teensy 3.2 as N2K-NMEA0183-N2K converter, which works fine besides the following:

The conversion of AIS NMEA0183 msgs into N2K AIS msg works perfectly when I take a look to your NMEA Simulator or the 
Actisense NMEAReader. But on the Raymarine MFD (Axiom) and on the Raymarine i70s Displays Class A ships are NOT shown at all, Class B ships work fine.

Unfortunately I do not have an N2K AIS device, so I cannot check if the Class A N2K msg are correct or what Raymarine does not like.

Do you have any idea, what I could do?

Best regards
Uwe

ttlappalainen, 2022-10-25:
Or maybe sent PGNs are not problems, but PGNs not send and required for Class A.

usauerbrey, 2022-10-25:
The converters reacts on msg 1,2,3, and 5 with
129038	AIS Class A Position Report							AIS VHF message 1, 2, 3
129794	AIS Class A Static and Voyage Rel Data					AIS VHF message 5
I do not receive other Class A related msgs on NMEA0183

ttlappalainen, 2022-10-26:
[image]
[image]

Above are pictures from real data produced with NMEA2000 AIS Receiver.

How about field 12 on 129038 and field 20 129794?

usauerbrey, 2022-10-26:
Bingo, the setting of the transceiver information in 129038 was wrong. The problem is in your lib:
void SetN2kPGN129038(tN2kMsg &N2kMsg, uint8_t MessageID, tN2kAISRepeat Repeat, uint32_t UserID,
                        double Latitude, double Longitude, bool Accuracy, bool RAIM, uint8_t Seconds,
                        double COG, double SOG, double Heading, double ROT, tN2kAISNavStatus NavStatus)
{
    N2kMsg.SetPGN(129038L);
    N2kMsg.Priority=4;
...
    N2kMsg.AddByte(0xff); // AIS transceiver information (5 bits)

Will you change it or should I do a change request for N2kMessages.cpp and for N2kMessages.h

Best Regards Uwe

ttlappalainen, 2022-10-26:
Try with this. There is new version of SetN2kPGN129038, which takes tN2kAISTransceiverInformation as parameter.
[src.zip](https://github.com/ttlappalainen/NMEA2000/files/9869601/src.zip)

ttlappalainen, 2022-10-26:
I'll make update. On update I'll remove duplicate and typo definition tN2kAISTranceiverInfo. It will be replaced with tN2kAISTransceiverInformation so there will be small compatibility problem, if you have used AIS messages using tN2kAISTranceiverInfo.

usauerbrey, 2022-10-26:
That's fine with me, I will change my code anyhow.
Thanks again

ttlappalainen, 2022-10-26:
Could you test changes? Then I do not need to setup test bed for AIS messages.

usauerbrey, 2022-10-26:
What to test exactly? The above problem with not showing the Class A ships is solve with the files you sent me

usauerbrey, 2022-10-26:
the rest, Class B etc is still working fine

ttlappalainen, 2022-10-26:
I mean I make change mentioned above and add zip here. Would you test it for me? It effects some AIS functions.

usauerbrey, 2022-10-26:
In the src.zip is just N2kMessages.cpp and N2kMessages.h. Did you change N2kTypes.h too?

ttlappalainen, 2022-10-26:
Not yet. I wrote "I'll make update. On update I'll remove..." and then thought if you could test it before update so that I do not need to setup test system.

usauerbrey, 2022-10-26:
I checked my converter. I had to change N2kaisti_Channel_A_VDL_reception to N2kaischannel_A_VDL_reception at some places. 
Everything seem to work fine, I found no problems ;-)

ttlappalainen, 2022-10-27:
Here is now updated files for testing. I decided to remove compatibility inlines for 129038, since message apparently does not work without tranceiver information in all cases so it is risk not to have it and users should be aware of it. I thought to use deprecated, but people often does not care of warnings.

Have you downloaded newest library version (below is not yet included)? It has some critical fixes in core.

[src.zip](https://github.com/ttlappalainen/NMEA2000/files/9875609/src.zip)

usauerbrey, 2022-10-27:
"I decided to remove compatibility inlines for 129038, since message apparently does not work without tranceiver information in all cases so it is risk not to have it and users should be aware of it."
I agree, it seems to be the better way.

Have you downloaded newest library version: yes it did

I checked my converter again. Everything works fine, I found no problems, thanks

ttlappalainen, 2022-10-27:
How about "communication state"? Do you get it from NMEA0183 and should it be added? It could be optional at end of parameter list.

usauerbrey, 2022-10-27:
Currently it is set in the NMEA2000 lib to 0xffff as you know ;-)
I have no idea, from which field in NMEA0183 I could take it:
Message 1,2,3 - Position Report
!AIVDM,1,1,,A,13u?etPv2;0n:dDPwUM1U1Cb069D,0*24
01	Message ID	1	
02	Repeat indicator	0	No repeat (default)
03	User ID (MMSI)	265547250	
04	Navigational status	0	Under way using engine
05	Rate of turn ROTAIS	-2.9	
06	SOG	13.9	
07	Position accuracy	0	Low (> 10 m) (default)
08	Longitude	11.8329767	
09	Latitude	57.6603533	
10	COG	40.4	
11	True heading	41	
12	Time stamp	53	
13	Special manoeuvre indicator	0	
14	Spare	0	
15	RAIM-flag	0	RAIM not in use (default)

usauerbrey, 2022-10-27:
But off course you should add it  optional at end of parameter list. So it can be set, if needed.
BTW the same is true for AISTransceiverInformation, there is also no corresponding field in NMEA0183

ttlappalainen, 2022-10-27:
I'll do it later. It is updated now. Please if you have time update your fork and your code and do hopefully final test.

Is your NMEA0183 parsers published somewhere?

usauerbrey, 2022-10-27:
For AIS parsing I am using a fork from https://github.com/KimBP/AIS. My fork can be found at https://github.com/usauerbrey/AIS

usauerbrey, 2022-10-28:
I did my tests again, everything works fine. I have to leave my boat today for the winter, it is on the hard right now. So I cannot do any further test until April ;-(
I hope I could help you a little, thanks again for your great libraries.
Best Regards Uwe

ttlappalainen, 2022-10-28:
Thanks for these tests. Helped me to avoid setting up test environment.

Where you are located? We have to take boats up for winter due to ice. I took Feelings up on 13.10. Normally I put she back to water around end of may.

usauerbrey, 2022-10-28:
My boat is in the med, in the Ionian Sea, Greece. The weather is still perfect, but we have to start the winter season because of the flights to Germany, which ends end of Oktober ;-(