ttlappalainen/NMEA2000 #292: List of confirmed working CAN transceivers?

anhorbc, 2022-10-28:
Hi!
Is there some kind of collection of confirmed working CAN transceivers with the NMEA2000 library?
Here is my findings with ESP32:
Texas Instruments SN65HVD230 - Marked "VP230" - Doesn't work, at least not for me.
Texas Instruments SN65HVD231QD- Marked "HV231Q" -  Works. 
Texas Instruments ISO1050 - Marked "ISO1050" - Works. (If you can get hold of one)
Microchip MCP2551 - Marked "MCP2551I" - No luck. 
Microchip MCP2562E - Marked "MCP2562E" - Works.
Sitcores SIT1050 - Marked "SIT1050" - No luck so far. Supposed to be a direct replaceable item for TJA1050.

Not sure what makes some chip to work and some not. To my understanding there is no driver for the actual transceiver?

/A

ttlappalainen, 2022-10-29:
No list. In principle all CAN tranceivers compatible with ESP32 should work. Note that e.g., MCP2551 is 5V chip and will not work as 3.3V. Also if it is powered 5V, ESP32 is 5V tolerant, so it can not be used without level shifter.

Also SN65HVD230 should work, but RS pin must be pulled down.

anhorbc, 2022-10-29:
Do you know of a way to debug controller<->transceiver communication? I have a prefab ESP32 board with onboard SIT1050 transceiver, so I am a bit confused why I can't get it to work with NMEA2000 library.

ttlappalainen, 2022-10-29:
With oscilloscope.

Have you configured Tx and Rx pins on ESP32 properly? See. e.g., https://github.com/ttlappalainen/NMEA2000/blob/master/Examples/TemperatureMonitor/TemperatureMonitor.ino line 6 and 7

aldas, 2023-02-02:
Maybe it helps someone in the future. I can add that M5stack Atom Canbus kit (CA-IS3050G transceiver)    https://shop.m5stack.com/products/atom-canbus-kit-ca-is3050g  works fine, just change the Rx/Tx pins and you are good to go.