ttlappalainen/NMEA2000 #293: N2kMessagesEnumToStr.h issues

agk1190, 2022-11-06:
I am using this library in a project and have noticed that in the `N2kMessagesEnumToStr` header file, there are a couple of potential errors.
My code will not compile without making the following changes...

Firstly, on line 104 of `N2kMessagesEnumToStr.h`, the `MakeN2kEnumTypeToStrFunc` function under the `tN2kMagneticVariationStrs` array has the argument `tN2kSpeedWaterReferenceTypeStrs`. Is this possibly a copy and paste error from the 2 lines above it (first 2 lines in block below)? Should it instead be `tN2kMagneticVariationStrs` which would then match the format of all the other code in the file?

```
// -------- original --------
const char* tN2kSpeedWaterReferenceTypeStrs[] = {"Paddle wheel","Pitot tube","Doppler","Correlation (ultra sound)","Electro Magnetic"};
MakeN2kEnumTypeToStrFunc(tN2kSpeedWaterReferenceType,tN2kSpeedWaterReferenceTypeStrs);

const char* tN2kMagneticVariationStrs[] = { "Manual","Automatic Chart","Automatic Table","Automatic Calculation","WMM 2000","WMM 2005","WMM 2010","WMM 2015","WMM 2020" };
MakeN2kEnumTypeToStrFunc(tN2kMagneticVariation, tN2kSpeedWaterReferenceTypeStrs);

// -------- fixed --------
const char* tN2kSpeedWaterReferenceTypeStrs[] = {"Paddle wheel","Pitot tube","Doppler","Correlation (ultra sound)","Electro Magnetic"};
MakeN2kEnumTypeToStrFunc(tN2kSpeedWaterReferenceType,tN2kSpeedWaterReferenceTypeStrs);

const char* tN2kMagneticVariationStrs[] = { "Manual","Automatic Chart","Automatic Table","Automatic Calculation","WMM 2000","WMM 2005","WMM 2010","WMM 2015","WMM 2020" };
MakeN2kEnumTypeToStrFunc(tN2kMagneticVariation, tN2kMagneticVariationStrs);
```

Secondly, when I try to use the `tN2kXTEMode` type from `N2kTypes.h`, I get 

```
.pio/libdeps/heltec_wifi_lora_32_V2/NMEA2000-library/src/N2kMessagesEnumToStr.h: In instantiation of 'void PrintN2kEnumType(T, Stream*, bool) [with T = tN2kXTEMode]':
src/decodeN2K.cpp:415:99:   required from here
.pio/libdeps/heltec_wifi_lora_32_V2/NMEA2000-library/src/N2kMessagesEnumToStr.h:38:35: error: no matching function for call to 'N2kEnumTypeToStr(tN2kXTEMode&)'
   const char *str=N2kEnumTypeToStr(a);
                   ~~~~~~~~~~~~~~~~^~~
```
 error unless I add the following code to the end of `N2kMessagesEnumToStr.h`.

```
const char* tN2kXTEModeStrs[] = { "Autonomous", "Differential", "Estimated", "Simulator" };
MakeN2kEnumTypeToStrFunc(tN2kXTEMode, tN2kXTEModeStrs);
```
Is this something that should or needs to be added to the `N2kMessagesEnumToStr` header file?

Please let me know if I am misunderstanding something or if this is something that should be added to the library.
Thanks,

ttlappalainen, 2022-11-06:
You may be the first using N2kMessagesEnumToStr, which explains errors. Your notices are correct. I'll add fixed. Would be also probably best to change every 
const char* tN2kXTEModeStrs[] -> const char* const tN2kXTEModeStrs[]
When arrays is defined as const, on some boards (ESP32, Teensy) these will be totally saved to flash. With current definition they will be loaded to RAM.

agk1190, 2022-11-07:
Thank you for fixing it, Timo. This library has been amazing so far!

I was not aware that const char* const saved to flash on ESP32, so thank you for that info.

agk1190, 2022-11-27:
I see that the changes made it into `master` on Nov 26, 2022. Thanks!

closed 2022-11-27 by agk1190