ttlappalainen/NMEA2000 #294: LilyGo T-CAN485 testing

MENIER, 2022-11-09:
Hello,

I have tested :https://github.com/Xinyuan-LilyGO/T-CAN485/tree/main/example/Arduino/CAN 
with CAN_cfg.speed = CAN_SPEED_250KBPS;
I have this log on my nmea2000:
18:35:38: New extended frame from 0x09F80100, DLC 8, Data 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
18:35:38: New extended frame from 0x09F80200, DLC 8, Data 0x00 0xFC 0x00 0x00 0x00 0x00 0xFF 0xFF 
18:35:38: New extended frame from 0x0DF80900, DLC 8, Data 0x3C 0x3F 0x90 0x0A 0x7B 0x30 0x10 0xFF 
18:35:38: New extended frame from 0x18EEFFCF, DLC 8, Data 0x92 0x8B 0x61 0x4E 0x00 0xC3 0x78 0xC0 
18:35:38: New extended frame from 0x09F801CF, DLC 8, Data 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
18:35:38: New extended frame from 0x09F802CF, DLC 8, Data 0xCF 0xFC 0x00 0x00 0x00 0x00 0xFF 0xFF 
18:35:38: New extended frame from 0x0DF809CF, DLC 8, Data 0x49 0x4B 0xA0 0x31 0x7B 0x30 0x10 0xFF 
18:35:38: New extended frame from 0x09F801CF, DLC 8, Data 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
18:35:38: New extended frame from 0x09F802CF, DLC 8, Data 0xCF 0xFC 0x00 0x00 0x00 0x00 0xFF 0xFF 
18:35:38: New extended frame from 0x0DF809CF, DLC 8, Data 0x49 0x4B 0xA0 0x31 0x7B 0x30 0x10 0xFF 
18:35:39: New extended frame from 0x09F801CF, DLC 8, Data 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
18:35:39: New extended frame from 0x09F802CF, DLC 8, Data 0xCF 0xFC 0x00 0x00 0x00 0x00 0xFF 0xFF 
18:35:39: New extended frame from 0x0DF809CF, DLC 8, Data 0x49 0x4B 0xA0 0x31 0x7B 0x30 0x10 0xFF 
18:35:39: New extended frame from 0x11F80ECF, DLC 8, Data 0x00 0x1C 0x01 0xAB 0xCF 0xAE 0x14 0xF8 
18:35:39: New extended frame from 0x11F80ECF, DLC 8, Data 0x01 0xC8 0x9A 0xDB 0x68 0x49 0xB4 0x08 
18:35:39: New extended frame from 0x11F80ECF, DLC 8, Data 0x02 0x9F 0x64 0xF5 0x00 0x00 0x00 0x00 
18:35:39: New extended frame from 0x11F80ECF, DLC 8, Data 0x03 0x00 0xFF 0xFF 0x80 0xFF 0xCF 0xF8 
18:35:39: New extended frame from 0x11F80ECF, DLC 8, Data 0x04 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 0xFF 
18:35:39: New extended frame from 0x09F801CF, DLC 8, Data 0x50 0x34 0xB3 0x08 0x00 0xAF 0x9D 0xDB 

I test with your code adding this defines and a setup to enable sn65hvd231d

#define LILYGO_ES32_CAN
#ifdef LILYGO_ES32_CAN
//ttyACM0
//LILYGO_ES32_CAN sn65hvd231d free PINMAP 8x2 connector
//GND         |GPIO_NUM_25
//GPIO_NUM_32 |GPIO_NUM_33
//GPIO_NUM_5  |GPIO_NUM_12
//GPIO_NUM_34 |GPIO_NUM_35 // input mode only?
//GPIO_NUM_18 |VDD
//GND         |VDD
//LILYGO_ES32_CAN PINMAP
#define ESP32_CAN_TX_PIN GPIO_NUM_26
#define ESP32_CAN_RX_PIN GPIO_NUM_27
//#define ESP32_CAN_TX_PIN GPIO_NUM_13 //D9 13
//#define ESP32_CAN_RX_PIN GPIO_NUM_12 //D8 12
#define ESP32_CAN_SE     GPIO_NUM_23
#define PIN_5V_EN		 GPIO_NUM_16 //!!!Rx default Serial2 //CGQ_EN
#define RS485_EN         GPIO_NUM_17 //!!!Tx default Serial2
#define RS485_SE         GPIO_NUM_19
#define RS485_RX         GPIO_NUM_21
#define RS485_TX         GPIO_NUM_22
#define WS2812_LED_PIN   GPIO_NUM_4
#define SD-MOSI          GPIO_NUM_15
#define SD-MISO          GPIO_NUM_2
#define SD-SCLK          GPIO_NUM_14
#define SD-CS            GPIO_NUM_13
#define RX2              GPIO_NUM_32
#define TX2              GPIO_NUM_33
#else
#define ESP32_CAN_TX_PIN GPIO_NUM_13 //D9 13
#define ESP32_CAN_RX_PIN GPIO_NUM_12 //D8 12
#define RX2 16
#define TX2 17
#endif
setupLilyGoTCAN485()
{
        pinMode(PIN_5V_EN, OUTPUT); //voltage enable
	digitalWrite(PIN_5V_EN, HIGH);
	pinMode(ESP32_CAN_SE, OUTPUT);//sn65hvd231d  select
	digitalWrite(ESP32_CAN_SE, LOW);
}

Nothing on LilyGo T-CAN485 the same  program work find on other ESP32 wemos card
Is there some body who had experiment with LilyGo T-CAN485?

ttlappalainen, 2022-11-10:
Board seems to just have ESP32, so it should work with my library. 2 notes
- On schematics I see that it has internal 120 ohm termination resistor. If you already have on your bus proper termination, this adds extra resistor and causes problems.
- From schematics I read that CAN rx is on GPIO 26 and tx on GPIO 27. In your definition #define ESP32_CAN_TX_PIN GPIO_NUM_26 #define ESP32_CAN_RX_PIN GPIO_NUM_27 you set opposite.
- It is not isolated so take care you do no cause ground loops.

MENIER, 2022-11-10:
Thanks for your speed answer!

Your are true on your 2 notes:
1) there is an error on config.h from:[https://github.com/Xinyuan-LilyGO/T-CAN485/blob/main/example/Arduino/CAN/config.h]. I have post an issue on Xinyuan repository
 the good defines is:
`#define ESP32_CAN_TX_PIN GPIO_NUM_27
#define ESP32_CAN_RX_PIN GPIO_NUM_26`
2) I have totaly forget this 120 Ohms resistor (I have removed it)

your code log this now:
_**NMEA2000 ESP32 ARCH
ESP32_CAN_TX_PIN
27
ESP32_CAN_RX_PIN
26**_
But nothing else...

Is there a way masking the Xinyuan log to find the pgn nmea2000 (to shake any hardware problem)?
in this log exemple  frame from 0x09F801CF
18:35:39: New extended frame from 0x09F801CF, DLC 8, Data 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
18:35:39: New extended frame from 0x09F802CF, DLC 8, Data 0xCF 0xFC 0x00 0x00 0x00 0x00 0xFF 0xFF.

MENIER, 2022-11-11:
Thank you for your help! I will test your code on Xinyuan to see if it's the good pgns:
`void CanIdToN2k(unsigned long id, unsigned char &prio, unsigned long &pgn, unsigned char &src, unsigned char &dst)`
My device may sending information right all is ok on my autopilot raymarine. 

I mean that is your code logs beceause in your code there is:
```
void tNMEA2000_esp32::CAN_init() {
 Serial.println("NMEA2000 ESP32 ARCH");
Serial.println("ESP32_CAN_TX_PIN");
Serial.println(TxPin);
Serial.println("ESP32_CAN_RX_PIN");
Serial.println(RxPin);
```
I will test DataDisplay2 soon

ttlappalainen, 2022-11-11:
Those prints are not from my code. See https://github.com/ttlappalainen/NMEA2000_esp32/blob/master/NMEA2000_esp32.cpp

Note taht you have add tranceiver enable code to my examples as in your code above. Library expects tranceivers are enabled by default. LilyGo has enabel signal wired.

MENIER, 2022-11-14:
Your are true it's not your code, i probably patch your code some months ago (I don't remuber) because of random #define on Sloeber (need to clean code after each #define modification). I have tested  DataDisplay2 it doesn't run.
my setup (first of setup):
`void setupLilyGoTCAN485()
{
pinMode(PIN_5V_EN, OUTPUT); //voltage enable
digitalWrite(PIN_5V_EN, HIGH);
pinMode(ESP32_CAN_SE, OUTPUT);//sn65hvd231d select
digitalWrite(ESP32_CAN_SE, LOW);
delay(1000);
}`
modify setup DataDisplay2:
`void setup() {
  setupLilyGoTCAN485();
  Serial.begin(115200); delay(500);
  OutputStream=&Serial;
//   while (!Serial) 
   
//  NMEA2000.SetN2kCANReceiveFrameBufSize(50);
  // Do not forward bus messages at all
  NMEA2000.SetForwardType(tNMEA2000::fwdt_Text);
  NMEA2000.SetForwardStream(OutputStream);
  // Set false below, if you do not want to see messages parsed to HEX withing library
  NMEA2000.EnableForward(false);
  NMEA2000.SetMsgHandler(HandleNMEA2000Msg);
//  NMEA2000.SetN2kCANMsgBufSize(2);
  NMEA2000.Open();
  OutputStream->print("Running...");
}`
my defines (first after the include Arduino.h) :
`#define LILYGO_ES32_CAN
#ifdef LILYGO_ES32_CAN
#define PIN_5V_EN        GPIO_NUM_16
#define ESP32_CAN_TX_PIN GPIO_NUM_27
#define ESP32_CAN_RX_PIN GPIO_NUM_26
#define ESP32_CAN_SE     GPIO_NUM_23
#define RS485_EN         GPIO_NUM_17
#define RS485_SE         GPIO_NUM_19
#define RS485_RX         GPIO_NUM_21
#define RS485_TX         GPIO_NUM_22
#define LED_WS2812       GPIO_NUM_4
#define SD_MOSI          GPIO_NUM_15
#define SD_MISO          GPIO_NUM_2
#define SD_SCLK          GPIO_NUM_14
#define SD_CS            GPIO_NUM_13

#else
#define ESP32_CAN_TX_PIN GPIO_NUM_13 //D9 13
#define ESP32_CAN_RX_PIN GPIO_NUM_12 //D8 12`
my new log from Xinyuan code (the GPS was not connected on my AIS VHF in my last log it's probably explain strange 129025 position from source 207  ) :
**New extended framePGN,127237,headingTrackControl,Priority,2,Source,172,Destination,255, from 0x09F105AC, DLC 8, Data 0xA2,0xFF,0x7F,0xFF,0x7F,0xFF,0x7F,0xFF,
New extended framePGN,127237,headingTrackControl,Priority,2,Source,172,Destination,255, from 0x09F105AC, DLC 8, Data 0xA3,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,
New extended framePGN,127250,vesselHeading,Priority,2,Source,204,Destination,255, from 0x09F112CC, DLC 8, Data 0xFF,0x3F,0x43,0xFF,0x7F,0xFF,0x7F,0xFD,
New extended framePGN,127251,rateOfTurn,Priority,2,Source,204,Destination,255, from 0x09F113CC, DLC 8, Data 0xFF,0x70,0xAD,0xFD,0xFF,0xFF,0xFF,0xFF,
New extended framePGN,127257,attitude,Priority,2,Source,204,Destination,255, from 0x09F119CC, DLC 8, Data 0xFF,0x3F,0x43,0xBA,0x00,0xCD,0x03,0xFF,
New extended framePGN,127245,rudder,Priority,2,Source,172,Destination,255, from 0x09F10DAC, DLC 8, Data 0xFC,0xF8,0xFF,0x7F,0xF2,0x14,0xFF,0xFF,
New extended framePGN,127250,vesselHeading,Priority,2,Source,204,Destination,255, from 0x09F112CC, DLC 8, Data 0xFF,0x3A,0x43,0xFF,0x7F,0xFF,0x7F,0xFD,
New extended framePGN,127251,rateOfTurn,Priority,2,Source,204,Destination,255, from 0x09F113CC, DLC 8, Data 0xFF,0x4A,0xB4,0xFD,0xFF,0xFF,0xFF,0xFF,
New extended framePGN,127257,attitude,Priority,2,Source,204,Destination,255, from 0x09F119CC, DLC 8, Data 0xFF,0x3A,0x43,0xB3,0x00,0xD7,0x03,0xFF,
New extended framePGN,127245,rudder,Priority,2,Source,204,Destination,255, from 0x09F10DCC, DLC 8, Data 0x00,0xFF,0xFF,0x7F,0xF2,0x14,0xFF,0xFF,
New extended framePGN,127245,rudder,Priority,2,Source,204,Destination,255, from 0x09F10DCC, DLC 8, Data 0xFC,0xF8,0xFF,0x7F,0xFF,0x7F,0xFF,0xFF,
New extended framePGN,65359,seatalkPilotHeading,Priority,7,Source,204,Destination,255, from 0x1CFF4FCC, DLC 8, Data 0x3B,0x9F,0xFF,0xFF,0xFF,0x3A,0x43,0xFF,
New extended framePGN,127245,rudder,Priority,2,Source,172,Destination,255, from 0x09F10DAC, DLC 8, Data 0xFC,0xF8,0xFF,0x7F,0xCF,0x14,0xFF,0xFF,
New extended framePGN,129033,timeDate,Priority,3,Source,219,Destination,255, from 0x0DF809DB, DLC 8, Data 0x6E,0x4B,0x60,0x3A,0x9F,0x1B,0x10,0xFF,
New extended framePGN,127250,vesselHeading,Priority,2,Source,204,Destination,255, from 0x09F112CC, DLC 8, Data 0xFF,0x35,0x43,0xFF,0x7F,0xFF,0x7F,0xFD,
New extended framePGN,127251,rateOfTurn,Priority,2,Source,204,Destination,255, from 0x09F113CC, DLC 8, Data 0xFF,0xD6,0xA2,0xFD,0xFF,0xFF,0xFF,0xFF,
New extended framePGN,127257,attitude,Priority,2,Source,204,Destination,255, from 0x09F119CC, DLC 8, Data 0xFF,0x35,0x43,0xB2,0x00,0xD8,0x03,0xFF,
New extended framePGN,127245,rudder,Priority,2,Source,204,Destination,255, from 0x09F10DCC, DLC 8, Data 0x00,0xFF,0xFF,0x7F,0xCF,0x14,0xFF,0xFF,
New extended framePGN,127237,headingTrackControl,Priority,2,Source,172,Destination,255, from 0x09F105AC, DLC 8, Data 0xC0,0x15,0x3C,0xC0,0x1F,0xFF,0x7F,0xFF,
New extended framePGN,127250,vesselHeading,Priority,2,Source,204,Destination,255, from 0x09F112CC, DLC 8, Data 0xFF,0x30,0x43,0xFF,0x7F,0xFF,0x7F,0xFD, 
New extended framePGN,127251,rateOfTurn,Priority,2,Source,204,Destination,255, from 0x09F113CC, DLC 8, Data 0xFF,0xCD,0x84,0xFD,0xFF,0xFF,0xFF,0xFF,**

MENIER, 2022-11-15:
I have patch your code NMEA2000_esp32.cpp in CAN_init() replacing:
MODULE_CAN->IER.U = 0xff;
to
MODULE_CAN->IER.U = 0xef;
IT'S RUN ALL IS OK
Some body can explain that?

ttlappalainen, 2022-11-15:
Are you sure you are using latest NMEA2000 library and NMRA2000_esp32? If you look code on github https://github.com/ttlappalainen/NMEA2000_esp32/blob/master/NMEA2000_esp32.cpp line 169, you see that ith has same setting and also comment why.

MENIER, 2022-11-15:
Your are true i haven't done any  software update on my NMRA2000_esp32 library. (I haven't any trouble with the old one with any board until this lilygo T-CAN485.)

ttlappalainen, 2022-11-16:
I prefer to check and update both libraries time to time. There my be important updates to improve stability. 

I do not prefer to do any patches to my libraries, since they will be then destoyed in updates. If there is some required changes, proper way it to open issue so they cam be fixed and make available for all users.

MENIER, 2022-11-16:
Thanks to your grate work about the NMEA2k!. 
Do you see any problem to convert the RS485 input to RS232 (NMEA183 to input my depth sounder) removing the resitors 120Ohms R3 R6 R8 (differentiall to comun mode)?

ttlappalainen, 2022-11-17:
RS485 is not RS232. Also NMEA0183 devices normally uses RS422. You should first study differencies of those physical layers, check the LilyGO 485 chip specifications and your depth sounder NMEA0183 specifications. Also note that even that would fit electrically, you connect depthsounder tx to LilyGO, but then accidentaly turn RS485 to output, there is risk that one of those will burn.

MENIER, 2023-01-13:
> RS485 is not RS232. Also NMEA0183 devices normally uses RS422. You should first study differencies of those physical layers, check the LilyGO 485 chip specifications and your depth sounder NMEA0183 [...]

closed 2023-01-13 by MENIER