ttlappalainen/NMEA2000 #295: PGN127750 Converter (Inverter/Charger) Status

gorlanovmax, 2022-11-15:
Hello dear Timo,

Have you heard about PGN127750?
It is specified on https://canboat.github.io/canboat/canboat.xml#pgn-list

Can we add it in library?

ttlappalainen, 2022-11-16:
Yes. It is specified on NMEA2000 and described on canboat.

Yes, if you have way to test it. Follow the coding and naming style and way for types in N2kTypes.h and NMEA2000StdTypes.h You need to create new types:
- DD342, Operating state
- DD343, Temperature state
- DD345, Low DC voltage state
- DD346, Ripple state

gorlanovmax, 2022-11-16:
I think, that it will be useful for another users.
May be add in repo.

Needed to add in Types, StdTypes and N2kMessages (h and cpp) if I right understood

ttlappalainen, 2022-11-16:
Maybe. I have not seen it in real life on any boater.

Yes.

gorlanovmax, 2022-11-16:
It is used, for example, for ship shore connection monitoring

ttlappalainen, 2022-11-16:
I know, but I do not kow that rich boat owners.

gorlanovmax, 2022-11-19:
Also in tN2kTempSource needed to add 
N2kts_ShaftSealTemperature=15