ttlappalainen/NMEA2000 #298: Teensy 4.1 CAN pins

diverd999, 2022-11-28:
Hi All,

All working fine so far on my project thanks to this great resource. I have a simple question though.

I'm using a Teensy 4.1 & have used the WindMonitor example as a starting point. It defaults to using CRX1 & CTX1 but I'd like to use one of the other CAN ports. What do I need to change to use CRX2/CTX2 or CRX3/CTX3?

Thanks Dave

ttlappalainen, 2022-11-29:
This is described on beginning of NMEA2000_CAN.h. If you use that, you just add define for overriding default settings.
```
#define NMEA2000_TEENSYX_CAN_BUS tNMEA2000_Teensyx::CAN2 // select CAN bus 2
#include <NMEA2000_CAN.h>
```

If you specify library object by yourself, then you define settings on constructor:
#include <NMEA2000_Teensyx.h>    // https://github.com/ttlappalainen/NMEA2000_Teensyx
tNMEA2000 &NMEA2000=*(new tNMEA2000_Teensyx(tNMEA2000_Teensyx::CAN2));

diverd999, 2022-11-29:
Hi Timo, thanks for the quick reply. I’ll give it a go today.

Thanks for all the work. It’s a fantastic resource.

Best regards,
Dave Fry