ttlappalainen/NMEA2000 #299: Actisense to UDP  -- (or a SeaSmart program to analyse N2K ?)

dagnall53, 2022-12-07:
I know that this has been asked earler , but I could not see an answer I could employ.

I would very much like to use the Free Actisense "NMEA Reader". and wirelessly explore what I have on N2000.
 Actisense program does not have UDP, but I can "connect" a UDP stream to a port it can look at: using HW Virtual Serial Port. I believe I can then simply "UDP" the strem that would otherwise have been sent Serial, and the Actisence program should work. 

I was therefore hoping that I could then find some code to allow me to modify the  **HandleStreamN2kMsg** example code, where it sends the msg very simply with : **`NMEA2000.SendMsg(N2kMsg,-1);`**
So i was hoping i could send to udp with **`UDPSEND(N2kMsg.Data);`**
where my UDPSEND(const char * buf) function contains Udp.print(buf); 
so of course I get a "invalid conversion from 'const unsigned char*' to 'const char*' [-fpermissive]" when I try to compile.

I have tried reinterpret_cast to give me a **`UDPSEND(reinterpret_cast<const uint8_t*>(N2kMsg.Data))`**

But that is also incorrect, and way beyond my competence level. I would greatly appreciate if you could show me the correct recast ..?

Also. it was mentioned that there are (were?) equivalent programs to the Actisense NMEA reader for SeaSmart.. but I cannot find any!.  That would obviously be a an alternate strategy for me. 

Many thanks for a fine library. 
Dagnall

ttlappalainen, 2022-12-07:
You are trying to send raw N2k data
You should use NMEA2000.SendInActisenseFormat(Udp);
where Udp is Stream class.

For analyzing my NMEA Simulator is far more better than Actisense and also listens UDP. Unfortunately only licenced version shows parsed messages.

dagnall53, 2022-12-07:
.. but I cannot seem to get it work.. :-( 
I had assumed you just placed in the setup..

is there a chance you could modify the actisenselistnersender example to show me how it works? 
I'm getting compile errors.. 'class tNMEA2000' has no member named 'SendInActisenseFormat'
Second.. what is your NMEA Simulator? is it the $300 one?
Cheers.

ttlappalainen, 2022-12-08:
Sorry my mistake - should have been N2kMsg.SendInActisenseFormat(Udp);

You do not need ActisenseListenerSender since you do not send anything. ActisenseListenerSender can be used e.g., with my NMEA Simulator sw.

It is enough to use example ActisenseListener. Replace line 24 with
NMEA2000.SetForwardStream(&Udp);

Commercial NMEA2000 Simulator with hw box is more expensive. https://acrossoceansystems.com/index.php/product/nmea2000-nmea0183-simulator/ It does not yet have WiFi. If you need it only for bus data analyzing via UDP and you get your device working, I can provide you some day test license. If you like you can buy just sw license from me, but you have to contact me with email for that.

dagnall53, 2022-12-11:
Sorry to keep on this, but thius seems to trigger some unsigned char to char issues ?/  Strange, as it sends fine over serial..
When I added the code to my modified  ActisenseListenerSender.ino, it crashes. ,
And Arduino exception decoder says:
' `Decoding stack results
0x4015468f: WiFiUDP::write(unsigned char) at C:\Users\admin\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.5\libraries\WiFi\src\WiFiUdp.cpp line 194
0x401546ad: WiFiUDP::write(unsigned char const*, unsigned int) at C:\Users\admin\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.5\libraries\WiFi\src\WiFiUdp.cpp line 201
0x400d4d3e: tN2kMsg::SendInActisenseFormat(Stream*) const at C:\Users\admin\Documents\Arduino\libraries\NMEA2000-master\src\N2kMsg.cpp line 876
0x400d5bf2: tNMEA2000::ForwardMessage(tN2kMsg const&) at C:\Users\admin\Documents\Arduino\libraries\NMEA2000-master\src\NMEA2000.cpp line 1770
0x400d5c19: tNMEA2000::ForwardMessage(tN2kCANMsg const&) at C:\Users\admin\Documents\Arduino\libraries\NMEA2000-master\src\NMEA2000.cpp line 1780
0x400d7e7d: tNMEA2000::ParseMessages() at C:\Users\admin\Documents\Arduino\libraries\NMEA2000-master\src\NMEA2000.cpp line 2292
0x400d2a18: loop() at C:\Users\admin\Documents\Arduino\ActisenseListenerSender/ActisenseListenerSender.ino line 100
0x400df1e9: loopTask(void*) at C:\Users\admin\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.5\cores\esp32\main.cpp line 50` `

In case you can spot something stupid I have done .. my sketch is here.. I simply swap the commenting on lines 81 and 82.

`// Demo: NMEA2000 library. Bus listener and sender. 
//   Sends all bus data to serial in Actisense format.
//   Send all data received from serial in Actisense format to the N2kBus.
//   Use this e.g. with NMEA Simulator (see. http://www.kave.fi/Apps/index.html) to send simulated data to the bus.
//   I have plans to add receiving functionality and data forwarding to NMEA Simulator. Meanwhile you can define
//   other stream to different port so that you can send data with NMEA Simulator and listen it on other port with 
//   Actisense NMEA Reader.

//#define N2k_CAN_INT_PIN 21
#define ESP32_CAN_TX_PIN GPIO_NUM_5 // Uncomment this and set right CAN TX pin definition, if you use ESP32 and do not have TX on default IO 16
#define ESP32_CAN_RX_PIN GPIO_NUM_4 // Uncomment this and set right CAN RX pin definition, if you use ESP32 and do not have RX on default IO 4

#include <Arduino.h>
#include <N2kMsg.h>
#include <NMEA2000.h>
#include <NMEA2000_CAN.h>
//#include "N2kDataToNMEA0183.h"
#include <ActisenseReader.h>

#include "Defines.h"
#include "WiFi_Event.h"
#include <ESPmDNS.h>
#include <WiFi.h>
#include <WiFiUdp.h>
#include <WebServer.h>
tActisenseReader ActisenseReader;

// Define READ_STREAM to port, where you write data from PC e.g. with NMEA Simulator.
#define READ_STREAM Serial       
// Define ForwardStream to port, what you listen on PC side. On Arduino Due you can use e.g. SerialUSB
#define FORWARD_STREAM Serial    

// Set the information for other bus devices, which messages we support
 // tN2kDataToNMEA0183 tN2kDataToNMEA0183(&NMEA2000, 0);
const unsigned long TransmitMessages[] PROGMEM={0};
const unsigned long ReceiveMessages[] PROGMEM={/*126992L,*/ // System time
                                              127250L, // Heading
                                              127258L, // Magnetic variation
                                              128259UL,// Boat speed
                                              128267UL,// Depth
                                              129025UL,// Position
                                              129026L, // COG and SOG
                                              129029L, // GNSS
                                              130306L, // Wind
                                              0};

#define MAX_NMEA2000_MESSAGE_SEASMART_SIZE 500 
//*****************************************************************************

 char* ssid     = "----------";
 char* password = "-----------";
extern WiFiUDP Udp;

Stream *ReadStream=&READ_STREAM;
Stream *ForwardStream=&FORWARD_STREAM;

//NMEA 2000 message handler
void HandleNMEA2000Msg(const tN2kMsg &N2kMsg) {
 // This is called on every message Rx, but I cannot get it to send over UDP.. 
 //incompatible types in assignment of 'const unsigned char [223]' to 'char [223]'
 //char buf[223];
 // buf = N2kMsg.Data; // recast ? 
 //  SendBufToUDP(buf);
 Serial.println("GOT N2K MESSAGE"); //NB this does not affect the Actisense Reader!
 
}

void setup() {

  SetWIFI("NMEA2K_ALS","","",ssid,password,"192.168.0.120",AP_and_STA); // Sets AP  EXT etc... and port 2000 for Udp  
  StartWiFi();
  // included in setup ?StartWiFiPorts(true);
  // Define buffers big enough
  NMEA2000.SetN2kCANSendFrameBufSize(300);
  NMEA2000.SetN2kCANReceiveFrameBufSize(300);
  
//  if (ReadStream!=ForwardStream) READ_STREAM.begin(115200);
  FORWARD_STREAM.begin(115200);
  //NMEA2000.SetForwardStream(ForwardStream);
  NMEA2000.SetForwardStream(&Udp); //(CRASHES) on send on Udp?? port 2000  UDP??  ?TL email of 8 dec 2022   
  NMEA2000.SetMode(tNMEA2000::N2km_ListenAndSend);
  // This (next) needs to be commented out for the ActisenseNMEA reader to work . 
  // but helps if just looking at serial  
  //NMEA2000.SetForwardType(tNMEA2000::fwdt_Text); // Show bus data in clear text
  NMEA2000.SetMsgHandler(&HandleNMEA2000Msg);

  NMEA2000.Open();
}

void HandleStreamN2kMsg(const tN2kMsg &N2kMsg) {
// Only called by the RECIEVE portion!!
 // NMEA2000.SendMsg(N2kMsg,-1);   
}

void loop() {
  CheckWiFi();
  NMEA2000.ParseMessages();
//  ActisenseReader.ParseMessages();
}
`

dagnall53, 2022-12-19:
Potentially solved.. 
It works with TCP but not UDP..

closed 2022-12-19 by dagnall53

ttlappalainen, 2022-12-20:
UDP should also have stream?

Other solution is not to use automatic forward stream - remove NMEA2000.SetForwardStream(&Udp); Instead set message handler and on message handler test that Udp is ok and then call N2kMsg.SendInActisenseFormat(&UDP);

dagnall53, 2022-12-20:
Thanks.. 
That is actually what I have done, I made a (copy) of the standard handler, but once I have the message ready,, I have three booleans that allow me to optionally send to UDP,TCP or Serial, depending on application. 
They all seem to work, and send data. But the “HW Virtual Serial port” sw I was using to link the wireless unit to the (Serialonly input) Actisense NMEA reader only seems to work correctly with the TCP link. But that is enough for me!..  Where the unit is USB accessible, then the serial link is obviously simpler.

ttlappalainen, 2022-12-21:
So does UDP sending work or does it crash? My NMEA Simulator has option to read actisense format from UPD stream.

dagnall53, 2022-12-21:
With the approach I used, UDP sending works. (At least does not crash!!!) .
I suspect it's just the HWVirtual port (HW-group.com) that does not like it!

PS &FYI.. I am in conversation with Andras, as I am exploring his AIS
conversions 0183>N2K , and they seem to work, he has also got some
additional 0183 Handlers, (GLL,MTW DBT MWD ) but they seem to need some
additions to your NMEA0183Messages library?