ttlappalainen/NMEA2000 #301: Not sure if PGN is coming across properly.

sergei, 2022-12-13:
I'm sending the SetN2kPGN130306 message from my ESP32 board, and that what I see on the debug output:

`ESP32N2kStream: 1291732 : Pri:2 PGN:130306 Source:15 Dest:255 Len:8 Data:f2,99,1,0,0,2,ff,ff`

I'm using the YDNU-02 USB gateway to monitor it. That what it shows while in RAW mode:

`00:09:47.511 R 0000020F F2 99 01 00 00 02 FF FF`

The payload seems to match, but I'm not sure if PGN came across properly. Sorry I'm not sure what RAW mode is, but I expected to see 130306 or 0x1FD02 somewhere.

Any help or pointers are appreciated. Just in case, my code is [here](https://github.com/sergei/sci2000/blob/master/mhu2nmea/main/N2KHandler.cpp)

ttlappalainen, 2022-12-13:
I do not have YDNU, so I can not test. Check that your hw is properly build up and you have proper termination. You could also switch YDNU to actisense mode and listen it with my NMEA Simulator.

You have vTaskDelay(1000 / portTICK_PERIOD_MS); in your loop, which I expect to generate 1000 ms delay. That will not work. You should run NMEA2000.ParseMessages(); without delay or something around 0.1 - 0.5 ms delay.

For synchronizing messages with NMEA2000 loop there is tN2kSyncScheduler. There is example e.g., on TemperatureMonitor. With vTaskDelay you get cumulative error to message sending.

sergei, 2022-12-13:
Thanks a lot. It seems I fell victim of my own arrogance. While struggling with low level CAN communications I implemented my own version of tNMEA2000_esp32 using ESP IDF TWAI driver (you can see it [here](https://github.com/sergei/sci2000/tree/master/mhu2nmea/components/NMEA2000_esp32_twai)). While knowing nothing about N2K data format I didn't set the id format to extended, quite naturally PGN is not transmitted correctly.

ttlappalainen, 2022-12-14:
What is better with twai driver? My current driver has been used in commercial NMEA2000 certified products. It is also independent of other libraries.

Remember to fix use of NMEA2000.ParseMessages();

sergei, 2022-12-14:
The sole reason I used the TWAI driver, is to get error information while struggling with CAN bus (see [here](https://github.com/hatlabs/discussions/discussions/38#discussioncomment-4392062), now might well go back to the proven solution.  And yes, I will change the SW based on your example.

sergei, 2022-12-14:
I'm a happy camper now. Can see data  decoded by USB gateway:
[image]

sergei, 2022-12-14:
> You should run NMEA2000.ParseMessages(); without delay or something around 0.1 - 0.5 ms delay.
The best I can do is call xQueueReceive with xTicksToWait set to 1. It translate to every 10 ms. If I set it to 0, then I starve the IDLE task and the watchdog kicks in. To make FSM bit lively I sent events to this queue from the driver each time the frame is either received or transmitted. 

What problems would I see if NMEA2000.ParseMessages() is called every 10ms instead of 0.1 ms?

ttlappalainen, 2022-12-16:
If you have transmit and receive with buffers big enough and with interrupts (forced on my ESP32 driver), you will see some variation in automated message timing and small but should not be too big on automates responces. If you are not certifying the device, it should work OK.

sergei, 2022-12-16:
Thanks a lot. No certification at all, just upgrading my own boat ancient B&G NET to NMEA 2000 one device at a time. Can't stomach full upgrade price at once :-)

closed 2022-12-16 by sergei