ttlappalainen/NMEA2000 #302: Am I doing storing/reading calibration properly?

sergei, 2022-12-16:
Could you please comment if I'm storing/reading my device calibration properly? 

I added two custom PGNs one to store calibration (130900)  and another to report it (130901). 

In order for 130901 to be transmitted I used the NMEA2000.SetISORqstHandler() to set the [callback](https://github.com/sergei/sci2000/blob/74f06bd73c3df1eb13fb7457d770a52bb076c56b/mhu2nmea/main/N2KHandler.cpp#L212) where I send the 130901.

Am I on a right path?  Is there some generic N2K PGNs to store/retrieve generic configuration items?

ttlappalainen, 2022-12-17:
Each proprietary message must start with fields
Field 1: MfgCode 11 bits
Field 2: reserved  2 bits. Must be set all 1
Field 3: Industry code 3 bits. Use Marine=4
After those fields you can have your own fields.

What is the point of having  unsigned char sid = N2kMsg.GetByte(Index);? I think you have not read meaning for SID.

Also what is point to queue event on HandleMsg? When you call NMEA2000.ParseMessages(); it will do callback to your HandleMsg and you can handle it immediately. Now you put it to queue and handle it on next loop on same context. It is just wasting of resources.

You can either have 2 proprietary messages - one for current settings and other for commanding. 

NMEA2000 way is to have just one for current settings and then you use group function commands for changing data. If your current setting PGN 130900 contains fields:
Field 1: MfgCode 11 bits
Field 2: reserved  2 bits. Must be set all 1
Field 3: Industry code 3 bits. Use Marine=4
Field 4: AWAOffset 2 bytes
Field 5: AWSMultiplier 2 bytes

This can be then requested with ISO request sourced or broadcast. Note that, when you receive message, you have to first check that fields 1,3 matches since there may be other device using same proprietary PGN. Other way is to request it with group funtion request, where one can specify PGN and fields 1,3 for reaquest parameters.

For using group function you inherit tN2kGroupFunctionHandler and write handlers for request and command. See tN2kGroupFunctionHandlerForPGN60928 in N2kGroupFunctionDefaultHandlers. Both request and command must have first  matching fields 1,3. For command group funtion it is a bit confusing since it must start with fields 1,3 even it does not set them.

You then register your group function handler with NMEA2000.AddGroupFunctionHandler(...);

Your PGN will be then requested by using PGN 126208 and by setting fields:
Field 1: Request Group Function Code = 0 (Request Message), 8 bits
Field 2: PGN = 130900, 24 bits
Field 3: Transmission interval = FFFF FFFF (Do not change interval), 32 bits
Field 4: Transmission interval offset = 0xFFFF (Do not change offset), 16 bits
Field 5: Number of Pairs of Request Parameters to follow = 2, 8 bits
Field 6: Field number of requested parameter = 1, 8 bits
Field 7: Value of requested parameter = mfg code, 16 bits
Field 8: Field number of requested parameter = 3, 8 bits
Field 9: Value of requested parameter = 4 (Marine), 8 bits

AWS offset will be then sent by sending 126208 with field data:
Field 1: Request Group Function Code = 1 (Command Message), 8 bits
Field 2: PGN = 130900, 24 bits
Field 3: Priority Setting = 0x8 (do not change priority),  4 bits
Field 4: Reserved =0xf, 4 bits
Field 5: Number of Pairs of Request Parameters to follow = 3, 8 bits
Field 6: Field number of commanded parameter = 1, 8 bits
Field 7: Value of commanded parameter = mfg code, 16 bits
Field 8: Field number of commanded parameter = 3, 8 bits
Field 9: Value of commanded parameter = 4 (Marine), 8 bits
Field 10: Field number of commanded parameter = 4, 8 bits
Field 11: Value of commanded parameter = AWSOffset, 16 bits

One option is to have PGN 130900 contains fields:
Field 1: MfgCode 11 bits
Field 2: reserved  2 bits. Must be set all 1
Field 3: Industry code 3 bits. Use Marine=4
Field 4: Data/Command, 1 byte
Field 5: AWAOffset, 2 bytes
Field 6: AWSMultiplier, 2 bytes

Then if you request it, you return Field 4: Data/Command = 0 (current settings)
If you want to change values you set Field 4: Data/Command = 1 (change settings)
- If you do not want to change some value, set it to NA
- if you want to restore defaults use value 0xfffe (restore default) for

sergei, 2022-12-17:
Awesome, thanks a lot for this detailed answer. Greatly appreciate it!

closed 2022-12-17 by sergei